raspberrypi/picamera2 #912: [OTHER] Autofocus value outside of 10

tennisparty, 2024-01-06:
Hello,

In an app I am setting an manual focus with this command which, according to the docs takes a LensPosition value between 0 and 10.
`picam2.set_controls({"AfMode": controls.AfModeEnum.Manual, "LensPosition": 10})`

I have also created an autofocus control. I am querying this via
`LensPosition = picam2.capture_metadata().get("LensPosition")`

Often the returned values from the autofocus command are above 10, in the range of 11 and 12. Is there a reason why the autofocus is returning values above 10?

Many thanks,
Sam

davidplowman, 2024-01-08:
I guess the problem is that it will generally depend on the type of module and we can't really document a meaningful limit without saying what module we're talking about. If you're referring to section 5.2.3 of the manual, it's telling you to look at the `camera_controls` to find the maximum, and using 10 only as a "reasonable" example for discussion. For the imx708, the maximum is in fact 12.  I guess it could be worded so as to make that clearer.

tennisparty, 2024-01-08:
Aah that makes sense. I'll adjust my app to suit, thanks so much, Sam