raspberrypi/picamera2 #913: set_overlay() for all outputs

MazrimT, 2024-01-06:
**Please only include one suggestion or features request per issue!**

**Is your feature request related to a problem? Please describe.**
The set_overlay() function only seems to work for previews and not for any actual output.
This is not clearly documented and also feels illogical to new user that it would not just generally be an overlay no matter the output.

**Describe the solution you'd like**
overlays should be taken into account in all outputs, wether it's just a file or FileOutput or FfmpegOutput is used.

**Describe alternatives you've considered**
in ffmpeg in general you can add things like overlays, so being able to specify more inputs to a single ffmpegoutput in ffmpeg format would work, but currently ffmpegoutput does not seem to work with input in the command.

**Additional context**
Add any other context or screenshots about the feature request here.

davidplowman, 2024-01-08:
Yes, the overlay feature is very deliberately only something for use on preview windows, for example for doing little bits of UI such as crosshairs, buttons or rectangles. It's not intended to be part of the final image.

If you want something on the final image, I'd recommend using the `pre_callback` to draw text or other graphics onto your images. There are examples of this kind of thing, for example [here](https://github.com/raspberrypi/picamera2/blob/main/examples/timestamped_video.py).

MazrimT, 2024-01-09:
Thank you for pointing me in the right direction!

closed 2024-01-09 by MazrimT