raspberrypi/picamera2 #916: JPEF or DNG images as buffers

dgalland, 2024-01-08:
It may be useful in certain applications to obtain the jpeg or dng image in a buffer rather than writing it to a file. This is not currently possible with the Request or Helpers methods. For example PiDNG convert can return the buffer if filename == "" but not the helper.

davidplowman, 2024-01-08:
Hi, does it not work to pass a BytesIO as the "file", a bit like [this](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_to_buffer.py) example? If it doesn't that would probably be a bug as I'm sure it's the intention!

dgalland, 2024-01-08:
Yes it probably works for Picamera2 methods but not for Request methods which ask for a fileName.
Something like this:
while .....  
    request = picam2.capture_request()
    array = request.make_array("raw")   or "main"
    convert in a jpeg or dng buffer

davidplowman, 2024-01-08:
I'm sorry, I still don't think I've understood what you want. So far as I know, this type of thing should work:
```
...
request = picam2.capture_request()
data = io.BytesIO()
request.save('main', data, format='jpeg')
```
Can you be more explicit about what it is that you can't do? Thanks!

dgalland, 2024-01-08:
Yes, this should work for a JPG or PNG but not for a DNG  ?

davidplowman, 2024-01-08:
PiDNG is a third party library (which we're very happy to be able to use, of course), so we would have to talk to the developer in question. I don't expect it would be difficult to fix, but obviously someone has to have the time to get round to it. It might be worth posting on the PiDNG repo, though I don't know if the author is able to monitor it very closely.

As a workaround one can always write to a memory file (in "/dev/shm"). It is a bit cumbersome, but probably gets you many of the benefits.

dgalland, 2024-01-08:
Yes but PIDNG is already able to return the buffer if filename=="" 
But this possibility disappears in the Helper.
This is just a comment because there are workarounds like using shm or calling directly  PiDNG by copying the Helper code

davidplowman, 2024-01-08:
Ah, I see. Well, I suppose I might arguably be happier to see PiDNG handle it because modules like PIL, cv2 do, though I'm not particularly averse to putting a wee bodge into `save_dng` to detect a BytesIO and work round it.

dgalland, 2024-01-13:
Hi David,
Yes but we would also have to change the Request method
    def save_dng(self, filename, name="raw"):
and also the Helpers ?
    def save_dng(self, buffer, metadata, config, filename):
The most consistent is perhaps to call PiDng with filename=""   ??

davidplowman, 2024-01-15:
Maybe have a look whether https://github.com/raspberrypi/picamera2/pull/923 fits the bill?

dgalland, 2024-01-15:
Yes that looks correct to me, thank you !

cpixip, 2024-01-16:
Great! - when will this show up in the official distribution?

davidplowman, 2024-01-17:
I'll see about getting another update out in the next week or so. In the meantime, you can always clone the repo or pip install directly from git to get early access.