raspberrypi/picamera2 #919: [BUG] Picamera2.global_camera_info returns no "Num" on devices with a single camera

signag, 2024-01-11:
On devices with multiple cameras, such as Pi5, Picamera2.global_camera_info() returns a dict which includes a "Num" element with values 0, 1.
On devices with a single camera only, such as Pi Zero, "Num" is not included in the returned dict.

It would have been convenient to initialize "Num" with 0 also in these cases because also there it is possible to instantiate a camera with cam=Picamera2(0)

davidplowman, 2024-01-12:
Hi, thanks for the report. I think the problem is actually that Pi Zeros are still running Bullseye which has not received the update to return the correct camera "Num". (Pi 4s and 5s get Bookwork, which has this change.) Would you be on a position to try out the 0_3_12_next branch in this repo? That's basically the Bullseye version with a few more recent patches.

Slightly longer term I believe the plan is to release a version of Bookworm for earlier Pis, so all this trouble will go away, but I'll have to check how that's progressing.

signag, 2024-01-15:
Hi, thanks for the reply. Actually, I came across the issue only on Bullseye systems.

I am working on a webserver, controlling Pi cameras ([https://github.com/signag/raspi-cam-srv](https://github.com/signag/raspi-cam-srv)), which also supports Pi 5 with two cameras. It was straightforward to work around the missing "Num" for the Pi Zero and Pi Zero 2 systems. 
Therefore, I did not try the suggested branch.

On the Pi 5, the "Num" value returned seems to be independent from the socket to which a camera is connected.
If only one camera is connected, it returns "Num" 0 in either socket. What are the rules if two cameras are connected? Is it possible to somehow know from the "Num" value which socket the camera is in?
I have seen (see [https://github.com/signag/raspi-cam-srv/blob/main/docs/Information.md](https://github.com/signag/raspi-cam-srv/blob/main/docs/Information.md)) that the "Location" parameter does not seem to distinguish the socket.

davidplowman, 2024-01-15:
The problem users have had is that when the system boots, the order in which cameras are discovered is not fixed. So sometimes "cam0" can appear first, sometimes "cam1" can appear first, and this determines which libcamera regards as camera 0 or 1.

However, the camera 'Id' that you get does indicate which connector the camera is attached to. There should be a number after `i2c@` which should identify it - the numbers will be different between Pi 5s and non-Pi 5s, but the same principle should hold, I believe. Using that you should be able work out the correct camera number to open. (On Bullseye images, where there's no "Num" field, I think you can just use the index of the camera in the list.)