raspberrypi/picamera2 #924: record full sensor area in different resolution

LeifSec, 2024-01-15:
Directly using `libcamera` one can record video of the full sensor area in different resolutions (in 16:9 ratio): `rpicam-vid  -t 5000 -o libcamera_vid.h264  --width <width> --height <height> --mode 2304:1296:10:P`

For `picamera2` I cannot figure out from the manual how to do this.

If I set full HD resolution not the total sensor area is used as for `libcamera-vid --with 1920 --height 1080 -o file.h264` :   (it is a imx708 sensor)

```
from picamera2 import Picamera2
picam2 = Picamera2()
video_configuration = picam2.create_video_configuration()
video_configuration["main"]["size"] = (1920, 1080)
picam2.configure(video_configuration)
print(picam2.video_configuration)
```
-> 

```
CameraConfiguration({'use_case': 'video', 'buffer_count': 6, 'transform': <libcamera.Transform 'identity'>, 'display': 'main', 'encode': 'main', 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'main': {'size': (1920, 1080), 'format': 'XBGR8888', 'stride': 7680, 'framesize': 8294400}, 'lores': None, 'raw': {'size': (2304, 1296), 'format': 'SBGGR10_CSI2P', 'stride': 2880, 'framesize': 3732480}, 'queue': True, 'sensor': {'output_size': (2304, 1296), 'bit_depth': 10}})
``` 

Trying to change sensor output_size
``` 
video_configuration["sensor"]["output_size"] = (1920, 1080)
``` 
did not change it: 
```
CameraConfiguration({'use_case': 'video', 'buffer_count': 6, 'transform': <libcamera.Transform 'identity'>, 'display': 'main', 'encode': 'main', 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'main': {'size': (1920, 1080), 'format': 'XBGR8888', 'stride': 7680, 'framesize': 8294400}, 'lores': None, 'raw': {'size': (2304, 1296), 'format': 'SBGGR10_CSI2P', 'stride': 2880, 'framesize': 3732480}, 'queue': True, 'sensor': {'output_size': (2304, 1296), 'bit_depth': 10}})
``` 

I do not want to zoom an area of the picture / sensor - so as I understand `"ScalerCrop" ` is not the right parameter to set.

davidplowman, 2024-01-15:
Hi, try something like this:
```
from picamera2 import Picamera2

picam2 = Picamera2
main = {'size': (1920, 1080)}
sensor = {'output_size': (2304, 1296)}
config = picam2.create_video_configuration(main, sensor=sensor)
```

LeifSec, 2024-01-15:
Tanks a lot - it works. (I do not know why setting `["sensor"]["output_size"] ` failed before.

Where is `(2304, 1206)` coming from? 

It is not in the camera properties:
``` 
>>> picam2.camera_properties
{'Model': 'imx708_wide_noir', 'UnitCellSize': (1400, 1400), 'Location': 2, 'Rotation': 180, 'PixelArraySize': (4608, 2592), 'PixelArrayActiveAreas': [(16, 24, 4608, 2592)], 'ColorFilterArrangement': 0, 'ScalerCropMaximum': (0, 0, 0, 0), 'SystemDevices': (20749, 20750, 20737, 20738, 20739)}
```

davidplowman, 2024-01-15:
It all depends on the "modes" the camera supports. Try looking at `picam2.sensor_modes` (always do this as first thing after making the picam2 object). This lists the different operating modes - after configuring the camera it will always report one of these. The imx708 has a full resolution (4608x2592) and 2x2 binned mode (2304x1296). Section 4.2.2.3 of the manual talks about this.

Akinetopsia1, 2024-02-16:
@davidplowman 
> Hi, try something like this:
> 
> ```
> from picamera2 import Picamera2
> 
> picam2 = Picamera2
> main = {'size': (1920, 1080)}
> sensor = {'output_size': (2304, 1296)}
> config = picam2.create_video_configuration(main, sensor=sensor)
> ```

i can't get this to work, i get 

```
TypeError: create_video_configuration() got an unexpected keyword argument 'sensor'
```

davidplowman, 2024-02-16:
What kind of Pi are you using and what OS (`cat /etc/os-release`). Maybe you're running Bullseye? I think the above only works on Bookworm. On Bullseye, try
```
from picamera2 import Picamera2

picam2 = Picamera2
main = {'size': (1920, 1080)}
raw = {'size': (2304, 1296)}
config = picam2.create_video_configuration(main, raw=raw)
```

Akinetopsia1, 2024-02-16:
Thanks for the quick reply!

i'm using a Raspberry Pi Zero 2 W with the latest OS release for it, which is still Bullseye indeed.

I'll try your updated method, however the problem seems to be something else as looking at the output the sensor appears to select the resolution i wanted to specify (selected unicam format):

```
Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
```

the output is still blurry as if upscaled from an even lower resolution and making it nearly impossible to focus. If i still have issues i'll open a new issue so as not to hijack this one.