raspberrypi/picamera2 #926: how to Optimize Raspberry Pi Camera V3 for Fast 12MP Image Capture in DIY Book Scanner

konos93, 2024-01-17:
I am currently working on a DIY book scanner project using a Raspberry Pi Camera V3 with 12 megapixels. I have tried using both libcamera and picamera2 to capture images, but I am facing performance issues. The capture process takes more than 6 seconds for each image, which is not ideal for my application.

has ever anybody use a camera on raspberry that capture images but not from the frame of the video? 
i need 12 mp because i need the most possible detail from the image of the book page.

i think that camera open and close in raspberry while i use the code but i did not find something that keep it open. ok probabply is of topic but the same sensors on smarphone can capture on the same resolution in much less than a sec .

```
import subprocess
import os
import re
import time
import RPi.GPIO as GPIO

output_folder = "your_output_folder"  # Replace with your desired output folder
base_filename = "image"
file_extension = "jpg"
gpio_pin = 18  # GPIO pin to use

if not os.path.exists(output_folder):
    os.makedirs(output_folder)

# Find the last image file in the folder
existing_files = [f for f in os.listdir(output_folder) if f.endswith(f".{file_extension}")]
if existing_files:
    last_file = max(existing_files, key=lambda x: int(re.search(r'\d+', x).group()))
    last_number = int(re.search(r'\d+', last_file).group())
else:
    last_number = 0

GPIO.setmode(GPIO.BCM)
GPIO.setup(gpio_pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    GPIO.wait_for_edge(gpio_pin, GPIO.FALLING)

    # Record the start time
    start_time = time.time()

    # Increment the counter
    last_number += 1

    # Generate the output filename
    output_filename = f"{base_filename}{last_number:04d}.{file_extension}"
    output_path = os.path.join(output_folder, output_filename)

    # Run the libcamera-jpeg command
    command = ["libcamera-jpeg", "-n", "-o", output_path]
    subprocess.run(command)

    # Calculate and print the elapsed time with 1 decimal place
    end_time = time.time()
    elapsed_time = round(end_time - start_time, 1)
    print(f"Image captured: {output_filename}, Elapsed Time: {elapsed_time} seconds")

```

result

```
User
[0:24:08.427953396] [2517]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:24:08.581691708] [2523]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:24:08.588225484] [2523]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:24:08.588380899] [2523]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:24:08.591796900] [2517]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-SBGGR10_CSI2P
[0:24:08.592701680] [2523]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
Mode selection for 4608:2592:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 10600
    SRGGB10_CSI2P,2304x1296/0 - Score: 8200
    SRGGB10_CSI2P,4608x2592/0 - Score: 1000
[0:24:13.816165872] [2517]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-YUV420 (1) 4608x2592-SBGGR10_CSI2P
[0:24:13.824487022] [2523]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
Still capture image received
Image captured: image0040.jpg, Elapsed Time: 9.4 seconds
```

dgalland, 2024-01-17:
Hi,

Running libcamera-jpeg as a command line in a subprocess is certainly not the most efficient way to do a capture, there is the cost of launching a process and the camera is restarted each time with probably an additional delay of stabilization.
You must therefore use picamera2 and take inspiration from the capture_jpeg.pi example

davidplowman, 2024-01-17:
> Hi,
>
> Running libcamera-jpeg as a command line in a subprocess is certainly not the most efficient way to do a capture, there is the cost of launching a process and the camera is restarted each time [...]

Agree 100%. Use Picamera2. Configure the camera in full resolution stills mode with at least 2 buffers. Leave the camera running.

Do the capture when the trigger happens.

If it's important that no pixel started to expose before the moment of the trigger, then call `capture_request` with `flush=True`.

It will look a bit like this (untested):
```
from picamera2 import Picamera2

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config)
picam2.start()

# wait for your trigger, then:
request = picam2.capture_request(flush=True)
request.save('main', "file.jpg")
request.release()
```

konos93, 2024-01-17:
i will try something like this 

import RPi.GPIO as GPIO
from picamera2 import Picamera2
import time
from datetime import datetime

# Set up GPIO 18 as an input with pull-up
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config)
picam2.start()

# Wait for GPIO 18 to go low
while GPIO.input(18):
    time.sleep(0.1)

# When GPIO 18 goes low, capture an image
request = picam2.capture_request(flush=True)

# Generate a timestamp and include it in the filename
timestamp = datetime.now().strftime("%Y%m%d_%H%M%S")
filename = f"file_{timestamp}.jpg"

request.save('main', filename)
request.release()

# Clean up GPIO
GPIO.cleanup()

konos93, 2024-01-18:
i am very satisfied with the results thanks for your comments and your great tools u develop 
i used that on rasbery pi 3b+

i will try to test with a second camera on raspberry pi5 .
how to use 2 camera with this  code ? (i am not sure here if i shall make a new question your repository ). will it be around 1 sec .
i will capture some book pages and i will upload them here asking more question about .

i used this with gpio 

```
import RPi.GPIO as GPIO
from picamera2 import Picamera2
import time
from datetime import datetime

# Set up GPIO 18 as an input with pull-up
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=1)
picam2.configure(config)
picam2.start()

try:
    while True:
        # Wait for GPIO 18 to go low
        while GPIO.input(18):
            time.sleep(0.1)

        # Record the start time
        start_time = time.time()

        # When GPIO 18 goes low, capture an image
        timestamp = datetime.now().strftime("%Y%m%d_%H%M%S")
        filename = f"file_{timestamp}.jpg"
        
        request = picam2.capture_request(flush=True)
        request.save('main', filename)
        request.release()

        # Record the end time
        end_time = time.time()

        # Calculate and print the time taken
        capture_time = end_time - start_time
        print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds")

        # Wait for GPIO 18 to go high again (to prevent multiple captures on a single press)
        while not GPIO.input(18):
            time.sleep(0.1)

except KeyboardInterrupt:
    pass
finally:
    # Clean up GPIO
    GPIO.cleanup()
    picam2.stop()
```
results

```
[1:26:37.145669248] [3223]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:26:37.311021339] [3229]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:26:37.317354895] [3229]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[1:26:37.317603540] [3229]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:26:37.327167389] [3223]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:26:37.427623251] [3232]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:26:37.433518476] [3232]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[1:26:37.433667329] [3232]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:26:37.452664456] [3223]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-SBGGR10_CSI2P
[1:26:37.453633356] [3232]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
Image captured: file_20240118_043222.jpg, Time taken: 1.06 seconds
Image captured: file_20240118_043224.jpg, Time taken: 1.09 seconds
Image captured: file_20240118_043226.jpg, Time taken: 0.95 seconds
```

and with keyboard 

```
import keyboard
from picamera2 import Picamera2
import time
from datetime import datetime

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=1)
picam2.configure(config)
picam2.start()

try:
    while True:
        # Wait for the space key to be pressed
        keyboard.wait("space")

        # Record the start time
        start_time = time.time()

        # Capture an image
        timestamp = datetime.now().strftime("%Y%m%d_%H%M%S")
        filename = f"file_{timestamp}.jpg"
        
        request = picam2.capture_request(flush=True)
        request.save('main', filename)
        request.release()

        # Record the end time
        end_time = time.time()

        # Calculate and print the time taken
        capture_time = end_time - start_time
        print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds")

except KeyboardInterrupt:
    pass
finally:
    # Clean up
    picam2.stop()
```

results i think is slightly slower

```
User
 [1:37:27.046278964] [3437]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:37:27.146640856] [3443]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:37:27.152654707] [3443]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[1:37:27.152879340] [3443]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:37:27.159286677] [3437]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:37:27.256174174] [3446]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:37:27.261940892] [3446]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[1:37:27.262091672] [3446]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:37:27.280776448] [3437]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-SBGGR10_CSI2P
[1:37:27.281792376] [3446]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
 Image captured: file_20240118_044312.jpg, Time taken: 1.88 seconds
 Image captured: file_20240118_044315.jpg, Time taken: 1.12 seconds
 Image captured: file_20240118_044317.jpg, Time taken: 1.01 seconds
 Image captured: file_20240118_044318.jpg, Time taken: 1.05 seconds

konos93, 2024-01-18:
on raspbery pi5 brilliant
 how fast would it be if i add a second camera? what changes should i do ?
 i didn't see so much different between keyboard and gpio 
Image captured: file_20240118_053824.jpg, Time taken: 0.29 seconds
Image captured: file_20240118_053825.jpg, Time taken: 0.39 seconds
Image captured: file_20240118_053826.jpg, Time taken: 0.38 seconds
Image captured: file_20240118_053827.jpg, Time taken: 0.41 seconds
Image captured: file_20240118_053828.jpg, Time taken: 0.38 seconds

davidplowman, 2024-01-18:
You can open a second camera with
```
picam2_1 = Picamera2(1)
```
and configure and start it in the same way.

To capture both images as quickly as possible after the trigger you could use a thread per camera. Each thread would wait for the trigger and do the capture in the same way as you did before. Doing it in a single thread is a bit  trickier because (I assume) you don't want to wait for the first capture to finish before asking for the second. Nonetheless, I think the 2 thread solution is probably better.

BTW, I notice you have `buffer_count=1` in your code. I would recommend increasing that to at least 2 if you can. The value 1 means that every other camera frame gets dropped, so that you're running at effectively half the framerate. This will increase the capture latency, though obviously there's no effect on the time spent encoding the image.

konos93, 2024-01-18:
ok i change the buffer to 3  should i do something for autofocus or for the module v3 doestn't need to ?

konos93, 2024-01-18:
[image]
it does not seem to auto focus

davidplowman, 2024-01-18:
TBH I'm not sure if it enables autofocus automatically or not, but it doesn't look like it. Check out section 5.2 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

konos93, 2024-01-18:
for some reason qt preview is gray .why ?
```
import RPi.GPIO as GPIO
import keyboard
from picamera2 import Picamera2
import time
from datetime import datetime
from libcamera import controls
import os

# Set up GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config)
picam2.start(show_preview=True)
picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})

try:
    while True:
        # Wait for the GPIO pin to be triggered (falling edge)
        GPIO.wait_for_edge(18, GPIO.FALLING)

        # Record the start time
        start_time = time.time()

        # Capture an image
        timestamp = datetime.now().strftime("%Y%m%d_%H%M%S")
        filename = f"z_img_{timestamp}.jpg"

        request = picam2.capture_request(flush=True)
        request.save('main', filename)
        request.release()

        # Get the size of the captured image in bytes
        file_size_bytes = os.path.getsize(filename)

        # Convert the file size to megabytes
        file_size_mb = file_size_bytes / (1024 ** 2)

        # Record the end time
        end_time = time.time()

        # Calculate and print the time taken and the size of the image in megabytes
        capture_time = end_time - start_time

        print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds, Size: {file_size_mb:.2f} MB")

except KeyboardInterrupt:
    pass
finally:
    # Clean up
    picam2.stop()
    GPIO.cleanup()
    
    ```
    
    results
    
    ```
    [2:17:23.969790214] [6849]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[2:17:23.984230690] [6860]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[2:17:23.994279532] [6860]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[2:17:23.996332571] [6849]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[2:17:24.017796535] [6863]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[2:17:24.031213862] [6863]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[2:17:24.040600482] [6849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[2:17:24.040722389] [6849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[2:17:24.040731426] [6849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[2:17:24.040740389] [6849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[2:17:24.041921761] [6849]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-BGGR16_PISP_COMP1
[2:17:24.042136076] [6863]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected CFE format: 4608x2592-PC1B
Image captured: z_img_20240118_184406.jpg, Time taken: 0.33 seconds, Size: 1.14 MB
Image captured: z_img_20240118_184407.jpg, Time taken: 0.31 seconds, Size: 1.18 MB
Image captured: z_img_20240118_184408.jpg, Time taken: 0.38 seconds, Size: 1.19 MB
Image captured: z_img_20240118_184408.jpg, Time taken: 0.36 seconds, Size: 1.19 MB
Image captured: z_img_20240118_184409.jpg, Time taken: 0.34 seconds, Size: 1.19 MB
```

davidplowman, 2024-01-18:
Don't know. Is this a colour or a NoIR camera? Do you have any other cameras to try?

konos93, 2024-01-18:
color camera i found that there also some apps 
https://github.com/raspberrypi/picamera2/tree/main/apps

i run [app_capture_af.py](https://github.com/raspberrypi/picamera2/blob/main/apps/app_capture_af.py)

lol
it does the job that i want 

but why 
show_preview=True doesnt work 

i mean in app_2 resuls is like this 

```
[2:36:30.168744736] [7530]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[2:36:30.182973516] [7536]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[2:36:30.193782837] [7536]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[2:36:30.195495669] [7530]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[2:36:30.219480716] [7539]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[2:36:30.236342215] [7539]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[2:36:30.240150785] [7530]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[2:36:30.240197248] [7530]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[2:36:30.240205526] [7530]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[2:36:30.240214951] [7530]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[2:36:30.241611672] [7530]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[2:36:30.241886468] [7539]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
```
with the qt app vieo

konos93, 2024-01-18:
ok i am done i am noob as hell this is what i need i should check more what u have develop thank u very much
https://github.com/raspberrypi/picamera2/blob/main/apps/app_full.py

konos93, 2024-01-19:
sorry that i reopen
this is some comments for app_full.py 
it needs some changes for diybookscanner like 
press space or use gpio to capture and save img like 
img_00040.jpg ,img_00041.jpg or with time like smartphones .
it also needs a second gui with preview and settings  for the second camera.  

now main examples from datasheet and scripts with qt preview work fine 

here why does not work the preview or stuck i mean its like starting it open a qt square with no image but cant press space or anthing 

import keyboard
from picamera2 import Picamera2, Preview
import time
from datetime import datetime
from libcamera import controls
import os

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config)
picam2.start(show_preview=False) # (show_preview=True) to see the grey qr code
#picam2.start_preview(Preview.DRM)
#picam2.start_preview(Preview.QTGL)
#picam2.start_preview(Preview.QT)
#picam2.start_preview(Preview.NULL)
picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous })
#picam2.set_controls({"AfMode": controls.AfModeEnum.Manual, "LensPosition": 4.5})# use the app_full.py to check the #better lens potition or use afmode and keep the images with the most size 

try:
    while True:
       
        # Wait for the space key to be pressed
        keyboard.wait("space")

        
        # Record the start time
        start_time = time.time()

        # Capture an image
        timestamp = datetime.now().strftime("%Y%m%d_%H%M%S")
        filename = f"z_img_{timestamp}.jpg"

        request = picam2.capture_request(flush=True)
        request.save('main', filename)
        request.release()

        # Get the size of the captured image in bytes
        file_size_bytes = os.path.getsize(filename)

        # Convert the file size to megabytes
        file_size_mb = file_size_bytes / (1024 ** 2)

        # Record the end time
        end_time = time.time()

        # Calculate and print the time taken and the size of the image in megabytes
        capture_time = end_time - start_time

        print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds, Size: {file_size_mb:.2f} MB")

except KeyboardInterrupt:
    pass
finally:
    # Clean up
    picam2.stop()

davidplowman, 2024-01-19:
Hi again, I'm afraid you're going to have to do some debugging. For example:

* Which line of code is getting stuck? Is it the `capture_request(...)`?
* Does it make a difference if you don't use `flush`?
* Does a different camera mode work? (e.g. try `create_preview_configuration()`)
* Do `libcamera-hello` and `libcamera-still` work?
* Are there any console messages?

As regards `app_full.py`, it's just an example. It's not designed for any purpose other than to educate users. If you want it to do something different, the software license lets you copy and reuse the code in any way that you want. If you would like to submit demo applications back to us, we're also very happy to take those.

konos93, 2024-01-25:
this seems to work fine for one camera and  preview with auto focus
i will add a second camera and a second preview window if i stuck again i will ask help here 
```
import keyboard
from picamera2 import Picamera2, Preview
import time
from datetime import datetime
from libcamera import controls
import os

picam2 = Picamera2()
config = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config)   
picam2.start_preview(Preview.QTGL)
preview_config = picam2.create_preview_configuration({"size": (4096, 2592)})
picam2.configure(preview_config)

picam2.start()

picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous })

try:
    while True:
       
        # Wait for the space key to be pressed
        keyboard.wait("space")

        
        # Record the start time
        start_time = time.time()

        # Capture an image
        timestamp = datetime.now().strftime("%Y%m%d_%H%M%S")
        filename = f"z_img_{timestamp}.jpg"

        request = picam2.capture_request(flush=False  )
        request.save('main', filename)
        request.release()

        # Get the size of the captured image in bytes
        file_size_bytes = os.path.getsize(filename)

        # Convert the file size to megabytes
        file_size_mb = file_size_bytes / (1024 ** 2)

        # Record the end time
        end_time = time.time()

        # Calculate and print the time taken and the size of the image in megabytes
        capture_time = end_time - start_time

        print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds, Size: {file_size_mb:.2f} MB")

except KeyboardInterrupt:
    pass
finally:
    # Clean up
    picam2.stop()
```

konos93, 2024-01-26:
does pi camera 2 support something like crosshair and grid lines ?

to use  one camera v3 
```
import keyboard
from picamera2 import Picamera2, Preview
import time
from datetime import datetime
from datetime import datetime
from libcamera import controls
import os

picam2 = Picamera2() # if u have one camera it will try to check it
#if u have alrerd two connected and wanna use only one use Picamera2(1) or Picamera2(1)
config = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config)   

picam2.start_preview(Preview.QTGL)
preview_config = picam2.create_preview_configuration({"size": (4096, 2592)})
picam2.configure(preview_config)

picam2.start()

picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous })

try:
    while True:
       
        # Wait for the space key to be pressed
        keyboard.wait("space")

        
        # Record the start time
        start_time = time.time()

        # Capture an image
        timeStamp = datetime.now().strftime("%Y%m%d-%H%M%S.%f")[:-4]
        filename = f"z_img_{timeStamp}.jpg"

        request = picam2.capture_request(flush=False  )
        request.save('main', filename)
        request.release()

        # Get the size of the captured image in bytes
        file_size_bytes = os.path.getsize(filename)

        # Convert the file size to megabytes
        file_size_mb = file_size_bytes / (1024 ** 2)

        # Record the end time
        end_time = time.time()

        # Calculate and print the time taken and the size of the image in megabytes
        capture_time = end_time - start_time

        print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds, Size: {file_size_mb:.2f} MB")

except KeyboardInterrupt:
    pass
finally:
    # Clean up
    picam2.stop_preview()
    picam2.stop()

```

for two camera v3 

```
import keyboard
from picamera2 import Picamera2, Preview
import time
from datetime import datetime
from libcamera import controls
import os

def capture_image(picam2, filename_prefix):
    # Record the start time
    start_time = time.time()

    # Capture an image
    timeStamp = datetime.now().strftime("%Y%m%d-%H%M%S.%f")[:-4]
    filename = f"{filename_prefix}_img_{timeStamp}.jpg"

    request = picam2.capture_request(flush=False)
    request.save('main', filename)
    request.release()

    # Get the size of the captured image in bytes
    file_size_bytes = os.path.getsize(filename)

    # Convert the file size to megabytes
    file_size_mb = file_size_bytes / (1024 ** 2)

    # Record the end time
    end_time = time.time()

    # Calculate and print the time taken and the size of the image in megabytes
    capture_time = end_time - start_time

    print(f"Image captured: {filename}, Time taken: {capture_time:.2f} seconds, Size: {file_size_mb:.2f} MB")

# Set up the first camera
picam1 = Picamera2(1)
config1 = picam1.create_still_configuration(buffer_count=3)
picam1.configure(config1)
picam1.start_preview(Preview.QTGL)
preview_config1 = picam1.create_preview_configuration({"size": (4096, 2592)})
picam1.configure(preview_config1)
picam1.start()
picam1.set_controls({"AfMode": controls.AfModeEnum.Continuous})

# Set up the second camera
picam2 = Picamera2(0)
config2 = picam2.create_still_configuration(buffer_count=3)
picam2.configure(config2)
picam2.start_preview(Preview.QTGL)
preview_config2 = picam2.create_preview_configuration({"size": (4096, 2592)})
picam2.configure(preview_config2)
picam2.start()
picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})

try:
    while True:
        # Wait for the space key to be pressed
        keyboard.wait("space")

        # Capture images from both cameras
        capture_image(picam1, "y")
        capture_image(picam2, "x")

except KeyboardInterrupt:
    pass
finally:
    # Close QtGL preview windows
    picam1.stop_preview()
    picam2.stop_preview()
    
    # Clean up both cameras
    picam1.stop()
    picam2.stop()

konos93, 2024-01-26:
ok i will use the app_capture_overlay.py before i use what script we made .

if u wanna have lens opposite from a glass u have to see in the middle the lens on a mirror  
[image]

and the scanner 

[image]

[davidplowman](https://github.com/davidplowman) if u ever be in athens let me know to crab u a beer .
your comments made a great work for future diybookscanner projects

konos93, 2024-01-26:
[image]

konos93, 2024-02-02:
em hello again
 i made this its a ^shape diybookscanner 
[image]

how to reduce the time that camera calibrate whiteness , brightness i dont know how to tell it 
it happen every time i flat a new page on the glass here is a video that probably help u
https://www.youtube.com/watch?v=aqaoUlu7En0

davidplowman, 2024-02-02:
Could you simply disable the AEC when you do the first capture?

konos93, 2024-02-02:
picam.set_controls({"AfMode": controls.AfModeEnum.Manual, "LensPosition": 5.1, "AeEnable": False})

works  beautiful .

if i use auto focus is there any way to check at what lens position the camera captured the page ?

davidplowman, 2024-02-02:
Yes, if you look in the image metadata of the captured image there should be some `'LensPosition'` metadata.

konos93, 2024-02-02:
should i use metadata_with_image.py or  i will use any program ?

davidplowman, 2024-02-02:
Looking at your code above, once you've captured your request you should be able to do:
```
lens_position = request.get_metadata()['LensPosition']
picam2.set_controls({'LensPosition': lens_position})
```

konos93, 2024-02-04:
is it possible to auto-focus on range of specific lenspotition around 4 and 6 ? 
i think that i need to change the way the preview work in what i made 

i used app_capture_af.py if i run it on two terminals for 2 camera it seems it work faster in the auto focus 
here is a video for help first is app_capture_af.py after 18 sec is  what i made  [(https://www.youtube.com/watch?v=0nzq46YIENE)]

the results are like that if i press one time capture button

```
>>> %Run app_capture_af.py
[0:04:40.089132883] [2470]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:04:40.114705396] [2479]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:04:40.127852140] [2479]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:04:40.127953231] [2479]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:04:40.128731185] [2479] ERROR V4L2 v4l2_device.cpp:353 'imx708': Unable to set controls: Device or resource busy
[0:04:40.140561332] [2479]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media3 using PiSP variant BCM2712_C0
[0:04:40.143722649] [2470]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:04:40.167272555] [2484]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:04:40.183179805] [2484]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:04:40.183282712] [2484]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:04:40.183956871] [2484] ERROR V4L2 v4l2_device.cpp:353 'imx708': Unable to set controls: Device or resource busy
[0:04:40.197139743] [2484]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media3 using PiSP variant BCM2712_C0
[0:04:40.200718019] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:04:40.200786592] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:04:40.200792240] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:04:40.200801333] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[0:04:40.201900006] [2470]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[0:04:40.202126040] [2484]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
[0:04:48.688420655] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:04:48.688495469] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:04:48.688503302] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:04:48.688512561] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[0:04:48.689610513] [2470]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-BGGR16_PISP_COMP1
[0:04:48.696944794] [2484]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected CFE format: 4608x2592-PC1B
[0:04:49.343240722] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:04:49.343311629] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:04:49.343318462] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:04:49.343328517] [2470]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[0:04:49.344277397] [2470]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[0:04:49.352228672] [2484]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B

```
if i run mine 
```
import keyboard
from picamera2 import Picamera2, Preview
import time
from datetime import datetime
from libcamera import controls
import os

def capture_image(picam, filename_prefix, save_path):
    # Record the start time
    start_time = time.time()

    # Capture an image
    timeStamp = datetime.now().strftime("%Y%m%d-%H%M%S.%f")[:-4]
    filename = os.path.join(save_path, f"{filename_prefix}_img_{timeStamp}.jpg")

    request = picam.capture_request(flush=False)
    request.save('main', filename)
    request.release()

    # Get the size of the captured image in bytes
    file_size_bytes = os.path.getsize(filename)

    # Convert the file size to megabytes
[... 87 lines cut ...]
[0:29:04.313816409] [5408]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:29:04.313862631] [5408]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:29:04.313870686] [5408]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[... cut ...]
[0:29:04.914585683] [5421]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected CFE format: 4608x2592-PC1B
Image captured: y_img_20240204-175520.39.jpg, Time taken: 0.24 seconds, Size: 1.72 MB
Image captured: x_img_20240204-175520.63.jpg, Time taken: 0.21 seconds, Size: 1.71 MB

```
with capture _af .py i can see better the range lens position should be . but still autofocus is neccesary because it may some letters be blurred .

so do u think if i change the way preview works will the scanner be faster ?

konos93, 2024-02-05:
ok again thanks for your help
all these issues happen after my first attempt to scan a whole book it was about 600 pager per hour and should increase it to 1100 .its necessary to have the least possible number of pages i need to recapture as a user of my scan.
 in app_capture af.py it is used something like a scale preview .that help to run focus faster and capture image in the best resolution . 
 
 i played around with my script trying to have a scale qt preview environment like app_capture af.py  ,
  i dont know exactly where these type of commands should go 
 picam2.switch_mode_and_capture_file(cfg, "test.jpg", signal_function=qpicamera2.signal_done)
 it didnt went well ,everything went black .but when restart and test some scripts  everything seems to work five , 

  can a deadlock destroy rasperi pi or any camera  ?
 
 after that i was afraid to try n error more stuff with camera.

 i made big buttons in app_capture _af . i open 2 terminals for each camera .in a third using puautogui i press space and pressed the buttons. its an ugly automation that seems to work fine for my needs .
 
 is there any keypress  in picmara2 .like press enter n capture?
 
using autofocus cycle seems better for book scanner than continuous . but still the staff with range could help in time n response .especially if there are blank pages or pages with letters only on the top like end of chapter.
i think that also can happen with the autofocus cycle after some 20 frames change to a specific manual focus n capture n return to af_cycle .
 af is neccecary because if u manual focus probably some text will be blurred . if i pressure with power the back of the book while i flat the pages lens potition numbers change . but with capture_af in the medatata_request is more helpful to check the lens_potition range .

konos93, 2024-02-05:
i tried to use it  with Preview.QTGL  and i faced black screen once and in the other the script is stopped. could this destroy raspberry or camera ? burn them or something . or everything will just power off

davidplowman, 2024-02-06:
I don't think anything should physically damage the Pi or the camera, so long as you don't plug the camera in or out while the Pi is powered, or connect anything to incorrect power supplies.

konos93, 2024-02-08:
hey again em i want focus in a whole area in the middle (rather than a spot ) .

1question how big is the area than normal focus used. 

do i use this part properly 

    picam2.set_controls({"AfMode": controls.AfModeEnum.Manual,
                         "LensPosition": lens_position_spinbox.value(),
                         "AfMetering" : controls.AfMeteringEnum.Windows,
                         "AfWindows" : [ (0, 0, 400, 500) ]}) #2 question is that in the middle????

here is what i used 

```
#!/usr/bin/python3

from libcamera import controls
from PyQt5 import QtCore
from PyQt5.QtGui import QPalette
from PyQt5.QtWidgets import (QApplication, QCheckBox, QHBoxLayout, QLabel,
                             QPushButton, QVBoxLayout, QWidget, QDoubleSpinBox)
from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2
import numpy as np
import time

STATE_AF = 0
STATE_CAPTURE = 1

def request_callback(request):
    lens_position = request.get_metadata().get('LensPosition', 'N/A')
    # Format the lens position to display up to 3 decimal places
    lens_position_str = f"{float(lens_position):.3f}" if lens_position != 'N/A' else 'N/A'
    label.setText(f"Lens Position: {lens_position_str}")
    

picam2 = Picamera2(1)
picam2.post_callback = request_callback
preview_width = 800
preview_height = picam2.sensor_resolution[1] * 800 // picam2.sensor_resolution[0]
preview_height -= preview_height % 2
preview_size = (preview_width, preview_height)
raw_size = tuple([v // 2 for v in picam2.camera_properties['PixelArraySize']])
preview_config = picam2.create_preview_configuration({"size": preview_size}, raw={"size": raw_size})
picam2.configure(preview_config)
if 'AfMode' not in picam2.camera_controls:
    print("Attached camera does not support autofocus")
    quit()
picam2.set_controls({"AfMode": controls.AfModeEnum.Auto,"AfSpeed": controls.AfSpeedEnum.Fast})
app = QApplication([])

def on_button_clicked():
    global state
    button.setEnabled(False)
    manual_focus.setEnabled(False)
    af_checkbox.setEnabled(False)
    state = STATE_AF if af_checkbox.isChecked() else STATE_CAPTURE
    if state == STATE_AF:
        picam2.autofocus_cycle(signal_function=qpicamera2.signal_done)
    else:
        do_capture()
    # Remove focus from QDoubleSpinBox
    lens_position_spinbox.clearFocus()

def on_continuous_af_toggled(checked):
    if checked:
        picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous,
                             "AfSpeed": controls.AfSpeedEnum.Fast,
                             "AfMetering" : controls.AfMeteringEnum.Windows,
                             "AfWindows" : [ (0, 0, 400, 500) ]})
[... 126 lines cut ...]
[4:15:52.194916979] [7105]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:15:52.197631567] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[4:15:52.197673660] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[4:15:52.197679178] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[4:15:52.197685382] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[4:15:52.198412214] [7091]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[4:15:52.198543807] [7105]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
AF cycle succeeded in 21 frames
Image saved: /home/konos93/Desktop/camera/cam1/bbb20240208165852.jpg
[4:16:06.213042902] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[4:16:06.213084790] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[4:16:06.213092402] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[4:16:06.213101568] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[4:16:06.213780715] [7091]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-BGGR16_PISP_COMP1
[4:16:06.220726390] [7105]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected CFE format: 4608x2592-PC1B
[4:16:06.749302211] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[4:16:06.749343803] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[4:16:06.749349562] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[4:16:06.749356896] [7091]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[4:16:06.749907765] [7091]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[4:16:06.757067051] [7105]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
```

konos93, 2024-02-08:
should i use ?

                     "AfMetering" : controls.AfMeteringEnum.Windows,
                     "AfWindows" : [ (2104, 1046, 400, 500) ]})
                     
on the autofocus parts in the manual it is locked so yeah

konos93, 2024-02-15:
[(https://www.youtube.com/watch?v=0PodyJVMFxg)]

on 9.00 min what other calculation while the camera capture can be manual so click to capture jpeg be faster ? 
focus can be manual . lighting ? or something other ?

davidplowman, 2024-02-15:
Can you post some code that shows the exact problem? (Please don't post all your code, just post the smallest application that you can which demonstrates the problem, preferably with just one camera. We could also add some timing, so that we obtain useful numbers for measuring the capture delay.)

I'm guessing that it is switching mode and restarting the camera that is slow. This is hard to avoid unless you run the camera permanently in the full resolution mode - but then you will have slower a slower framerate in preview mode.

Maybe your UI also need some indication when the capture has finished?

konos93, 2024-02-15:
i think here is the reason 

def do_capture():

    timestamp = time.strftime("%Y%m%d%H%M%S")
    filename = f"{SAVE_DIRECTORY}/aaa{timestamp}.jpg"
    cfg = picam2.create_still_configuration(buffer_count=3)
    picam2.switch_mode_and_capture_file(cfg, filename, signal_function=qpicamera2.signal_done)
    print(f"Image saved: {filename}")
    
but still work fine 

manual focus is faster than auto focus , i think if i also regulate iso or others probably it will be faster .
`
  I'm guessing that it is switching mode and restarting the camera that is slow. This is hard to avoid unless you run the camera permanently in the full resolution mode - but then you will have slower a slower framerate in preview mode.`
  
yeah its still faster the preview with re-scale in the app window  rather than without it .

`Maybe your UI also need some indication when the capture has finished?`

the capture button change colors so i know that during this time i wont move the book

```
#!/usr/bin/python3

from libcamera import controls
from PyQt5 import QtCore
from PyQt5.QtGui import QPalette
from PyQt5.QtWidgets import (QApplication, QCheckBox, QHBoxLayout, QLabel,
                             QPushButton, QVBoxLayout, QWidget, QDoubleSpinBox)
from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2
import numpy as np
import time

STATE_AF = 0
STATE_CAPTURE = 1

def request_callback(request):
    lens_position = request.get_metadata().get('LensPosition', 'N/A')
    # Format the lens position to display up to 3 decimal places
    lens_position_str = f"{float(lens_position):.3f}" if lens_position != 'N/A' else 'N/A'
    label.setText(f"Lens Position: {lens_position_str}")
    

picam2 = Picamera2(0)
picam2.post_callback = request_callback
preview_width = 800# change me to check manual focus
preview_height = picam2.sensor_resolution[1] * preview_width // picam2.sensor_resolution[0]
preview_height -= preview_height % 2
preview_size = (preview_width, preview_height)
raw_size = tuple([v // 2 for v in picam2.camera_properties['PixelArraySize']])
preview_config = picam2.create_preview_configuration({"size": preview_size}, raw={"size": raw_size})
picam2.configure(preview_config)
if 'AfMode' not in picam2.camera_controls:
    print("Attached camera does not support autofocus")
    quit()
picam2.set_controls({"AfMode": controls.AfModeEnum.Auto,"AfSpeed": controls.AfSpeedEnum.Fast})
app = QApplication([])

def on_button_clicked():
    global state
    button.setEnabled(False)
    manual_focus.setEnabled(False)
    af_checkbox.setEnabled(False)
    state = STATE_AF if af_checkbox.isChecked() else STATE_CAPTURE
    if state == STATE_AF:
        picam2.autofocus_cycle(signal_function=qpicamera2.signal_done)
    else:
        do_capture()
[... 135 lines cut ...]
[0:22:30.694454306] [3021]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:22:30.785981425] [3021]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:22:30.789708405] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:22:30.789776128] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:22:30.789782961] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:22:30.789789979] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[0:22:30.790527387] [3008]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[0:22:30.790669887] [3021]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
Image saved: /home/konos93/Desktop/camera/cam0/aaa20240215131647.jpg
[0:22:36.954651476] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:22:36.954872124] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:22:36.954938661] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:22:36.954990976] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[0:22:36.956214143] [3008]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-BGGR16_PISP_COMP1
[0:22:36.964617456] [3021]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected CFE format: 4608x2592-PC1B
[0:22:37.477066525] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:22:37.477170007] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:22:37.477200025] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:22:37.477235748] [3008]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[0:22:37.477813322] [3008]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-BGGR16_PISP_COMP1
[0:22:37.484878876] [3021]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
```

davidplowman, 2024-02-15:
I'm sorry, I haven't really understood what you're asking.

At 9:00 in the video, you click the capture button. It seems reasonably quick to me. But is this actually too slow?

konos93, 2024-02-17:
> I'm sorry, I haven't really understood what you're asking.
> 
> At 9:00 in the video, you click the capture button. It seems reasonably quick to me. But is this actually too slow?

look i am the one who should apologize most times here i dont know even what exactly i want to ask. and when i ask many times i define what i am looking into. for me its the first time i read a lot so much a datasheet look examples and ask for help here

i run app_full.py i think its slightly faster . it is still good .

but if u have moved images 1 out 10 in 400 pages is about 40 images. of course that didnt happen i had to recapture about 10 out of 400 . 

in app_full.py u can regulate stuff like brightness and other . in capture_af which is the script based on the video if i could regulate stuff like ISO, aperture, shutter speed i think that capture button speed will be better because less calculations needed .

i will look out for it and i will inform. again thanks for your help

closed 2024-07-27 by konos93