raspberrypi/picamera2 #929: [HOW-TO] Remove qt 'framework' after closing QTGL /QT Previews

sandyol55, 2024-01-22:
I'd like to use matplot lib to analyse some data after timing preview frames (as per #885 frame sequential attempts).
If a QTGL preview has been open then I get:-
ImportError: Cannot load backend 'TkAgg' which requires the 'tk' interactive framework, as 'qt' is currently running

I note from comments in qt_previews.py that the intention is that the Qt app doesn't close.
Also that stopping or closing the picam process doesn't seem to 'release' Qt either?

Is this just a feature to live with or can qt be 'Quit' somehow when no longer required?

davidplowman, 2024-01-22:
Can you give the simplest example that provokes an error? When I try this:
```
import matplotlib.pyplot as plt
plt.plot(range(4))
plt.show()
```
it works, though there are complaints about QApplication::exec not being run in the main thread.

(There is a very long and unhappy story where Qt really really wants to take over the entire application and run as the main thread. If you try to run it as a background kind of window, as in the case of Picamera2 previews, it fights you tooth and nail, which is why there's so much awkwardness in the way it deals with Qt. Trying to work around all the ways that Qt attempts to stop you, basically.

Though as noted, if I start a matplotlib window, it "works", only you now have two QApplications. Probably, the whole thing is fundamentally borked at this point, and it will probably segfault just when you least expect it. If anyone can tell me how to drive these GUI frameworks so as to achieve all this, I'd love to know.

I think the "proper" way to use Picamera2 and Qt may be to force people actually to make a real Qt application. Then maybe you can put a Picamera2 window and a matplotlib window into the same all-controlling application. But I don't really know, so it's mostly a case of wishing anyone who tries this the best of luck! To be fair, if anyone did this, it would be a nice demo to add to the apps folder. I'm afraid I know even less about Tk.)

sandyol55, 2024-01-22:
This should be enough to trigger what I'm seeing:-
```from picamera2 import Picamera2,Preview
import matplotlib.pyplot as plt
import time

picam2 = Picamera2()

picam2.start_preview(Preview.QTGL)
time.sleep(2)

picam2.stop_preview()
picam2.stop()

plt.plot(range(4))
plt.show()
```

Was just curious whether I could run mathplotlib after a QT related preview.  Workaround has been to store the timestamps in a file and analyse outwith the test program.
Thanks

davidplowman, 2024-01-22:
Hmm, that's exactly what I tried, and it "worked" (subject to the various caveats about it probably being a total abuse of Qt). Does [this](https://matplotlib.org/stable/users/explain/figure/backends.html) shed some light?

sandyol55, 2024-01-22:
Many thanks for the link.
```
matplotlib.use('qtagg')
```
resolves the issue.

Got all the warnings but the plot was created.

No need to 'quit' qt,  just work with it ?

closed 2024-01-22 by sandyol55