raspberrypi/picamera2 #931: [HOW-TO]

JosephDemarest, 2024-01-23:
**Describe what it is that you want to accomplish**
Add an effect, for example a colourmap, over the preview output.

**Describe alternatives you've considered**
I spent a few hours trying to extend qglpicamera2 with some custom logic, but i hit a roadblock and figured you guys are significantly more knowledgeable on the library than i am. (obviously).

**Additional context**
the cameras will always be capturing in raw mode.

davidplowman, 2024-01-23:
Hi, thanks for the question. A few thoughts:

* If you're happy actually to _change_ the image itself, then you can use a `pre_callback`. For example, have a look what [this example](https://github.com/raspberrypi/picamera2/blob/main/examples/timestamped_video.py) does to alter the image by writing text on it.
* If you don't want to change the image itself, maybe you can add something over the top using an overlay. See [this example](https://github.com/raspberrypi/picamera2/blob/main/examples/overlay_gl.py).
* If you want to do something more sophisticated that can't be done with an overlay, then you probably need to edit or extend the preview windows. I'm sure it must be possible to add or change the graphics shaders to accomplish all kinds of different things - but it's not something I really know too much about. All they do at the moment is resample the image, with optional horizontal/vertical mirroring. Maybe there's someone else out there who can help?

JosephDemarest, 2024-01-23:
"If you're happy actually to change the image itself, then you can use a pre_callback. For example, have a look what [this example](https://github.com/raspberrypi/picamera2/blob/main/examples/timestamped_video.py) does to alter the image by writing text on it."

That is some great advice. Thank you. MY problem is completely solved now. I don't know how i missed pre_callback when post_callback is right there... lol.

I have one more question, if you don't mind. What settings/controls apply to the camera when taking a raw image? I want to get an output that isn't modified by anything, contrast, brightness, AEC/AWB/ etc. The only thing I want to set is resolution, and integration time.

JosephDemarest, 2024-01-23:
Thank you. That information is invaluable. 
Have a fantastic day