raspberrypi/picamera2 #933: [OTHER] Properly set up initial values for exposure time and gain in video configuration

ilirosmanaj, 2024-01-24:
I am trying to start my camera in video configuration with predefined values for most of the camera controls. From those, the most important ones for me are `ExposureTime` and `AnalogueGain`.

Unfortunately, even though I set this in the initial configuration, that seems to not be respected as the initial value and also is not respected after recording for long periods of time as well (approx ~20s).

To validate whether the settings are being applied correctly, I am checking:

- `picam2.controls` property
     - Some interesting note about this one: The moment I do `picam2.start_recording`, this property goes to an empty set value. Is this expected?
- `picam2.camera_controls` property
- `picam2.capture_metadata()` function

Please see a test script below:

```
from picamera2.encoders import H264Encoder
from picamera2 import Picamera2
import time

picam2 = Picamera2()

video_config = picam2.create_video_configuration(
    main={"size": (2304, 1296), "format": "RGB888"}, 
    lores={"size": (192, 108)},
    encode="lores", 
    buffer_count=4,
    queue=False,
    controls={
        "Brightness": 0,
        "Contrast": 2,
        "Saturation": 1,
        "AeConstraintMode": 0,
        "AeEnable": False,
        "ExposureTime": 1500,
        "AnalogueGain": 1.5,
        "AeExposureMode": 0,
        "ExposureValue": 0,
        "AeMeteringMode": 1,
        "AwbEnable": False,
        "AwbMode": 0,
        "ColourGains": (1, 1),
        "AfMetering": 0,
        "AfMode": 0,
        "AfRange": 1,
        "AfSpeed": 1,
        "LensPosition": 17,
        'FrameDurationLimits': (8333, 8333)
    }
)

picam2.configure(video_config)
encoder = H264Encoder(bitrate=10000000)

print(f"\n\n-Camera 'controls' before start recording: {picam2.controls}")
print(f"\n\n-Camera 'camera_controls' before start recording: {picam2.camera_controls}")
picam2.start_recording(encoder, "test.h264")

print(f"\n\n-Camera 'controls' after start recording: {picam2.controls}")
print(f"\n\n-Camera 'camera_controls' before start recording: {picam2.camera_controls}")
metadata = picam2.capture_metadata()
print(f"\n\n-Camera metadata after start recording: {metadata}")

time.sleep(20)

metadata = picam2.capture_metadata()
print(f"\n\n-Camera metadata after 10s of recording: {metadata}")

picam2.stop_recording()

print(f"\n\n-Camera 'controls' after stop recording: {picam2.controls}")
print(f"\n\n-Camera 'camera_controls' after stop recording: {picam2.camera_controls}")
```

This outputs:

```
[95:40:28.268171184] [966927]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[95:40:28.561431925] [966956]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media1
[95:40:28.561669999] [966956]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[95:40:28.607918962] [966956]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@0/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media3
[95:40:28.608362369] [966956]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[95:40:28.639095981] [966927]  INFO Camera camera.cpp:1033 configuring streams: (0) 2304x1296-RGB888 (1) 192x108-YUV420 (2) 2304x1296-SBGGR10_CSI2P
[95:40:28.641100758] [966956]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA

-Camera 'controls' before start recording: <Controls: {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (8333, 8333), 'Brightness': 0, 'Contrast': 2, 'Saturation': 1, 'AeConstraintMode': 0, 'AeEnable': False, 'ExposureTime': 1500, 'AnalogueGain': 1.5, 'AeExposureMode': 0, 'ExposureValue': 0, 'AeMeteringMode': 1, 'AwbEnable': False, 'AwbMode': 0, 'ColourGains': (1, 1), 'AfMetering': 0, 'AfMode': 0, 'AfRange': 1, 'AfSpeed': 1, 'LensPosition': 17}>

-Camera 'camera_controls' before start recording: {'ColourGains': (0.0, 32.0, None), 'AfMetering': (0, 1, 0), 'AnalogueGain': (1.1228070259094238, 16.0, None), 'Saturation': (0.0, 32.0, 1.0), 'Contrast': (0.0, 32.0, 1.0), 'AeMeteringMode': (0, 3, 0), 'AfMode': (0, 2, 0), 'AeConstraintMode': (0, 3, 0), 'AeEnable': (False, True, None), 'ExposureTime': (13, 0, None), 'AfPause': (0, 2, 0), 'AfRange': (0, 2, 0), 'NoiseReductionMode': (0, 4, 0), 'ScalerCrop': ((0, 0, 128, 128), (0, 0, 4608, 2592), (0, 0, 4608, 2592)), 'Sharpness': (0.0, 16.0, 1.0), 'AwbEnable': (False, True, None), 'FrameDurationLimits': (17849, 112075593, None), 'ExposureValue': (-8.0, 8.0, 0.0), 'AwbMode': (0, 7, 0), 'AeExposureMode': (0, 3, 0), 'AfSpeed': (0, 1, 0), 'AfTrigger': (0, 1, 0), 'AfWindows': ((0, 0, 0, 0), (65535, 65535, 65535, 65535), (0, 0, 0, 0)), 'Brightness': (-1.0, 1.0, 0.0), 'LensPosition': (0.0, 32.0, 1.0)}

-Camera 'controls' after start recording: <Controls: {}>

-Camera 'camera_controls' before start recording: {'ColourGains': (0.0, 32.0, None), 'AfMetering': (0, 1, 0), 'AnalogueGain': (1.1228070259094238, 16.0, None), 'Saturation': (0.0, 32.0, 1.0), 'Contrast': (0.0, 32.0, 1.0), 'AeMeteringMode': (0, 3, 0), 'AfMode': (0, 2, 0), 'AeConstraintMode': (0, 3, 0), 'AeEnable': (False, True, None), 'ExposureTime': (13, 0, None), 'AfPause': (0, 2, 0), 'AfRange': (0, 2, 0), 'NoiseReductionMode': (0, 4, 0), 'ScalerCrop': ((0, 0, 128, 128), (0, 0, 4608, 2592), (0, 0, 4608, 2592)), 'Sharpness': (0.0, 16.0, 1.0), 'AwbEnable': (False, True, None), 'FrameDurationLimits': (17849, 112075593, None), 'ExposureValue': (-8.0, 8.0, 0.0), 'AwbMode': (0, 7, 0), 'AeExposureMode': (0, 3, 0), 'AfSpeed': (0, 1, 0), 'AfTrigger': (0, 1, 0), 'AfWindows': ((0, 0, 0, 0), (65535, 65535, 65535, 65535), (0, 0, 0, 0)), 'Brightness': (-1.0, 1.0, 0.0), 'LensPosition': (0.0, 32.0, 1.0)}

-Camera metadata after start recording: {'SensorTemperature': 40.0, 'AfPauseState': 0, 'ExposureTime': 17208, 'DigitalGain': 4.0, 'AfState': 0, 'SensorTimestamp': 344429065338000, 'LensPosition': 5.0, 'ScalerCrop': (0, 0, 4608, 2592), 'ColourCorrectionMatrix': (1.557413935661316, -0.27196478843688965, -0.28544917702674866, -0.42025768756866455, 1.7205229997634888, -0.3002624809741974, -0.03535439446568489, -0.5996246337890625, 1.634979009628296), 'ColourTemperature': 4000, 'AnalogueGain': 1.4992679357528687, 'ColourGains': (1.0, 1.0), 'AeLocked': False, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'Lux': 400.0, 'FrameDuration': 17849}

-Camera metadata after 10s of recording: {'SensorTimestamp': 344449091325000, 'ScalerCrop': (0, 0, 4608, 2592), 'ColourCorrectionMatrix': (1.557413935661316, -0.27196478843688965, -0.28544917702674866, -0.42025768756866455, 1.7205229997634888, -0.3002624809741974, -0.03535439446568489, -0.5996246337890625, 1.634979009628296), 'FocusFoM': 459, 'ExposureTime': 17208, 'SensorTemperature': 60.0, 'AfPauseState': 0, 'LensPosition': 17.0, 'FrameDuration': 17849, 'AeLocked': True, 'AfState': 0, 'DigitalGain': 1.0206844806671143, 'AnalogueGain': 1.4992679357528687, 'ColourGains': (1.0, 1.0), 'ColourTemperature': 4000, 'Lux': 25.9088191986084, 'SensorBlackLevels': (4096, 4096, 4096, 4096)}

-Camera 'controls' after stop recording: <Controls: {}>

-Camera 'camera_controls' after stop recording: {'ColourGains': (0.0, 32.0, None), 'AfMetering': (0, 1, 0), 'AnalogueGain': (1.1228070259094238, 16.0, None), 'Saturation': (0.0, 32.0, 1.0), 'Contrast': (0.0, 32.0, 1.0), 'AeMeteringMode': (0, 3, 0), 'AfMode': (0, 2, 0), 'AeConstraintMode': (0, 3, 0), 'AeEnable': (False, True, None), 'ExposureTime': (13, 0, None), 'AfPause': (0, 2, 0), 'AfRange': (0, 2, 0), 'NoiseReductionMode': (0, 4, 0), 'ScalerCrop': ((0, 0, 128, 128), (0, 0, 4608, 2592), (0, 0, 4608, 2592)), 'Sharpness': (0.0, 16.0, 1.0), 'AwbEnable': (False, True, None), 'FrameDurationLimits': (17849, 112075593, None), 'ExposureValue': (-8.0, 8.0, 0.0), 'AwbMode': (0, 7, 0), 'AeExposureMode': (0, 3, 0), 'AfSpeed': (0, 1, 0), 'AfTrigger': (0, 1, 0), 'AfWindows': ((0, 0, 0, 0), (65535, 65535, 65535, 65535), (0, 0, 0, 0)), 'Brightness': (-1.0, 1.0, 0.0), 'LensPosition': (0.0, 32.0, 1.0)}
```

Is there something obvious that I am missing? Again, my goal is to just start the camera on a specific ExposureTime and AnalogueGain value.

davidplowman, 2024-01-24:
Hi, thanks for raising this.

Indeed there may be some slightly awkward sequencing going on down in libcamera. When you set a number of controls, it's not defined what order they happen in, so it's quite easy for things to get confusing. In your case, it's setting `AeEnable` to `False` that causes libcamera to turn off the AEC/AGC and adopt the "current" values. But most controls don't apply immediately, so your values (1500 exposure and 1.5 gain) haven't appeared yet.

The easiest workaround might be not to set `AeEnable` to `False`, because setting the exposure and gain explicitly does this anyway. Ditto AwbEnable. Setting the colour gains disables it, so may as well not do it again (in case there are similar things happening, not sure).

I think the code could do better here, but that's something that needs looking into down in libcamera itself. But give that a try as a workaround and see if things are more predictable.

To your other point, libcamera (and therefore Picamera2) doesn't maintain a list of "current control values". That's in part because setting a control value does not mean it has happened, or at least, not immediately. So you have to wait and watch the metadata to see what's actually going on. `camera_controls` is just a list of the available controls (and limits for them), and `controls` is merely where you set values that you want to send (and is cleared once they're sent). But again, this only means "sent" and not "happened", usually there several frames delay before that - watch the metadata.

Hope that helps.

ilirosmanaj, 2024-01-24:
Hi @davidplowman,

Thanks a lot for your response, all makes sense. 

So my scenario is this way:

- I want to enable Autoexposure
- But want to set the default value of exposure time/analogue gain to some specific value so that it starts closer to the range where I expect it to be

Is there a way to do so?

davidplowman, 2024-01-24:
I believe it normally starts out at an exposure of 1000us and a gain of 1x. You can actually change these values in the tuning file (imx708.json, or one of the wide/noir variants for the v3 camera). Find "channels:" in the tuning file, and just inside the next curly brace put something like
```
"default_exposure_time": 5000,
"default_analogue_gain": 1.5,
```
Make sure you're editing the tuning file that you're using - I normally type a garbage character at the top of the file and check the camera _doesn't_ start to be sure, it's so easy to get this wrong otherwise.

I 've never actually tried this, but the settings appear to be available!

ilirosmanaj, 2024-01-25:
Thanks for this, @davidplowman - can you point me where you see these settings?

The tunning file on my pi does not have any `channels` key anywhere, and I could not find documentation for it in the manual either (https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

davidplowman, 2024-01-25:
Ah, you're running on a Pi with older software. Find where the tuning file has "metering_modes" and put these fields just above that, in the same curly braces.

This is all libcamera stuff, so the Picamera2 manual wouldn't explicitly talk about it. The libcamera alogithm and tuning guide should, though it's quite possible they haven't been documented there either.

ilirosmanaj, 2024-01-26:
Hi @davidplowman - thanks a lot for the note.

Can you please help me understand what do you mean with "older software"? Not sure I got that

davidplowman, 2024-01-29:
Newer versions of the software have this "channels" keyword in the JSON tuning file. Older versions don't. Currently, however, the Imager tool only recommends "older" Bullseye images for Pis older than a Pi 4, though this is being worked on so that Bookworm images can be recommended there too.

ilirosmanaj, 2024-01-29:
Thank you @davidplowman. I might need some more support. Could you please provide me with an email of yours, so I can explain it there?

davidplowman, 2024-01-29:
Generally I prefer to explain stuff in public forums like GitHub, just in case anyone else might learn from or wants to follow the discussion. Would it work for you to continue filing questions here as the need arises?

ilirosmanaj, 2024-01-29:
Sure, no issues. We have a product based quite a lot in Picamera2, and our team has plenty of questions that come continuously and are quite blockers for us. The idea of the email was to evaluate if you could do some consultancy with us. Let me know if this works for you - but in general, it's really helpful so far; thank you very, very much for the above.

davidplowman, 2024-01-29:
Hi again, happy to keep answering questions here. If you wanted any kind of more formal arrangement, you could speak with our applications team who provide more dedicated support, @JamesH65 should be able to tell you more.

JamesH65, 2024-01-29:
Hi, please email applications@raspberrypi.com for industrial technical support.