raspberrypi/picamera2 #934: [BUG] stdin of another process cannot be used as Outpt to `start_recording`

puterboy, 2024-01-26:
I am trying to pass the stdin of another process as the `output` of `start_recording(encoder, output)`
However I get errors of form:
``` 
File "/home/mysuer/test.py", line 273, in video_pipeline
    pi_camera.start_recording(encoder, my_proc.stdin, quality=Quality.HIGH) 
   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1743, in start_recording
    self.start_encoder(encoder, output, pts=pts, quality=quality, name=name)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1664, in start_encoder
    _encoder.output = output
    ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 177, in output
    raise RuntimeError("Must pass Output")
RuntimeError: Must pass Output
```
where `proc.stdin` is generated by a command of form:
```
FNULL = open(os.devnull, 'w')
my_proc = psutil.Popen(my_cmd.split(' '), stdin=subprocess.PIPE, stdout=FNULL, stderr=subprocess.PIPE)
```

Indeed, `isinstance(my_proc.stdin, Output)` returns `false`
while `print(my_proc.stdin)` returns `<_io.BufferedWriter name=5>`

In `picamera`, I was able to use `my_proc.stdin` as an Output to start_recording.
Why can't the stdin of another process be used as the output of `start_recording` in `picamera2`

Is this a bug or am I doing something wrong here?

davidplowman, 2024-01-29:
Hi, good question. Have you tried
```
output = FileOutput(my_proc.stdin)
picam2.start_recording(encoder, output)
```
Let us know if that doesn't work. Thanks!

puterboy, 2024-01-29:
That works! Thanks!

closed 2024-01-29 by puterboy