raspberrypi/picamera2 #939: [BUG] ImportError: cannot import name 'Orientation' from 'libcamera' (/usr/lib/python3/dist-packages/libcamera/__init__.py)

juanmf, 2024-01-30:
**Describe the bug**
After upgrading to picamera2-0.3.16 this line causes issues [File "/usr/local/lib/python3.9/dist-packages/picamera2/utils.py", line 1,](https://github.com/raspberrypi/picamera2/commit/1ab79ac3ff945252a2c4d94818809c4ebb884117#diff-fd46c318073166c21c0aa2be3834e1d2c74d401d5db752de3ed53f3faf6eb9f0R1)

Added a [comment](https://github.com/raspberrypi/picamera2/commit/1ab79ac3ff945252a2c4d94818809c4ebb884117#r138050203) in conflicting commit with more details

**To Reproduce**
upgrade libcamera & picamera2:
```
...
Successfully installed picamera2-0.3.16
$libcamera-hello --version
libcamera-apps build: 7e4d3d71867f 17-07-2023 (07:34:42)
libcamera build: v0.0.5+83-bde9b04f

```

**Expected behaviour**
normal import behavior

**Console Output, Screenshots**
```
Traceback (most recent call last):
...
    import picamera2 as Picam2
  File "/usr/local/lib/python3.9/dist-packages/picamera2/__init__.py", line 9, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 24, in <module>
    import picamera2.utils as utils
  File "/usr/local/lib/python3.9/dist-packages/picamera2/utils.py", line 1, in <module>
    from libcamera import ColorSpace, Orientation, Rectangle, Size, Transform
ImportError: cannot import name 'Orientation' from 'libcamera' (/usr/lib/python3/dist-packages/libcamera/__init__.py)
```

**Hardware :**
Raspberri Pi 4B

uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

(same happened on Linux 5.13.32)

davidplowman, 2024-01-30:
Can you say where you have got your version of libcamera from?

It sounds to me like you're running something horribly old, and any even vaguely recent version of Picamera2 will never work.

If you have a spare SD card, are you able simply to install an up-to-date image to test your camera? Can you say why this doesn't work for you? Thanks.

dgalland, 2024-01-30:
@davidplowman 
What is not obvious is the compatibility of the libcamera/picamera2 versions.
For example, currently for me it is OK with the current distributions 3.14 and up to date libcamera distrib.
Should I expect a similar problem if I wanted to try 2.16?

davidplowman, 2024-01-30:
We always recommend updating libcamera, rpicam-apps and Picamera2 from apt together, because we always ensure they work together. The root of the problem is that libcamera doesn't have a stable API, though I think things are getting better with some of the version numbering, so that's something we could try to improve.

I think we'd make Picamera2 have a minimum supported libcamera version. We're unlikely to keep newer versions of Picamera2 backwards compatible with older libcameras unless it's straightforward to do and doesn't complicate code paths internally.

juanmf, 2024-01-30:
okay, I could start from scratch (before I was tied to kernel 5.13.32) due to the [camera driver issue](https://github.com/INNO-MAKER/CAM-OV9281RAW-V2/issues/8).

Initially it didn't detect the camera again but some of [these steps in issue 218](https://github.com/raspberrypi/picamera2/issues/218#issuecomment-1181352074) made the trick.

setup for reference, after clean image of :

`Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux`

installed;
```commandline
sudo apt-get update
sudo apt-get upgrade
sudo apt-get autoremove

sudo apt install -y python3-libcamera python3-kms++
sudo apt install -y python3-pyqt5 python3-prctl libatlas-base-dev ffmpeg python3-pip 
sudo apt install -y python3-opencv
sudo apt install -y opencv-data vim

> https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#setup-virtual-environment-3157129

sudo apt-get install -y i2c-tools libgpiod-dev python3-libgpiod
sudo apt install --upgrade python3-setuptools
sudo apt install python3.11-venv

$ sudo vim /boot/firmware/config.txt
Add at the end: 
[all]
camera_auto_detect=0
dtoverlay=ov9281
dtparam=i2c_vc=on
```

Ready to start debugging my own app. Thanks!
just a note from your comment on https://github.com/raspberrypi/picamera2/issues/849#issuecomment-1914261726

```
request = picam2.capture_request(flush=True)
array = request.make_array('main')
request.flush()
```
> AttributeError: 'CompletedRequest' object has no attribute 'flush'

I'm closing this issue. Thanks for your fast response.

closed 2024-01-30 by juanmf

Chetan2312, 2024-06-07:
I tried all above changes mentioned in the last comment by @juanmf , ye i am facing the same issue.
`File "/usr/local/lib/python3.9/dist-packages/vilib-0.2.1-py3.9.egg/vilib/__init__.py", line 2, in <module>
    from .vilib import Vilib
  File "/usr/local/lib/python3.9/dist-packages/vilib-0.2.1-py3.9.egg/vilib/vilib.py", line 25, in <module>
    from picamera2 import Picamera2
  File "/usr/local/lib/python3.9/dist-packages/picamera2/__init__.py", line 9, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 24, in <module>
    import picamera2.utils as utils
  File "/usr/local/lib/python3.9/dist-packages/picamera2/utils.py", line 1, in <module>
    from libcamera import ColorSpace, Orientation, Rectangle, Size, Transform
ImportError: cannot import name 'Orientation' from 'libcamera' (/usr/lib/python3/dist-packages/libcamera/__init__.py)`

davidplowman, 2024-06-07:
It sounds as though you may be running an older version of Pi OS and have tried to install a version of Picamera2 that is newer? I would recommend:

* Upgrade to Pi OS Bookworm if you can.
* If you need to use an older version of Pi OS, only use the version of Picamera2 available in apt, as this will be the correct version for the OS.

Chetan2312, 2024-06-07:
@davidplowman , I am using Bullseye OS.

```
 $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

I will try with Bookworm OS once and Update. Thanks for the quick response.

zoldaten, 2025-02-27:
@Chetan2312 
```
pip3 uninstall picamera2
sudo apt install -y python3-libcamera
pip3 install numpy==1.26.4
```