raspberrypi/picamera2 #942: Can't use picamera2 in a Tkinter GUI

Carl0sC0elh0, 2024-02-01:
Hello everyone, I'm developing a Tkinter app to help me with some image processing. My objective is to take a photo from the picamera v3 module and store it for further analysis. the problem is that the code doens't run and aborts the app with this output

`QObject::moveToThread: Current thread (0x7f580286a0) is not the object's thread (0x7f581f82b0).
Cannot move to target thread (0x7f580286a0)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/Carlos/FISCHER/lib/python3.11/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.`

I'm trying to use threading as well so that the tkinter app keeps running while the camera takes the shot, but even without threading the app closes when it reaches the segment to take a shot. Any advice on this? Is it a dependencies problem? Thanks in advance

tvoverbeek, 2024-02-01:
Tkinter is not supported on Wayland (default on Pi3b+, Pi4 and Pi5).
You need to be running X11.
On Bookworm run raspi_config, goto Advanced Settings, Select Wayland and select X11 and Openbox.
I do not know if this relevent to your issue.

bnd762, 2024-02-05:
I solved a similar problem using Pygame. This actually works quite well.
https://github.com/raspberrypi/picamera2/issues/607#issuecomment-1730933656

Carl0sC0elh0, 2024-02-05:
Thank you all for your suggestions but it was indeed a dependencies conflict or something like that. My venv didn't have picamera2 and libcamera2 installed and couldn't install them via pip.   I solved it by creating a new venv with --system-site-packages. Is it a good fix? I don't know, im just a noob with a PI4. 

Thank you all and have a good week!

davidplowman, 2024-02-05:
Yes, the easiest solution for a venv (unless you have some very particular requirements which mean it isn't appropriate) is to use --system-site-packages.