raspberrypi/picamera2 #944: [HOW-TO] Run AutoFocus only on a specific range of lenses

ilirosmanaj, 2024-02-05:
My end goal is simple, but to do so, I need some understanding of Picamera2 and libcamera as well. 

Let me explain my problem in a couple of bullet points: 

- I want to use camera autofocus, which runs only when I trigger it
- I want to use the Fast autofocus speed
- I want to use a Macro autofocus range, limited in lens position from 13 to 18. Defaulting to 13
- When the autofocus is triggered, I want to use PDAF in a way that it acts as quickly as possible
    - Potentially, I would want to allow PDAF to do only one range of scans (as in starting from lens 13 and going up, but then stopping as quickly as it finds some suitable capture)
    - I would be cool with PDAF only testing rounded lens positions (e.g., only values 13, 14, 15, 16, 17, or 18). Currently, this spends plenty of time in fractional values (e.g. 13.55555)

Note: what I am trying to capture is my hand, which is in front of the camera (about 10-15 cm from the camera itself).

Here is my camera tunning file (adding only the autofocus part here):

```
{
            "rpi.af":
            {
                "ranges":
                {
                    "normal":
                    {
                        "min": 1.0,
                        "max": 32.0,
                        "default": 1.0
                    },
                    "macro":
                    {
                        "min": 13.0,
                        "max": 18.0,
                        "default": 13.0
                    }
                },
                "speeds":
                {
                    "normal":
                    {
                        "step_coarse": 2.0,
                        "step_fine": 0.5,
                        "contrast_ratio": 0.75,
                        "pdaf_gain": -0.03,
                        "pdaf_squelch": 0.2,
                        "max_slew": 4.0,
                        "pdaf_frames": 20,
                        "dropout_frames": 6,
                        "step_frames": 4
                    },
                    "fast":
                    {
                        "step_coarse": 1.0,
                        "step_fine": 1.0,
                        "contrast_ratio": 0.75,
                        "pdaf_gain": -0.05,
                         # modified from 0.2, which is the default
                        "pdaf_squelch": 0.4,
                        "max_slew": 5.0,
                         # modified from 16, which is the default
                        "pdaf_frames": 10,
                        "dropout_frames": 6,
                        "step_frames": 2
                    }
                },
                "conf_epsilon": 8,
                "conf_thresh": 12,
                "conf_clip": 512,
                "skip_frames": 5,
                "map": [ 0.0, 420, 35.0, 920 ]
            }
        }
[... 273 lines cut ...]
2024-02-05 10:28:37,796 - root - DEBUG: Processed frame: 61
2024-02-05 10:28:37,800 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.0
2024-02-05 10:28:37,815 - root - DEBUG: Processed frame: 62
2024-02-05 10:28:37,818 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.0
2024-02-05 10:28:37,836 - root - DEBUG: Processed frame: 63
2024-02-05 10:28:37,839 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.0
2024-02-05 10:28:37,859 - root - DEBUG: Processed frame: 64
2024-02-05 10:28:37,863 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,880 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,881 - root - DEBUG: Processed frame: 65
2024-02-05 10:28:37,899 - root - DEBUG: Processed frame: 66
2024-02-05 10:28:37,902 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,918 - root - DEBUG: Processed frame: 67
2024-02-05 10:28:37,921 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,926 - root - DEBUG: Processed frame: 68
2024-02-05 10:28:37,945 - root - DEBUG: AF Success on lens position: 13.492711067199707
2024-02-05 10:28:37,963 - root - DEBUG: Processed frame: 69
2024-02-05 10:28:37,966 - root - DEBUG: Finished
```

So this took 2s already, which is way too much for what I want. I would need it at a maximum of 200ms speed.

To wrap up, I need to run the AF on lenses [13, 14, 15, 16, 17, 19] as quickly as possible in a trustable way.

I do expect this to be a combination of picamera and libcamera as well, so happy to perform tunning as much as needed. I failed to find some good explanation of the PDAF algorithm and their parameters so that I would know how to change.

ilirosmanaj, 2024-02-05:
@davidplowman a very interesting problem I have been checking recently. Hope you can help!

davidplowman, 2024-02-06:
Here are some suggestions:

* If your search is always starting at at LP = 13, maybe set the lens position to 13 in your configuration (add `'LensPosition': 13` to the controls for creating the configuration).
* You may actually have more luck running at 30fps than 80fps. If you run faster than 30 then algorithms like AF will start skipping frames.
* It looks to me from the log like it's currently taking about 1.5s. Can you reduce the range any more? I suspect that doing an AF search over a close range like this is always going to be somewhat slow because the image is so sensitive to lens movement. But if it's still taking too long, @njhollinghurst may have some suggestions on other tuning changes.

ilirosmanaj, 2024-02-06:
Thanks for the response, @davidplowman @njhollinghurst.

- Have set up the frame rate to 30pfs
- I am not sure it's respecting the lens position; even when I set it from the config, it always starts at 1. Does it need some time to pick up? Or the lens position isn't respected if you set the AF mode to auto and manual.
-  I've tried lowering the confidence threshold and epsilon but it is not stable, and the produced images are also not very well-focused

I am unsure what the next reasonable step would be in this case. Is there a way to completely disable CDAF?

closed 2025-01-02 by ilirosmanaj