raspberrypi/picamera2 #945: [HOW-TO] Use text color via `cv2.putText()` in `lores` stream

pwhack, 2024-02-06:
Hi!

I'm applying a timestamp to both the `main` and `lores` streams via a `pre_callback` function, much like [this example](https://github.com/raspberrypi/picamera2/blob/main/examples/timestamped_video.py). The problem is that no combination of values in the text color tuple fed to `cv2.putText()` produces color when operating on the `lores` stream. I can only vary the brightness of a grayscale via the first element of the tuple. Is there any way to tell `cv2.putText()` how to apply colorful text on that stream? Everything is working fine (with colors) in the `main` stream.

I'm guessing the issue is due to the `lores` stream being in YUV420 format on my RPi4. Is that correct? Do I need to perform some sort of conversion of the `lores` stream before attempting to use `cv2.putText()`? Will that slow down the `pre_callback` function too much? Though the text is in grayscale, the final output is otherwise fine, the image is not visibly skewed to some other colorspace.

I've tried using an RGB-to-YUV calculator (even though OpenCV likes to operate in BGR) to no avail. Then I wrote a loop to increment each of the text color tuple elements one at a time to brute force something on screen. It seems only the first element does anything, where 0 appears black and 255 appears white with grayscale in between. The second and third elements seem to do nothing.

Below is a quick minimal script to demonstrate the behavior on the `lores` stream:

```
#!/usr/bin/env python3
import time
import cv2

from picamera2 import MappedArray, Picamera2
from picamera2.encoders import H264Encoder

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(lores={'size': (640, 480)}))

color = (255, 0, 0) # White
origin = (0, 30)
font = cv2.FONT_HERSHEY_SIMPLEX
scale = 1
thickness = 2

def apply_timestamp(request):
    timestamp = time.strftime('%Y-%m-%d %X')
    with MappedArray(request, 'lores') as m:
        cv2.putText(m.array, timestamp, origin, font, scale, color, thickness)

picam2.pre_callback = apply_timestamp

encoder = H264Encoder(10000000)

picam2.start_recording(encoder, 'test.h264', name='lores')
time.sleep(5)
picam2.stop_recording()
```

And a screen grab of what `color = (255, 0, 0)` produces:

[image]

Thanks for any help you can provide!

sandyol55, 2024-02-07:
If we can bend the requirements a bit, this will offer a coloured band where the text is, ie both the added text and the background video are tinted.
```
#!/usr/bin/env python3
import time
import cv2

from picamera2 import MappedArray, Picamera2
from picamera2.encoders import H264Encoder

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(lores={'size': (640, 480)}))

y,u,v=200,140,140
color = (y, u, v)
origin = (0, 30)
font = cv2.FONT_HERSHEY_SIMPLEX
scale = 1
thickness = 2

def apply_timestamp(request):
    timestamp = time.strftime('%Y-%m-%d %X')
    with MappedArray(request, 'lores') as m:
        cv2.putText(m.array, timestamp, origin, font, scale, color, thickness)
        m.array[480:490,:]=u
        m.array[600:610,:]=v

picam2.pre_callback = apply_timestamp

encoder = H264Encoder(10000000)

picam2.start_recording(encoder, 'test.h264', name='lores')
time.sleep(5)
picam2.stop_recording()
```

pwhack, 2024-02-12:
Well these are both wonderful examples I can use. Thank you so much!