raspberrypi/picamera2 #949: [OTHER] ImportError after installing v0.3.16

ajkelsey, 2024-02-08:
I upgraded to the lastest version of picamera2 using pip. When I try to import the module I receive the following error:
```
>>> from picamera2 import Picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/speedcam/.local/lib/python3.9/site-packages/picamera2/__init__.py", line 9, in <module>
    from .picamera2 import Picamera2, Preview
  File "/home/speedcam/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 24, in <module>
    import picamera2.utils as utils
  File "/home/speedcam/.local/lib/python3.9/site-packages/picamera2/utils.py", line 1, in <module>
    from libcamera import ColorSpace, Orientation, Rectangle, Size, Transform
ImportError: cannot import name 'Orientation' from 'libcamera' (/usr/lib/python3/dist-packages/libcamera/__init__.py)

```

davidplowman, 2024-02-09:
Hi, it sounds like you might be running Bullseye? You would need to install Bookworm to run 0.3.16, or otherwise downgrade back to 0.3.12 (in fact, I expect apt wouldn't offer you anything later).

Using Pi 3s (or earlier), the imager tool won't offer you Bookworm as an option, though you can still choose it if you don't specify the device. I believe it should work OK on most devices. There is work in progress to have an "official" Bookworm for these models of Pi, but I don't know exactly when it will be ready.

As an interim, there's also a 0_3_12 branch in this repository where I sometimes put "hot fixes", exactly for folks still on Bullseye. Sorry that the situation here remains a bit complicated for now than it should be.

ajkelsey, 2024-02-09:
Yes, it appears I am running Bullseye. I have a pi4, so I will make the upgrade to Bookworm. Thank you for pointing that out.

ajkelsey, 2024-02-09:
So upgraded to Bookworm. Python 3.11.2 is installed. Picamera2 0.3.16 is installed. I import picamera and try to create an object and receive an error.

closed 2024-02-09 by ajkelsey