raspberrypi/picamera2 #959: [BUG] MJPEGEncoder - TypeError: memoryview: casts are restricted to C-contiguous views

signag, 2024-02-20:
Hello,

**Bug Description**

I am working on a web server for controlling Raspberry Pi cameras with picamera2 ([raspi-cam-srv](https://github.com/signag/raspi-cam-srv))
This includes an MJPEG live stream.

Working with camera module 2, I try to use several stream sizes which are available in the 7 sensor modes.
The following sizes have been tested:
(640, 480) - OK    
(1640, 1232) - NOK    
(1920, 1080) - OK    
(3280, 2464) - NOK    
For stream sizes marked above with NOK, I get the following exception:
```
Exception during process_requests()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 85, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1265, in process_requests
    encoder.encode(encoder.name, req)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 210, in encode
    self._encode(stream, request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/libav_mjpeg_encoder.py", line 81, in _encode
    frame = av.VideoFrame.from_ndarray(m.array, format=self._av_input_format, width=self.width)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "av/video/frame.pyx", line 463, in av.video.frame.VideoFrame.from_ndarray
  File "av/video/frame.pyx", line 44, in av.video.frame.copy_array_to_plane
TypeError: memoryview: casts are restricted to C-contiguous views
```

**To Reproduce**

To reproduce the issue, I have attached the following test program which is an excerpt of the streaming server of raspiCamSrv.
[testPicamera2MJPEGEncoder.py.txt](https://github.com/raspberrypi/picamera2/files/14343894/testPicamera2MJPEGEncoder.py.txt)

**Expected behaviour**

The MJPEGEncoder should support all screen sizes of the available sensor modes.

**Console Output, Screenshots**

This is the output obtained when running the test program:   
[testOutput.txt](https://github.com/raspberrypi/picamera2/files/14343893/testOutput.txt)

**Hardware**

RPi 5   
Attached cameras: 
- (0): module 2 camera ("Model": "imx219")   
- (1): module 3 camera ("Model": "imx708")   

OS: Bookworm   
Debian version: 12.5

**Additional Context**

The above error does not occur when using the different screen sizes with camera module 3.

davidplowman, 2024-02-20:
Hi, thanks for the report.

Yes, there do seem to be some problems passing images to libav when there is padding on the end of each row of pixels (as hardware often requires for alignment purposes). Specifically it's the PyAV wrapper that doesn't like non-contiguous arrays. Doing yet another copy to make them contiguous seems quite painful, so it would probably be better to patch PyAV to cope - I'll have a look.

davidplowman, 2024-02-20:
OK, I think the top commit here https://github.com/davidplowman/PyAV should do it. Obviously that'll have to get built into a package and distributed to our apt repo.

signag, 2024-02-21:
Hi,
I compiled your latest PyAV from [https://github.com/davidplowman/PyAV](https://github.com/davidplowman/PyAV) and deployed av to my virtual environment.

Now all stream sizes, listed in the sensor modes, can be successfully configured.
I tested also some extreme ones like (3280, 52) which work too.

Thank you very much

davidplowman, 2024-02-21:
Thanks for the confirmation. I'll get the version in our apt repository updated.

davidplowman, 2024-02-21:
I think the apt repo for Bookworm has also been updated now if that helps.