raspberrypi/picamera2 #970: [HOW-TO]Run capture_circular_stream.py

micksulley, 2024-03-03:
I am trying to use the capture_circular_stream.py from examples, to record video and to stream output to my desktop.
It appears to run and capture video but on my desktop I see a series of strange characters like this
���gBÀÚ@è@���@¡ Æ¨���hÎ<€��eˆ„t‚˜ �!¡oÿøà�B8�ˆÇVð
My desktop is running Firefox in Linux Mint.  
If I run the mjpeg_server.py example it works fine.

Thanks
Mick

davidplowman, 2024-03-04:
Could you please include a short script that demonstrates what you're trying to do and shows the problem? If possible, please keep the script down to no more that a dozen or so line. Thanks.

micksulley, 2024-03-04:
I've just discovered a problem with my SD card.  I don't know if it's related but I will create a new one and try again.  I'll get back to you.

micksulley, 2024-03-04:
OK new SD card which passes the test.
I am running "capture_circular_stream.py" from the examples on GITHUB.  The terminal output looks OK  and it records h264 clips when movement is detected, but the video stream is not working, I see a long long string of characters e.g. ���gBÀÚ@è@���@¡ Æ¨���hÎ<€��eˆ„H)ƒ �"iÿéöÇ1Àaä~•?ð†×ÛÿÂ¿ÿnÇWþß÷„1Û¢§§Þ6È'€ƒý—eêÎ­¿þð¾U–Ãå„vá¥÷êÎ$Ëÿ!ª‰þËÿÂ{ß¯ÿøŒ.ô!´ßÿÞÚú/K¥èŒì

My desktop browser (Firefox) is pointed to http://pi-5-wi:10001/index.html

If I run "mjpeg_server.py", again from the examples, it works fine with browser pointed to http://pi-5-wi:8000/index.html
Just looking at the code it seems that "mjpeg_server.py" uses port 8000 and "capture_circular_stream.py" uses port 10001.  Is that correct?
Thanks
Mick

davidplowman, 2024-03-04:
You're right with the port numbers. But bear in mind that the circular example is just streaming a vanilla h.264 stream to a web socket, and browsers don't know how to interpret this. I would recommend a tool like ffplay, try pointing that at the address instead. (Even vlc won't play a basic h.264 stream.)

micksulley, 2024-03-04:
Thanks for your help David, I now have it working.  In case anyone else needs to know how -
My pi is "pi-5-wi" running "capture_circular_stream.py" from the examples on GITHUB
On my desktop (Linux Mint) terminal I run 
ffplay -i tcp://pi-5-wi:10001
and it shows me the live stream.
Lots more to sort out but that is now working:)
Mick

lurch, 2024-03-25:
I guess this issue can be closed then?

micksulley, 2024-03-25:
Still got lots to sort out but this issue can be closed.

closed 2024-03-25 by micksulley