raspberrypi/picamera2 #975: [OTHER]

IncredibleRichie, 2024-03-07:
**Please only include one item/question/problem per issue!**
I want to leave a info for all users facing:
```
ERROR V4L2 v4l2_videodevice.cpp:1697 /dev/video13[27:out]: Failed to queue buffer 0: Invalid argument
ERROR RPISTREAM rpi_stream.cpp:276 Failed to queue buffer for ISP Input
```

After an `apt upgrade`  i got this error.

Downgrading everything that got updated before resolves the issue.

I´ve used  `sudo apt install $(sudo find /var/cache/apt/archives/ -name *deb  |grep -v partial |tr "\n" " ")` to downgrade.

IDK if I missed something but it seems others having the issue too: 

https://www.reddit.com/r/klippers/comments/1ay9k74/picam_stopped_working/

kbturk, 2024-03-26:
I don't know the solution but I'm also having the same problem running the example programs in this library. Downgrading worked for me too. Thanks for posting that.

IncredibleRichie, 2024-04-15:
@davidplowman do you have any ideas about this issue?

davidplowman, 2024-04-16:
Hi, would you be able to:

* Say what kind of Pi you are using.
* Say what camera you are using (looks like imx477).
* Install a fresh copy of the latest OS. Let it perform all its updates (`sudo apt update` and `sudo apt full-upgrade`).
* Confirm your kernel version (output of `uname -a`) and OS version (`cat /etc/os-release`).
* Confirm your libcamera version (output of `libcamera-hello --version`).
* Don't install or touch anything else, just try the camera with `libcamera-hello -t 0`. Does it work?
* If that works, try Picamera2. If that fails, please provide the shortest script that you can which fails and the console output. Also please check `dmesg` to see if there are any errors at the end.

Thanks!

mgoutelle, 2024-04-16:
Hi,
I sure can, but only on Thursday when I'm back at the office, to have access to the proper hardware. I will update this comment then.