raspberrypi/picamera2 #978: LensPosition isn't Accurate in Meters? [OTHER]

quitmeyer, 2024-03-11:
Im using an arducam 64mp with a rpi4 on bookworm

I noticed in the manual that it says the LensPosition is in "diopters" meaning a unit that is 1/meters

but i realized with our setup the focus plane is always exactly 30cm away. With autofocus on manual focus i tend to set the lens position to 7.82 which would suggest the focus plane is 12cm away. (as if i measured and set the diopters i would do 1/,3 = 3.33 for lensposition, correct?)

I didn't file this as a bug because
1) maybe arducam messed something up

or 
2) maybe the diopters just isn't that accurate in general? or maybe not accurate in macro (like, "don't complain you are only off by 20 cm")?

and 
3) it's not a big deal, i can just set all my devices in their own generic focus units where 7.82 works well for 30cm

but im curious if im doing anything wrong here?

davidplowman, 2024-03-12:
No, I doubt you're doing anything wrong. I can only really speak for the camera module 3, but in that case the dioptre values should be taken as "best effort". All the modules have minor manufacturing differences, and the lens position is susceptible to many factors including gravity if your module is not completely level. So you will definitely see some variation.

Finally, I don't know anything about the Arducam module. They may have calibrated it carefully, or they may not - you'd have to ask them!

quitmeyer, 2024-03-12:
ok terrific! thanks @davidplowman

closed 2024-03-12 by quitmeyer