raspberrypi/picamera2 #984: [HOW-TO]  Advice required about broken pipe error on camera

Brunnian, 2024-03-15:
[logfile 2024-03-15.log](https://github.com/raspberrypi/picamera2/files/14615699/logfile.2024-03-15.log)
[TimeLapseCamera0.79.zip](https://github.com/raspberrypi/picamera2/files/14615719/TimeLapseCamera0.79.zip)

I have been trying to build a timelapse camera, to capture sunrises.  This was going remarkably well, once I debugged my software.  The thing consists of a python script which is called at boottime (RTC alarm) from the chron queue, calculates the time of sunrise, waits, takes the photo, and then calls a command to save photos to the cloud.  The thing then starts tomorrow's boot time, updates the RTC and halts. Hurrah!

(If I manually boot it takes a test image, sends it to the cloud, sets the RTC but does not halt, so I can access the system during the day for management and updating).  

But toward the end of february, it started to fail when starting from the RTC alarm/chron queue.  Not every day, but increasingly more so.  There is a log file attached of such a failure, but basically the root cause seems to be:

>   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1123, in start_self.camera.start(controls)
>   RuntimeError: Failed to start camera: Broken pipe

I have turned on camera debugging as described in https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf but found little help on expanding further upon what is displayed.  The debug statements I did get just says the same thing, a broken pipe.

If I run the script manually from within Thonny, it works every time.  I don't see the broken pipe like that.

I don't think it can be something to do with the boot-up from the chron queue running as root, because it sometimes works.

**So, questions:**
1.  Can anyone suggest why the camera pipe is sometimes broken and sometimes not?  Is this a bug to be reported?
2.  I am considering trapping the broken pipe and restarting the camera.  I am not sure exacctly what part of the initialisation process, or the photography process, needs to be re-done if I do that.  Any suggestions?  Could the broken pipe be discovered during the set configuration action?
3.  Am I missing some other bit of initialisation?

davidplowman, 2024-03-15:
Hi, thanks for the question. It looks to be a V4L2 error, so quite low down in libcamera itself (or the drivers even below that). 

From the log file I see you're running a Pi 5 with an HQ cam.  I assume you've updated to all the latest software? I think it may be worth asking in the Raspberry Pi camera forum, as you'll get a wider audience there - this list is mostly for stuff that we can pin down to the Python layer. Thanks.

Brunnian, 2024-03-15:
@davidplowman  - yes, I have kept everything up to date.

I will try them too

Brunnian, 2024-03-17:
[logfile 2024-03-17 normal.zip](https://github.com/raspberrypi/picamera2/files/14627429/logfile.2024-03-17.normal.zip)
Just for completeness, here is a log file from a couple of days later that shows it working normally.

Not seen anything from the other query yet, and I trawled the V4L2 pages with increasing bemusement.  I also notice there is a new kernel, so I shall update to that.

Brunnian, 2024-04-08:
For interest, I have modified the setting of the RTC so that the system wakes up 120 seconds before it is required, rather than 90 seconds, to see if that would help.  It has not failed with this pipe error since.

The script is run from the crontab at boot, so I wondered if e.g. the file system was not yet stable at 90 seconds.  It seemed unlikely, because it is, after all, executing the script from the file system, but maybe some other thing needed to be established.  I will continue to monitor.