raspberrypi/picamera2 #985: [BUG] Potential kernel issue relating to v4l2 with latest kernel, which I believe affects picamera2

chrisruk, 2024-03-16:
**Describe the bug**

After upgrading to the latest kernel `6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`
noticed I didn't seem to be getting many videos from my Raspberry Pi camera running `capture_circular_stream.py`.

After looking in my journal, I found interesting kernel messages (listed later).

**To Reproduce**

Leave `capture_circular_stream.py` running over long time period and monitor journalctl.

**Expected behaviour**

No kernel issue.

**Console Output, Screenshots**
```
Mar 15 23:42:31 pi2 kernel: vc_sm_cma_vchi_rx_ack: received response 2476813, throw away...
Mar 15 23:42:31 pi2 kernel: INFO: task capture_circula:817 blocked for more than 120 seconds.
Mar 15 23:42:31 pi2 kernel:       Tainted: G         C         6.6.20+rpt-rpi-v8 #1 Debian 1:6.6.20-1+rpt1
Mar 15 23:42:31 pi2 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Mar 15 23:42:31 pi2 kernel: task:capture_circula state:D stack:0     pid:817   ppid:1      flags:0x00000004
Mar 15 23:42:31 pi2 kernel: Call trace:
Mar 15 23:42:31 pi2 kernel:  __switch_to+0xe8/0x168
Mar 15 23:42:31 pi2 kernel:  __schedule+0x380/0xd70
Mar 15 23:42:31 pi2 kernel:  schedule+0x64/0x108
Mar 15 23:42:31 pi2 kernel:  schedule_preempt_disabled+0x2c/0x50
Mar 15 23:42:31 pi2 kernel:  __mutex_lock.constprop.0+0x2ac/0x608
Mar 15 23:42:31 pi2 kernel:  __mutex_lock_slowpath+0x1c/0x30
Mar 15 23:42:31 pi2 kernel:  mutex_lock+0x50/0x68
Mar 15 23:42:31 pi2 kernel:  v4l2_m2m_fop_poll+0x38/0x80 [v4l2_mem2mem]
Mar 15 23:42:31 pi2 kernel:  v4l2_poll+0x54/0xc8 [videodev]
Mar 15 23:42:31 pi2 kernel:  do_sys_poll+0x2a4/0x5b8
Mar 15 23:42:31 pi2 kernel:  __arm64_sys_ppoll+0xb0/0x140
Mar 15 23:42:31 pi2 kernel:  invoke_syscall+0x50/0x128
Mar 15 23:42:31 pi2 kernel:  el0_svc_common.constprop.0+0x48/0xf0
Mar 15 23:42:31 pi2 kernel:  do_el0_svc+0x24/0x38
Mar 15 23:42:31 pi2 kernel:  el0_svc+0x40/0xe8
Mar 15 23:42:31 pi2 kernel:  el0t_64_sync_handler+0x100/0x130
Mar 15 23:42:31 pi2 kernel:  el0t_64_sync+0x190/0x198
```

I grepped my logs and found this message only seemed to appear after upgrading to the latest kernel.

**Hardware :**
Pi 4 and Raspberry Pi Auto Focus Camera 

Many thanks!

davidplowman, 2024-03-18:
Hi Chris, thanks for the report. Any idea how long it needs to be running, is several days enough do you think? Do you think it's also still an issue in the latest 6.6 kernel which was generally released at the end of last week?

@lurch Andrew, I don't suppose I could tempt you into giving this one a try (pretty please)?

ajkelsey, 2024-03-18:
@chrisruk I'm having a similar issue (#986). How many days did your script run? Mine ran for about 250 hours. Do you know how many files were written before it started to write 0 bytes? I'm wondering if it's a number of files or a length of time. I've started a counter in my script to collect that information.

chrisruk, 2024-03-18:
Hi David, 

I'm fairly sure the first time I noticed this was with the latest kernel (6.6.20+rpt-rpi-v8).

I've grepped the logs for 'blocked for', alas it seems to occur very infrequently for me, so it may be extremely hard to replicate.

It seems to occur a number of times in a row, but it looks like I've only seen it on Dec 09, Feb 14, Mar 15, Mar 16.

Note that my script starts/stops each day via cron, running the script for around 13hrs a day.

@ajkelsey I'm not running the same process continuously I'm a afraid and I don't think I've seen 0 byte files unless I'm mistaken.

davidplowman, 2024-03-19:
Well, let's put this on the list of things to try and reproduce - that's usually always the first thing to try and do!

lurch, 2024-03-25:
I've now been running the `capture_circular_stream.py` script continuously for a whole week, and haven't seen any errors.
@chrisruk Is there still plenty of free disk-space on your SD card? (I'm just guessing, but I wonder if a lack of disk-space _might_ cause a `hung_task_timeout`? :shrug: )

davidplowman, 2024-03-25:
It seems quite suspicious to me that the `vchi` error message has also been observed now in issue https://github.com/raspberrypi/picamera2/issues/986. So I do believe there's something going on here, just it's proving very difficult to provoke at our end.

On that other issue, the following suggestion has been made:

> Running with start_debug=1 in config.txt, and using sudo vclog -m and sudo vclog -a after it goes wrong would be useful.

Chris, would you be able to give that a try, then maybe we can learn something more if it happens to you again?

davidplowman, 2024-03-26:
@6by9 Could you perhaps answer Chris's question, re. whether `sudo vclog -a` should be run immediately? Thanks!

6by9, 2024-03-26:
There should be relatively few asserts in the firmware on a normal system, so timing shouldn't be critical. 
(If you're getting lots of asserts logged under normal situations, then let us know as they ought to be fixed).

By default there is relatively little logging enabled on the firmware, so whilst there will be more churning through the `vclog -m` buffer, it again shouldn't be too verbose.

chrisruk, 2024-03-26:
Thanks for that information!  Will post an update if the issue occurs again, with more log info.

davidplowman, 2024-04-05:
Hi Chris, and thanks for the update. I've been running the other case we have showing these `vc_sm_cma_vchi_rx_ack` errors (https://github.com/raspberrypi/picamera2/issues/986), so far for about 10 days without any problems. Given my failure to provoke any errors, and the fact that I'm on the latest OS and kernel (which is 6.6), I wonder if it might be worth everyone making sure they're fully up to date (usual caveats about backing stuff up apply!), just to be sure that I'm chasing something that is still there! But the SD card tip is probably a good one, it's always something to try (like dodgy power supplies or long cables!).

chrisruk, 2024-04-09:
Hi David, replacing the SD card has meant videos are now recorded throughout the time period my script runs :) (Before it seemed to be stopping receiving new frames after a period of time)

Feel free to close this issue now.  I'm thinking the hanging task error might have been related to issues with the SD card.

davidplowman, 2024-04-10:
Cool, thanks for the update! If it comes back... you know where we are :)