raspberrypi/picamera2 #986: [OTHER] After several days, picamera starts to write 0 byte video files.

ajkelsey, 2024-03-17:
I have a script that will automatically capture a video and write it to the drive. It will do this several hundred times per day. I have found that after several days, the video files written are 0 bytes in size. I have plenty of drive space, RAM, and CPU available when this occurs. When I kill the script and restart, it works fine. The script ran for approx. 250 hours and 23 minutes before this happened.

I've included some of the relevant code below. When I run the stop_video() function, I do run it in a separate thread.

```
def init_camera():
    try:
        p_cam = Picamera2()
        width = int(config['resolution_x'])
        height = int(config['resolution_y'])
        cam_config = p_cam.create_video_configuration(main={'size': (width, height)}, buffer_count=12)
        p_cam.configure(cam_config)
        encdr = H264Encoder()
        outpt1 = CircularOutput(buffersize=int(config['cam_pre_record']) * 30)
        p_cam.start_recording(encdr, outpt1)
        if config['http_server'] == '1':
            try:
                http_streamer(p_cam, encdr)
                logger.info('HTTP server initialized.')
            except Exception:
                logger.exception('HTTP server exception has occured.')
        logger.info("Camera initialized.")
    except Exception:
        logger.exception('Camera exception has occured.')
    return p_cam, outpt1

def start_video(f_date):
    if config['camera_facility'] == '1':
        path_name = '/opt/speedcam/images/' + f_date
        filename = path_name + '/' + f_date + '.h264'
        try:
            if exists(path_name) == False:
                os.mkdir(path_name)
            if exists(filename) == False:
                output.fileoutput = filename
                output.start()

                logger.debug(filename + " saved.")
        except Exception:
            logger.exception('Image capture exception.')
    else:
        filename = 'unknown'
    return filename

def stop_video(filename):
    if config['camera_facility'] == '1':
        time.sleep(int(config['cam_post_record']))
        output.stop()
    if os.path.getsize(filename) == 0:
        print('Video file size is 0.')
        logger.warn('Video file size is 0.')
```

davidplowman, 2024-03-18:
Hi, thanks for the report. Could you perhaps cut this down into a fully self-contained script that is also _as short as possible_. That would give us something we can use to try to reproduce the problem.

Please also confirm what kind of Pi you are using, what the camera is, and that all the software is fully up to date. Please confirm OS version (`cat /etc/os-release`), kernel version (`uname -a`), libcamera version (`libcamera-hello --version`) and Picamera2 version (`apt list python3-picamera2`). Thanks!

ajkelsey, 2024-03-18:
I want to say I updated all the software prior to running the script, but I don't remember. If it wasn't, it shouldn't be terribl;y out of date because I did upgrade the OS to bookworm this year.

```
Raspberry pi 4 8MB RAM
Arducam for Raspberry Pi HQ Camera, 12.3MP IMX477

OS Version:  
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm

Kernel Version: Linux speedcam.grandroyal.org 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

libcamera Version: 
rpicam-apps build: 9ae39f85ae6b 30-01-2024 (13:24:24)
libcamera build: v0.1.0+118-563cd78e

Picamera2 Version: 
python3-picamera2/stable,stable 0.3.17-1 all [upgradable from: 0.3.16-1]
python3-picamera2/now 0.3.16-1 all [installed,upgradable to: 0.3.17-1]
```

```
#!/bin/python

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput
import time
from threading import Thread

def init_camera():
    p_cam = Picamera2()
    cam_config = p_cam.create_video_configuration(main={'size': (1920, 1080)}, buffer_count=12)
    p_cam.configure(cam_config)
    encdr = H264Encoder()
    outpt1 = CircularOutput(buffersize=90)
    p_cam.start_recording(encdr, outpt1)
    return p_cam, outpt1

def start_video():
    global index
    file = f'video{index:0>5d}.h264'
    index += 1
    output.fileoutput = file
    output.start()
    return 

def stop_video():
    time.sleep(3)
    output.stop()

if __name__ == "__main__":
    index = 1
    picam, output = init_camera()
    while True:
        try:
            start_video()
            cam_thread = Thread(target=stop_video(), daemon=True)
            cam_thread.start()
            time.sleep(180)
        except KeyboardInterrupt:
            break
    picam.stop()
```

davidplowman, 2024-03-18:
Is there anything printed to the console, or anything suspicious in `dmesg` when it's not working? And when it's not working, is it receiving camera frames at all? It feels similar to https://github.com/raspberrypi/picamera2/issues/985, so it would be nice to confirm whether that's really true or not.

ajkelsey, 2024-03-18:
I did not check dmesg when this happened and I have since rebooted. I have made a note to do so the next time it happens. 

How can I check if it is receiving camera frames?

davidplowman, 2024-03-18:
I guess we could make print out a message occasionally, maybe add
```
count = 0

def callback(request):
    global count
    count += 1
    if count % 30 == 0:
        print(count)
```
and then `p_cam.pre_callback = callback` just after `p_cam.start_recording(encdr, outpt1)`.

ajkelsey, 2024-03-18:
I checked journalctl and there isn't anything related to the kernel at the time this started to happen.

I've put the callback in start_video() because where you suggested is only run once. Right now I am getting the frame count printed ever second with increase of 30. What is the expected output if the camera stops receiving frames? The counter stops increasing?

davidplowman, 2024-03-19:
Yes, it's just counting frames, so if that count is still coming out then it's receiving camera frames. Anyway, let's try and be sure whether the problem is related to the camera stopping (most likely, I think) or something else.

davidplowman, 2024-03-19:
I've started it running on my own Pi 4. I've changed the 180s delay to 15s, hoping that it may misbehave faster. Obviously if there's anything we can find to make it fail more quickly that would be good to know, if it's really going to take 250 hours then progress is going to be slow!

ajkelsey, 2024-03-23:
After about 5 days, the counter stopped at 12756420. However, videos continue to be recorded and are greater than 0 bytes.

ajkelsey, 2024-03-23:
Interesting. Not long after my above post, the videos became 0 bytes. The last video correctly written was at 11:46 and the first 0 byte video was 11:55, March 23. The camera was initialized at 13:00 on March 18. The counter was 12756420 when it stopped, and if I figured this correctly, the counter stopped at March 23 11:06:54.

Camera initialized: March 18, 13:00
Counter stopped: March 23, 11:06:54
Last correct video recorded: March 23, 11:46
First 0 byte video recorded: March 23, 11:55

A copy of dmesg:
[speedcam_dmesg.txt](https://github.com/raspberrypi/picamera2/files/14733037/speedcam_dmesg.txt)

dmesg sample:
```
[431719.138286] vc_sm_cma_vchi_rx_ack: received response 51307935, throw away...
[431852.797596] INFO: task speedcam.py:163714 blocked for more than 120 seconds.
[431852.797615]       Tainted: G         C         6.1.0-rpi8-rpi-v8 #1 Debian 1:6.1.73-1+rpt1
[431852.797618] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[431852.797621] task:speedcam.py     state:D stack:0     pid:163714 ppid:1591   flags:0x0000000c
[431852.797631] Call trace:
[431852.797634]  __switch_to+0xe8/0x160
[431852.797651]  __schedule+0x328/0x960
[431852.797657]  schedule+0x64/0x104
[431852.797662]  schedule_preempt_disabled+0x2c/0x4c
[431852.797668]  __mutex_lock.constprop.0+0x2c4/0x640
[431852.797674]  __mutex_lock_slowpath+0x1c/0x30
[431852.797679]  mutex_lock+0x60/0x70
[431852.797685]  v4l2_m2m_fop_poll+0x38/0x74 [v4l2_mem2mem]
[431852.797709]  v4l2_poll+0x54/0xc0 [videodev]
[431852.797762]  do_sys_poll+0x29c/0x5b0
[431852.797770]  __arm64_sys_ppoll+0xb0/0x13c
[431852.797775]  invoke_syscall+0x50/0x120
[431852.797783]  el0_svc_common.constprop.0+0x4c/0xf4
[431852.797789]  do_el0_svc+0x34/0xd0
[431852.797794]  el0_svc+0x2c/0x84
[431852.797799]  el0t_64_sync_handler+0xf4/0x120
[431852.797804]  el0t_64_sync+0x18c/0x190

```

davidplowman, 2024-03-25:
So is it completely repeatable that you get this error? And you're running that script that you posted [here](https://github.com/raspberrypi/picamera2/issues/986#issuecomment-2004227866)?

I've been trying that myself, but with a 15 second wait. It ran quite happily for several days until it had filled its 64GB disk and then stopped with a "regular" error. Maybe I need to try it again with the longer delay and see if it does the same as yours after 5 days?

@6by9 Does the stack trace [above](https://github.com/raspberrypi/picamera2/issues/986#issuecomment-2016551531) mean anything to you, by any chance?

davidplowman, 2024-03-25:
I notice that this v4l2/vchi related message _is_ the same one we see in https://github.com/raspberrypi/picamera2/issues/985.

6by9, 2024-03-25:
The blocked task stack trace isn't useful - it only tells you the final thread that is blocked.

```
 vc_sm_cma_vchi_rx_ack: received response 51307935, throw away...
```
and probably the few lines before that would be the interesting bit.
vcsm_cma has received an ack for a message we are not (possibly no longer) waiting for.

Running with `start_debug=1` in config.txt, and using `sudo vclog -m` and `sudo vclog -a` after it goes wrong would be useful.

davidplowman, 2024-03-25:
Thanks for the tip. I'll keep trying to reproduce this, but in the meantime, @ajkelsey would you be able to try that?

ajkelsey, 2024-03-25:
> So is it completely repeatable that you get this error? And you're running that script that you posted [here](https://github.com/raspberrypi/picamera2/issues/986#issuecomment-2004227866)?
>

No, that is the shortened script you had requested. The full script includes radar, temperature sensor, and a couple schedulers. This is the first time I have checked for this error, so I don't know if it happens every time the files are 0 bytes. I suspect that it is, but I will have to wait for this to happen again to confirm.

> and probably the few lines before that would be the interesting bit.
> vcsm_cma has received an ack for a message we are not (possibly no longer) waiting for.

@6by9 The full file is linked in the post. The line above happens days before. 

> Running with `start_debug=1` in config.txt, and using `sudo vclog -m` and `sudo vclog -a` after it goes wrong would be useful.

 Will do.

6by9, 2024-03-25:
Spontaneously getting a `vc_sm_cma_vchi_rx_ack` failure is bizarre, as it waits with a `wait_for_completion_interruptible`. That would imply an interrupted system call. I don't think stopping encoding is going to trigger that, so I can't think how that is happening.

ajkelsey, 2024-03-27:
Results from today. A reboot has not occurred since my last report.

[vclog_a2.txt](https://github.com/raspberrypi/picamera2/files/14774597/vclog_a2.txt)
[vclog_m2.txt](https://github.com/raspberrypi/picamera2/files/14774598/vclog_m2.txt)
[speedcam_dmesg3.txt](https://github.com/raspberrypi/picamera2/files/14774599/speedcam_dmesg3.txt)

Should I keep reporting?

davidplowman, 2024-03-27:
That's probably OK for now, thanks, until we have more idea what to look for.

We're seeing a load of errors related to `vc_image_helper_get_sdram_page_size` (which is pretty hard to fathom, I must say). Just to clarify, are these coming out spontaneously and are then followed by zero byte files, or do they come out when you quit the program? Same question as regards the `vc_sm_cma_import_dmabuf`/`__vb2_queue_cancel` messages in `dmesg` - do they appear spontaneously, or only when you try and quit?

Thanks! (FWIW, I'm still trying to reproduce this, though without success so far. Actually I'm running an up-to-date image with a 6.6 kernel, though I have no reason to suppose there's anything different about it.)

popcornmix, 2024-03-27:
There was a [recent](https://github.com/raspberrypi/firmware/commit/45319db29eb5e4f67feab5c4194bc1f28c574ed0) bug for to firmware that could fix an issue with `vc_image_helper_get_sdram_page_size`.

Might be worth trying a firmware update with rpi-update (usual caveats about using rpi-update apply).

ajkelsey, 2024-03-27:
`vc_image_helper_get_sdram_page_size` is spontaneous. I can see it rolling the log over right now with a fresh restart of the script.

`vc_sm_cma_import_dmabuf` and `__vb2_queue_cancel` happened when interrupting the script.

Let me know if there is anything else I can help with.

davidplowman, 2024-03-27:
Thanks. And `vc_image_helper_get_sdram_page_size` is directly associated with the problem, that is, if you haven't seen that, then it works?

Are you able to try a firmware update, as was suggested? (Maybe on a spare SD card, or after carefully backing up anything important!)

@popcornmix Presumably `sudo rpi-update` is the correct thing to do, or is there a particular version to try? (Or might it be better to do the `sudo apt update` and `sudo apt full-upgrade` procedure?)

ajkelsey, 2024-03-27:
> And `vc_image_helper_get_sdram_page_size` is directly associated with the problem, that is, if you haven't seen that, then it works?

It works while that is happening. It seems to be appearing constantly.

I will look into the firmware update.

6by9, 2024-03-27:
Without the firmware fix you will get masses of asserts from video encode.
The update should solve it (or using a firmware from before Feb 29, 2024)

popcornmix, 2024-03-27:
> @popcornmix Presumably sudo rpi-update is the correct thing to do, or is there a particular version to try? (Or might it be better to do the sudo apt update and sudo apt full-upgrade procedure?)

Yes rpi-update is correct. The apt firmware doesn't contain the fix yet.

ajkelsey, 2024-04-05:
https://github.com/raspberrypi/picamera2/issues/985#issuecomment-2039572233

My current version:
Linux 6.6.22-v8+ aarch64

ajkelsey, 2024-04-08:
I do not run that shorter script. 

Yes, it has failed using 6.6 kernel.

What's interesting is that I was recently getting failures multiple times per day. Rebooting didn't seem to change it, but after power cycling, it was no longer failing daily. 

In my more complicated script, it is possible for it to want to start a new recording before the previous one has stopped, but it looks like that isn't possible. I used this code to test it:

```
#!/bin/python

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput
import time
from threading import Thread

def init_camera():
    p_cam = Picamera2()
    cam_config = p_cam.create_video_configuration(main={'size': (1920, 1080)}, buffer_count=12)
    p_cam.configure(cam_config)
    encdr = H264Encoder()
    outpt1 = CircularOutput(buffersize=90)
    p_cam.start_recording(encdr, outpt1)
    return p_cam, outpt1

def start_video():
    global index
    global file
    file = f'video{index:0>5d}.h264'
    index += 1
    output.fileoutput = file
    output.start()
    print(f'{file} has started.')
    return 

def stop_video():
    global file
    time.sleep(10)
    output.stop()
    print(f'{file} has stopped.')

if __name__ == "__main__":
    index = 1
    file = ''
    picam, output = init_camera()
    while True:
        try:
            start_video()
            cam_thread = Thread(target=stop_video(), daemon=True)
            cam_thread.start()
            # time.sleep(1)
        except KeyboardInterrupt:
            break
    picam.stop()
```

davidplowman, 2024-04-08:
Yes, I'm still struggling here because I've never managed to see this problem. Can you just confirm for me again, what's the simplest/shortest script for which you're quite sure you've seen the problem? I will download a fresh copy of the latest image and then try exactly that.

ajkelsey, 2024-04-08:
I have only run my full script to achieve the problem. I have not tried anything else. I am happy to provide that, but I am using it with a radar device. The output of the radar is via a terminal connection. It can probably be mimicked with another device and a script that generates the text output if you are interested.

davidplowman, 2024-04-08:
OK, I see. The current approach doesn't seem to be getting anywhere much, but if you're able to provide a script that we can run in exactly the same way that you do, and which has been confirmed to produce the problem, then I'm happy to set that running again here and give the whole thing another try!

ajkelsey, 2024-04-08:
Yeah, I have a feeling there is something else involved than just picamera. I'll keep my eyes peeled for more clues. Thanks for having a look into it.

davidplowman, 2024-04-09:
One more thought... you mentioned radios, and in the past we have certainly seen radio interference cause corruption on the camera ribbon cable leading to cameras "stopping". I don't know if this might be an issue in your case, but mitigation for interference might be worth considering (shorter cables, shielded cables, moving equipment around..).

ajkelsey, 2024-04-10:
Interesting. That could explain why I was getting camera failure once per day vs the once every 10 days or so. I was messing around inside the enclosure and the cable may have been placed back in a different way. Have you seen shielded pi camera cables?

davidplowman, 2024-04-11:
Raspberry Pi makes shielded orange cables, though online most (all?) seem to have the narrow Pi 5 / CM4 style connector at one end. Here in the office we have ones with wide connectors at both ends, so I don't know why those don't seem to be available. Maybe a third party would sell one, but I don't have any specific information.

ajkelsey, 2024-05-16:
I haven't done much with this for the last month or so. I did happen to come across the following error in dmesg. I am sharing it in case it is relevant/helpful. 

I do plan to explore the interference theory.

```
[544233.904016] ------------[ cut here ]------------
[544233.904030] WARNING: CPU: 3 PID: 3030596 at drivers/media/common/videobuf2/videobuf2-core.c:2024 __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[544233.904072] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc cdc_acm brcmfmac hci_uart btbcm bluetooth brcmutil imx477 binfmt_misc cfg80211 ecdh_generic raspberrypi_hwmon ecc rpivid_hevc(C) bcm2835_unicam bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp(C) rfkill v4l2_dv_timings v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_fwnode videobuf2_vmalloc v4l2_async videobuf2_dma_contig videobuf2_memops libaes videobuf2_v4l2 w1_therm videodev snd_bcm2835(C) vc_sm_cma(C) raspberrypi_gpiomem videobuf2_common w1_gpio wire mc cn nvmem_rmem uio_pdrv_genirq uio i2c_dev dm_mod fuse ip_tables x_tables ipv6 vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper v3d drm_kms_helper gpu_sched drm_shmem_helper drm drm_panel_orientation_quirks i2c_mux_pinctrl i2c_mux i2c_brcmstb snd_soc_core snd_compress snd_pcm_dmaengine i2c_bcm2835 snd_pcm snd_timer snd backlight
[544233.904234] CPU: 3 PID: 3030596 Comm: speedcam.py Tainted: G        WC         6.6.28+rpt-rpi-v8 #1  Debian 1:6.6.28-1+rpt1
[544233.904241] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[544233.904244] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[544233.904250] pc : __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[544233.904265] lr : __vb2_queue_cancel+0x38/0x2a0 [videobuf2_common]
[544233.904280] sp : ffffffc0819dba70
[544233.904283] x29: ffffffc0819dba70 x28: ffffff8101754578 x27: 0000000000000009
[544233.904291] x26: 0000000000000001 x25: 0000007fac28b6d8 x24: ffffff8101754380
[544233.904300] x23: ffffff8104b90160 x22: ffffff8124048298 x21: ffffff8102f21d30
[544233.904308] x20: ffffff8124048340 x19: ffffff8124048298 x18: 0000000000000003
[544233.904315] x17: 0000000100000000 x16: ffffffd5bc0d42c0 x15: ffffffc0819db5c0
[544233.904324] x14: 0000000000000004 x13: ffffff8104600028 x12: 0000000000000000
[544233.904331] x11: ffffff8063d9c798 x10: ffffff8063d9c6d8 x9 : ffffffd5bc12b778
[544233.904339] x8 : ffffffc0819db970 x7 : 0000000000000000 x6 : 0000000000000228
[544233.904346] x5 : ffffff80401a3b40 x4 : fffffffe010068e0 x3 : 000000008015000c
[544233.904354] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000001
[544233.904362] Call trace:
[544233.904365]  __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[544233.904381]  vb2_core_queue_release+0x2c/0x60 [videobuf2_common]
[544233.904396]  vb2_queue_release+0x18/0x30 [videobuf2_v4l2]
[544233.904414]  v4l2_m2m_ctx_release+0x30/0x50 [v4l2_mem2mem]
[544233.904439]  bcm2835_codec_release+0x64/0x110 [bcm2835_codec]
[544233.904451]  v4l2_release+0xec/0x100 [videodev]
[544233.904530]  __fput+0xbc/0x288
[544233.904539]  ____fput+0x18/0x30
[544233.904543]  task_work_run+0x80/0xe0
[544233.904550]  do_exit+0x30c/0x988
[544233.904555]  do_group_exit+0x3c/0xa0
[544233.904559]  get_signal+0x980/0x9b0
[544233.904563]  do_notify_resume+0x15c/0x1370
[544233.904569]  el0_svc+0xd8/0xe8
[544233.904576]  el0t_64_sync_handler+0x100/0x130
[544233.904582]  el0t_64_sync+0x190/0x198
[544233.904586] ---[ end trace 0000000000000000 ]---
[544233.904591] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000024a78687 in active state
```

davidplowman, 2024-05-20:
Thanks for that. I recall we did have some problems in the firmware a little while ago where the encoders were eating buffers and not releasing them. This would lead to zero length output files, and the kind of stack trace you're seeing. Have you done a `sudo apt update` and `sudo apt full-upgrade` recently? I think the fixed firmware should be available now, so that would be worth a try.

davidplowman, 2024-05-28:
Yes, I think that was it.