raspberrypi/picamera2 #989: [HOW-TO] ERROR EGL_NOT_INITIALIZED

SwordTechCorp, 2024-03-22:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
I wish to run the sample Qt program provided in the documentation with EGLFS backend. (On RaspberryPi OS Lite (Bookworm))
The problem is, the script raised an error at the following line
    qpicamera2 = QGlPicamera2(picam2, width=800, height=600, keep_ar=False)
The trace is as follows:
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 97, in __init__
    self.egl = EglState()
               ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 27, in __init__
    self.choose_config()
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 43, in choose_config
    eglInitialize(self.display, major, minor)
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
        err = EGL_NOT_INITIALIZED,
        baseOperation = eglInitialize,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7f675c5ac0>,
                c_int(0),
                c_int(0),
        ),
        result = 0
)
**Describe alternatives you've considered**
I tried `qpicamera2 = QPicamera2(picam2, width=800, height=600, keep_ar=False)` and that did work. Yet I still wish to get the egl running.

**Additional context**
Started from a fresh Lite Image (Bookworm) and installed necessary packages with
`sudo apt install python3-pyqt5`
`sudo apt install python3-picamera2`
Hello World Qt program can run with EGLFS backend normally.

SwordTechCorp, 2024-03-23:
It seems that the EGL display is retrieved from X11, which does not exist in RaspberryPi OS Lite. Still looking for a solution to use GL with Qt EGLFS backend.

davidplowman, 2024-03-25:
I'm afraid I don't really know anything about EGL, EGLFS, Qt, X, Wayland or DRM so I'm unlikely to be able to help. The audience on the issues here is not very wide, so you may have more luck asking elsewhere, especially somewhere more graphics- than camera-oriented. But I did ask around in the office here to see if anyone could help. I wouldn't say I had a great deal lot of luck, but to try and summarise what folks said:

* There was some suggestion that this sounded like it might be difficult. EGL does normally ask the window manager for the display, and if you don't have a window manager, then you will have to figure out where else you might get it from. Maybe there are some frame buffer drivers, or perhaps you could run a very minimal version of X (or some other windowing system). Presumably there are systems that have dealt with this, but it didn't sound like anyone knew much more.
* There was also the opinion that even if you can get past the initialisation, dmabufs won't work with EGLFS, and these are what the QGlPicamera2 widget relies upon. So basically, there may a second significant hurdle to overcome if you get that far.

So I'm sorry not to be more help, but if you do find anything out about how one might accomplish these things, then please do let us know.

SwordTechCorp, 2024-03-26:
Hi and thank you so much for your suggestions. I've adopted the software rendering now as I find what you said in a hard way. Here is what I've found.

- Firstly, one has to make sure their `EGL_BACKEND` environment variable is set to `drm` prior to the initialization of eglDisplay other wise they will get the error: EGL_NOT_INITIALZIED.
- Even if one has set the environment variable, another error will occur when trying to call `eglCreateWindowSurface()` as the native window handle passed into the  function is 1 when Qt is running in EGLFS backend, without window manager. This invalid pointer will lead to a segmentation fault.

I was hoping there was a way we can integrate the EGL rendering task into the Qt framework instead of having to connect to the native window. (Basically using Qt itself as the Window Manager) For the sake of time, I didn't explore deeper into this possibility. But I may try to find out a way later. 

p.s. As far as I know, QQuickItem and QQuickPaintedItem look plausible as they are compatible with EGLFS mode and they are by default running with low-level GPU API. We may integrate the rendering task into the Scene Graph which Qt Quick provides.

Thank you again and should anyone have any ideas, feel free to leave a note.

popcornmix, 2024-03-27:
While I know nothing about eglfs, I did see it's used here: https://github.com/warpme/minimyth2
which reports eglfs as working with Pi4 and Pi5.

Maybe you could work out how that is configured.

SwordTechCorp, 2024-04-02:
@popcornmix Thanks! I'll check it out.