raspberrypi/picamera2 #995: [BUG] "Failed to reserve DRM plane" when running python overlay_drm.py in RPi5 Composite

ArtemKorr, 2024-03-31:
**Please only report one bug per issue!**
I'm attempting to display the output of the Picamera2 on a DRM (Direct Rendering Manager) preview window and then output it to analog video. However, when I launch the application, I encounter the following error:
I previously inquired about this issue in the repository responsible for the kernel.  https://github.com/raspberrypi/linux/issues/5948#issuecomment-1941545899 
```
raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
```
Could you advise on how to resolve this issue and what I might be doing wrong in this scenario?
**Describe the bug**
I'm having trouble seeing the DRM (Direct Rendering Manager) preview on an RPI5 when using the composite video output.
 **Hardware :**
I'm using Raspberry Pi 5
**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.
 
**Additional context**

***uname -a***
`Linux raspberrypi 6.6.22-v8-16k+ #1747 SMP PREEMPT Wed Mar 27 17:06:56 GMT 2024 aarch64 GNU/Linux
***vcgencmd version***
```
2024/02/16 15:28:41 
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)
```
```
`cat /boot/config.txt
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
disable_overscan=1
# Enable audio (loads snd_bcm2835)
suddtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
#display_auto_detect=1
#[pi4]
#dtoverlay=imx296
#dtoverlay=ov5647o
#[pi5]
#dtoverlay=imx708

# Automatically load initramfs files, if found
#auto_initramfs=1
#[pi5]
#hdmi_force_hotplug=1
#hdmi_enable_4kp60=1

[pi4]
force_turbo=1
arm_freq=1900
core_freq=600
h264_freq=600
isp_freq=600
v3d_freq=600
hevc_freq=600
#sdram_freq=3200
arm_freq_min=1900
core_freq_min=600
gpu_freq_min=600
[... 528 lines cut ...]
[    6.142471] imx708 6-001a: camera module ID 0x0302
[    6.145716] rp1-cfe 1f00110000.csi: Using sensor imx708_wide for capture
[    6.145838] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-csi2_ch0] node id 0 successfully as /dev/video0
[    6.145881] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-embedded] node id 1 successfully as /dev/video1
[    6.145922] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-csi2_ch2] node id 2 successfully as /dev/video2
[    6.145954] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-csi2_ch3] node id 3 successfully as /dev/video3
[    6.145987] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_image0] node id 4 successfully as /dev/video4
[    6.146023] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_image1] node id 5 successfully as /dev/video5
[    6.146057] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_stats] node id 6 successfully as /dev/video6
[    6.146107] rp1-cfe 1f00110000.csi: Registered [rp1-cfe-fe_config] node id 7 successfully as /dev/video7
[    6.497449] Adding 3146784k swap on /var/swap.  Priority:-2 extents:119 across:28003616k SS
[    7.076295] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    7.076305] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    7.078967] pps pps0: new PPS source ptp0
[    7.079026] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    7.108911] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.169963] systemd[852]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   19.557506] pl011-axi 1f0003c000.serial: no DMA platform data
[   19.557568] pl011-axi 1f00040000.serial: no DMA platform data
[   22.117364] drm-rp1-vec 1f00144000.vec: [drm] User-defined mode not supported: "1280x720": 60 74440 1280 1336 1472 1664 720 721 724 746 0x20 0x6
[   54.089289] drm-rp1-vec 1f00144000.vec: [drm] User-defined mode not supported: "1280x720": 60 74440 1280 1336 1472 1664 720 721 724 746 0x20 0x6
[   61.403661] drm-rp1-vec 1f00144000.vec: [drm] User-defined mode not supported: "1280x720": 60 74440 1280 1336 1472 1664 720 721 724 746 0x20 0x6
[   61.476027] source of link 'rp1-cfe-fe_config':0->'pisp-fe':1 is not a V4L2 sub-device, driver bug!
[   61.476051] rp1-cfe 1f00110000.csi: Using a link rate of 900 Mbps
```

davidplowman, 2024-04-02:
Hi, thanks for the report. I don't have the right hardware to do any work on this, but am happy to do my best to help if we can establish what it is that needs fixing. Can I assume you have checked out the latest version of Picamera2, and have added your local version to your PYTHONPATH?

If you run the most basic example
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start(show_preview=True)
```
can you say exactly where it is going wrong, and what the immediate cause of the error is? Thanks!

ArtemKorr, 2024-04-03:
Hello. thanks for the answer.
I use    **picamera2 0.3.16** 
I tried the most basic example that you sent me and got this
[image]

It is important that this does not work only if you combine the output.
For example, if I take an HDMI converter then RCA, then since Linux works with HDMI for DRM output, then there are no problems

davidplowman, 2024-04-03:
Presumably your application is using one of the 32bpp `"XRGB8888"` type of output formats, which would translate into something like `pykms.PixelFormat.XRGB8888`, and this is being rejected. Do you know what kind of pixel format this display will accept, one of the 24bpp ones perhaps?

ArtemKorr, 2024-04-03:
Also https://github.com/raspberrypi/rpicam-apps/blob/70894c0c46855955442284aba9dedb66d8d4bfbf/preview/drm_preview.cpp#L90 I also tried to run this code in C++ but I get “No suitable enabled connector found” due to the fact that count_connectors turns out to be empty

ArtemKorr, 2024-04-03:
> Presumably your application is using one of the 32bpp `"XRGB8888"` type of output formats, which would translate into something like `pykms.PixelFormat.XRGB8888`, and this is being rejected. Do you know what kind of pixel format this display will accept, one of the 24bpp ones perhaps?

Well, this display works with what the analog output gives it. in this case pal.

davidplowman, 2024-04-03:
OK, this is totally beyond my understanding (I know basically nothing about DRM or KMS). @6by9 Does any of this mean anything to you?

ArtemKorr, 2024-04-03:
We previously talked about this and this is what he wrote to me.
https://github.com/raspberrypi/linux/issues/5948#issuecomment-1942096270

ArtemKorr, 2024-04-03:
I tried changing the format in drm_preview.py from AB24 to XR24, but still failed

6by9, 2024-04-03:
> OK, this is totally beyond my understanding (I know basically nothing about DRM or KMS). @6by9 Does any of this mean anything to you?

As discussed previously, Picamera2 (and rpicam-apps) are assuming they're running against vc4 as the DRM renderer, and hence have a wide choice of formats, scaling, and planes.

RP1 for DSI, DPI, and VEC (and also most SPI displays) only have a single primary plane, and only support XRGB8888, XBGR8888, RGB888, BGR888, and RGB565 formats. Asking for any other format or requiring an overlay plane will fail.

Picamera2 and rpicam-apps need to fall back to using the primary plane in the absence of a suitable overlay plane, and ensure that their pixel format choice is supported by the DRM device.

ArtemKorr, 2024-04-03:
[drm_preview.zip](https://github.com/raspberrypi/picamera2/files/14852360/drm_preview.zip)
Ok, if I understood you correctly, I tried to make the changes you say. I attached the file in a zip archive. now that line has passed, but it breaks below 
[image]

davidplowman, 2024-04-03:
Hmm, so it seems to be failing to create a `pykms.DmabufFramebuffer`. I assume you've tried a different format or two. Perhaps this kind of display doesn't support dmabufs? It would be nice if there was a way to know that. Maybe we need to allocate a `pykms.DumbFramebuffer` instead and copy the image into that (though it will involve lots of pixel copying...)?

@6by9 Does the above sound plausible, do you think?

davidplowman, 2024-04-03:
OK, that's interesting. I don't have any means of reproducing this behaviour, unfortunately. @ArtemKorr Can you check your image dimensions match the display exactly?

ArtemKorr, 2024-04-03:
OK. When I set the photo size to be the same as my display (704*432) I don’t get any errors.. But I don’t see the image on the screen either

davidplowman, 2024-04-03:
I guess you're sure that the image actually does contain something? (Perhaps you are able to save it as a JPEG and view it somewhere else?)

ArtemKorr, 2024-04-03:
This is a live stream from a camera... but the camera is working. shoots, so the images must contain something exactly

ArtemKorr, 2024-04-03:
Ok I found a bug in myself. I decided to remove the frame size by default. Now everything works
[image]

davidplowman, 2024-04-03:
Great glad it's working! I'd like to get these changes packaged into a PR that can be merged into our code base, so that in future this will "just work" for everyone, and you won't need special code edits of your own.

Would you be able to commit your changes and post your tree to a Github fork of Picamera2? Then I can tidy them up a bit and make sure I haven't broken anything else. Then I'll create a PR which can be merged. I'd be extremely grateful if you'd have a moment to test that PR before I finally merge it. Would that be OK? Thanks very much for your help with this issue.

davidplowman, 2024-04-04:
Well, I'll make up a PR in the next day or so, and maybe you could give it a try just to check that I've fixed your issue correctly. That would be great!

6by9, 2024-04-04:
FKMS is deprecated.

vc4-kms-v3d when rendering through vc4 (the composition hardware for HDMI, DSI, DPI, and VEC on Pi0-4, and for HDMI on Pi5) allows lots of options that currently picamera2 assumes are present.
With the RP1 outputs, or with other random DRM drivers (eg SPI displays, DisplayLink USB devices, etc), those features aren't available. Picamera2 needs to handle these situations properly. 
It also really wants a mechanism for passing in `dev_path` when creating the card, so you can choose which display device to render to (https://github.com/tomba/pykms/blob/master/kms/card.py#L24)

Largely the PR is to make sure that nothing gets missed out from what works for you. I'm trying to get it working on an SPI display at present, but with limited success.