raspberrypi/picamera2 #997: [BUG] need to configure twice to get correct analog gain range.

fiepfiep, 2024-04-01:
I made a driver for a sensor called mira050 which supports various bitmodes/analog gains.
in the 8bit mode, the range should be 1x-16x. However, first time, the max is 1.06 -- how come?
for some reason, i have to configure the sensor twice to make picamera2 get the correct anagain range..

```
#DARK CONDITIONS
bit_mode = selected_mode["bit_depth"]
im_data = pd.DataFrame()
print(f'you have chosen mode {selected_mode}')
#attempt nr 1
with Picamera2() as picam2:
    preview_config = picam2.create_preview_configuration(main={"size": selected_mode["size"]},
        raw={"format": selected_mode["unpacked"],
            "size": selected_mode["size"],
        })
    picam2.configure(preview_config)
    picam2.start()

    print(f'analog gain limits,attempt 1, : {picam2.camera_controls["AnalogueGain"]}')

  #attempt nr 2
with Picamera2() as picam2:
    preview_config = picam2.create_preview_configuration(main={"size": selected_mode["size"]},
        raw={"format": selected_mode["unpacked"],
            "size": selected_mode["size"],
        })

    picam2.configure(preview_config)
    picam2.start()
    print(f'analog gain limits,attempt 2,: {picam2.camera_controls["AnalogueGain"]}')
```

returns 

```

you have chosen mode {'format': SGRBG8, 'unpacked': 'SGRBG8', 'bit_depth': 8, 'size': (576, 768), 'fps': 120.08, 'crop_limits': (0, 0, 576, 768), 'exposure_limits': (132, 1000764, None)}

[0:15:08.047184742] [5146]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70-dirty (2024-02-15T15:15:09+01:00)
[0:15:08.061534799] [5190]  WARN RPI raspberrypi.cpp:1357 Mismatch between Unicam and CamHelper for embedded data usage!
[0:15:08.063826540] [5190]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/mira050@36 to Unicam device /dev/media4 and ISP device /dev/media0
[0:15:08.070712451] [5146]  INFO Camera camera.cpp:1028 configuring streams: (0) 576x768-XBGR8888 (1) 576x768-SGRBG8
[0:15:08.071349705] [5190]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/mira050@36 - Selected sensor format: 576x768-SGRBG8_1X8 - Selected unicam format: 576x768-GRBG

analog gain limits,attempt 1, : (1.0, 1.055999994277954, None)

[0:15:08.700646723] [5146]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70-dirty (2024-02-15T15:15:09+01:00)
[0:15:08.712244487] [5196]  WARN RPI raspberrypi.cpp:1357 Mismatch between Unicam and CamHelper for embedded data usage!
[0:15:08.713218683] [5196]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/mira050@36 to Unicam device /dev/media4 and ISP device /dev/media0
[0:15:08.719249229] [5146]  INFO Camera camera.cpp:1028 configuring streams: (0) 576x768-XBGR8888 (1) 576x768-SGRBG8
[0:15:08.719597356] [5196]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/mira050@36 - Selected sensor format: 576x768-SGRBG8_1X8 - Selected unicam format: 576x768-GRBG

analog gain limits,attempt 2,: (1.0, 16.0, None)

```

fiepfiep, 2024-04-02:
Solved - was a driver issue. 
I had to specify the gain range in the *_set_pad_format function.

closed 2024-04-02 by fiepfiep

davidplowman, 2024-04-02:
Hi, it's a bit difficult to know what to suggest here. This is presumably either a camera driver or a libcamera bug, and not really anything to do with Picamera2.

I did try this with a v3 camera where the minimum gain is 1.12, and this seems to be advertised correctly as soon as the camera is configured, so there's no obvious sign of a problem there. Of course, I can't try your camera. Have you put some debug messages into your driver to see what analogue gain ranges it's advertising to libcamera? I guess you must have written a CamHelper? Can you say what gain code/value conversions that is being asked to do?