raspberrypi/picamera2 #1009: [OTHER]raspberrypi picamera2 error

mogomogomogo, 2024-04-12:
**Please only include one item/question/problem per issue!**
I'm trying to run a camera operating code
import time
from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration()
picam2.configure(preview_config)

picam2.start()
try:
    while True:
        time.sleep(1)

except KeyboardInterrupt:
   picam2.stop_preview()
   
   An error message appears.
   
Traceback (most recent call last):
  File "/home/pi/video2.py", line 5, in <module>
    picam2 = Picamera2()
             ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in __init__
    camera_num = self.global_camera_info()[camera_num]['Num']
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
IndexError: list index out of range
Exception ignored in: <function Picamera2.__del__ at 0x7f8a8b9a80>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 404, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 604, in close
    if self._preview:
       ^^^^^^^^^^^^^
AttributeError: 'Picamera2' object has no attribute '_preview'

christobal96, 2024-04-12:
Same Problem here.. code worked fine for weeks, yesterday i installed pandas via pip, now I get this error.

davidplowman, 2024-04-12:
Hi, thanks for the report. From what you say, am I correct in thinking that even this:
```
from picamera2 import Picamera2
picam2 = Picamera2()
```
would fail?

It sounds to me like your camera is not being recognised. What output do you get if you type `libcamera-hello --list-cameras` in a terminal window?

davidplowman, 2024-04-12:
> Same Problem here.. code worked fine for weeks, yesterday i installed pandas via pip, now I get this error.

This sounds like a different problem. I'm afraid I can't really say what would get broken by installing pandas. I would recommend installing a fresh version of the latest Raspberry Pi OS and verifying that the camera works. Then install the _most minimal_ set of modules that makes it fail, and file a separate bug report detailing _exactly_ what you did. Thanks.

mogomogomogo, 2024-04-15:
When enter this code 'libcamera-hello --list-cameras'

Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

This is how it is output to the terminal.

davidplowman, 2024-04-15:
@mogomogomogo  Thanks for the update. It seems like your camera is recognised OK. Can you confirm that
```
from picamera2 import Picamera2
picam2 = Picamera2()
```
fails, and that the problem is repeatable on a clean and updated  install of the latest version of the OS? Then we can try to reproduce that. Thanks.

mogomogomogo, 2024-04-15:
import time
from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration()
picam2.configure(preview_config)

picam2.start()
try:
while True:
time.sleep(1)

except KeyboardInterrupt:
picam2.stop_preview()
When I wrote the code, I got an error message

from picamera2 import Picamera2, Preview
import time

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration()
picam2.configure(preview_config)

picam2.start()
try:
while True:
time.sleep(1)

except KeyboardInterrupt:
picam2.stop_preview()

I just changed the order of the first and second lines, but it's running properly

thanks.

IncredibleRichie, 2024-04-15:
@davidplowman do you have any ideas about this issue?

mogomogomogo, 2024-05-02:
> @mogomogomogo Thanks for the update. It seems like your camera is recognised OK. Can you confirm that
>
> ```
> from picamera2 import Picamera2
> picam2 = Picamera2()
> ```
>
> fails, and that the problem is [...]

<New problem>

import cv2

from picamera2 import Picamera2

# Grab images as numpy arrays and leave everything else to OpenCV.

face_detector = cv2.CascadeClassifier("/usr/share/opencv4/haarcascades/haarcascade_frontalface_default.xml")
cv2.startWindowThread()

picam2 = Picamera2()
picam2.configure(picam2.create_preview_configuration(main={"format": 'XRGB8888', "size": (640, 480)}))
picam2.start()

while True:
    im = picam2.capture_array()

    grey = cv2.cvtColor(im, cv2.COLOR_BGR2GRAY)
    faces = face_detector.detectMultiScale(grey, 1.1, 5)

    for (x, y, w, h) in faces:
        cv2.rectangle(im, (x, y), (x + w, y + h), (0, 255, 0))

    cv2.imshow("Camera", im)
    cv2.waitKey(1)
    
    When you enter the code, you get this error message.
    
    [0:10:55.104459588] [2216]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
Traceback (most recent call last):
  File "/home/pi/picv.py", line 9, in <module>
    picam2 = Picamera2()
             ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 245, in __init__
    camera_num = self.global_camera_info()[camera_num]['Num']
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
IndexError: list index out of range

help me please...

davidplowman, 2024-05-02:
So, did you run just the two line code snippet that I suggested (and nothing else):
```
from picamera2 import Picamera2
picam2 = Picamera2()
```
And is that failing?

It still looks to me like your camera is not being detected. What does
```
libcamera-hello --list-cameras
```
report?

mogomogomogo, 2024-05-06:
When you connect the camera and 
libcamera-hello --list-cameras enter the corresponding code, 
it says "No camera available!"

Is it a problem with the camera module?

Ahmerssqws, 2024-05-14:
> When you connect the camera and libcamera-hello --list-cameras enter the corresponding code, it says "No camera available!"
> 
> Is it a problem with the camera module?

did you get the solution

Ahmerssqws, 2024-05-14:
> **Please only include one item/question/problem per issue!** I'm trying to run a camera operating code import time from picamera2 import Picamera2, Preview
> 
> picam2 = Picamera2() picam2.start_preview(Preview.QTGL)
> 
> preview_config = picam2.create_preview_configuration() picam2.configure(preview_config)
> 
> picam2.start() try: while True: time.sleep(1)
> 
> except KeyboardInterrupt: picam2.stop_preview()
> 
> An error message appears.
> 
> Traceback (most recent call last): File "/home/pi/video2.py", line 5, in picam2 = Picamera2() ^^^^^^^^^^^ File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in **init** camera_num = self.global_camera_info()[camera_num]['Num'] ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^ IndexError: list index out of range Exception ignored in: <function Picamera2.**del** at 0x7f8a8b9a80> Traceback (most recent call last): File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 404, in **del** self.close() File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 604, in close if self._preview: ^^^^^^^^^^^^^ AttributeError: 'Picamera2' object has no attribute '_preview'
> 
> please help..

same problem .please help if you get the solution

brunnerant, 2024-05-16:
I also have the same problem since two days 😢

Kannospark, 2024-05-17:
I have the same question QAQ.

mogomogomogo, 2024-05-19:
import cv2
from picamera import PiCamera
from picamera.array import PiRGBArray

name = 'Jiwon'

cam = PiCamera()
cam.resolution = (512, 304)
cam.framerate = 10
rawCapture = PiRGBArray(cam, size=(512, 304))

img_counter = 0

while True:
    for frame in cam.capture_continuous(rawCapture, format="bgr", use_video_port=True):
        image = frame.array
        cv2.imshow("Press space to take a photo", image)
        rawCapture.truncate(0)

        k = cv2.waitKey(1)
        rawCapture.truncate(0)
        if k % 256 == 27:
            break
        elif k % 256 == 32:
            img_name = "dataset/" + name + "/image_{}.jpg".format(img_counter)
            cv2.imwrite(img_name, image)
            print("{} written!".format(img_name))
            img_counter += 1

    if k % 256 == 27:
        print("Escape hit, closing...")
        break

cv2.destroyAllWindows()

Let me know what's wrong with this code

python jiwondata2.py
Traceback (most recent call last):
  File "/home/pi/jiwondata2.py", line 2, in <module>
    from picamera import PiCamera
  File "/usr/local/lib/python3.11/dist-packages/picamera-1.13-py3.11.egg/picamera/__init__.py", line 72, in <module>
  File "/usr/local/lib/python3.11/dist-packages/picamera-1.13-py3.11.egg/picamera/exc.py", line 41, in <module>
  File "/usr/local/lib/python3.11/dist-packages/picamera-1.13-py3.11.egg/picamera/mmal.py", line 49, in <module>
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libmmal.so: cannot open shared object file: No such file or directory

mogomogomogo, 2024-05-19:
from picamera2 import Picamera2
import cv2
from picamera.array import PiRGBArray

name = 'Jiwon'

cam = Picamera2()
cam.resolution = (512, 304)
cam.framerate = 10
rawCapture = PiRGBArray(cam, size=(512, 304))

img_counter = 0

while True:
    for frame in cam.capture_continuous(rawCapture, format="bgr", use_video_port=True):
        image = frame.array
        cv2.imshow("Press space to take a photo", image)
        rawCapture.truncate(0)

        k = cv2.waitKey(1)
        rawCapture.truncate(0)
        if k%256 == 27:
            break
        elif k%256 == 32:
            img_name = "dataset/" + name + "/image_{}.jpg".format(img_counter)
            cv2.imwrite(img_name, image)
            print("{} written!".format(img_name))
            img_counter += 1

    if k%256 == 27:
        print("Escape hit, closing...")
        break

cv2.destroyAllWindows()

Let me know what's wrong with this code

Traceback (most recent call last):
  File "/home/pi/jiwondata2.py", line 3, in <module>
    from picamera.array import PiRGBArray
  File "/usr/local/lib/python3.11/dist-packages/picamera-1.13-py3.11.egg/picamera/__init__.py", line 72, in <module>
  File "/usr/local/lib/python3.11/dist-packages/picamera-1.13-py3.11.egg/picamera/exc.py", line 41, in <module>
  File "/usr/local/lib/python3.11/dist-packages/picamera-1.13-py3.11.egg/picamera/mmal.py", line 49, in <module>
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libmmal.so: cannot open shared object file: No such file or directory

mogomogomogo, 2024-05-19:
https://core-electronics.com.au/guides/face-identify-raspberry-pi/

I'd like to proceed with what's on this link.

I'm curious about the data collection and machine learning codes.

tyokoyamax, 2024-05-20:
I experienced a similar phenomenon.
In my case, the camera cable was connected to the wrong place.
Raspberry Pi has “DISPLAY” and “CAMERA”.
If the camera cable is connected to “DISPLAY”, “No camera available!” is displayed and the camera is not recognized.

mogomogomogo, 2024-05-20:
Camera recognition is working normally.....

davidplowman, 2024-05-20:
There seem to be a lot of problems being raised all in this one issue, so I think I'm going to close it because it's too difficult for me to keep track of everything. Please open a new issue for every new problem.

Before doing that, a couple of comments:

If `libcamera-hello --list-cameras` says "no cameras detected", then nothing at all will work because no cameras are being detected. That needs to be fixed before doing anything else.

I saw one of the posts importing from `picamera.array`. The PiCamera (not Picamera2) module is for the legacy camera stack and will never work with the current camera stack. Picamera2 gives you numpy arrays directlys using the `capture_array` method.

closed 2024-05-20 by davidplowman

mogomogomogo, 2024-05-20:
What code is used to learn data using collected face image data?

Kannospark, 2024-05-25:
Actually it is not the problem of picamera2. The main reason is your computer can't correctly detect your camera. I updated my rasberrypi system, and it could detect the camera after that. Now my problem is solved. 
Hope this will help you.

Ahmerssqws, 2024-05-25:
> Actually it is not the problem of picamera2. The main reason is your computer can't correctly detect your camera. I updated my Rasberry Pi system, and it could detect the camera after that. Now my problem is solved. Hope this will help you.

I also updated and then upgraded the RPI system but till still not work .Kinldy help

davidplowman, 2024-05-28:
I think the first thing you should do is to download a fresh version of Raspberry Pi OS (use a spare SD card if you have one), let it perform its updates when it first boots, and then (without changing anything else) try
```
rpicam-hello
```
Do you see a camera preview window for about 5 seconds?

SummerDay231, 2024-09-04:
I have solved this problem by add the current user to video group. 
`usermod -a -G video [username]`

mimodia1, 2025-03-13:
In /boot/firmware/config.txt i change this two thing:
camera_auto_detect=0
dtoverlay=ov9281

because i'm using this camera model, which it's not RPi oficial.