raspberrypi/picamera2 #1010: [BUG] Unable to get raw image in raspberry pi 5

tf4ong, 2024-04-13:
[image]
I am trying to get raw images using a rqspberry pi 5 with a monchrome imx296. 
Here is my code 
[image]
I was using this code in raspberry pi 4 and it worked fine and what changes have been made vs rpi4.
With libcamera-raw it gave me a MONO_PISP_COMP1 format on the pi5 vs R10 on the pi4.

davidplowman, 2024-04-15:
Hi, please could you say what camera you are using. Please confirm that a clean install of the latest OS shows the problem. Could you also post the exact script (as short as possible, please!) that fails in the comments here so that I can copy and paste it - then I can try it for myself. Thanks!

davidplowman, 2024-04-16:
Thanks very much for the update. There's a fix [here](https://github.com/raspberrypi/picamera2/pull/1012) (actually just the second commit), and I'll merge that shortly for the next release.

Note that a Pi 5 will give you an "R16" raw image - after configuring the camera, check
```
picam2.camera_configuration()['raw']
```
so your loop should look like this:
```
while True:
    raw_image = picam2.capture_array('raw')
    cv2.imshow('img', raw_image.view(np.uint16))
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
```

tf4ong, 2024-04-16:
Thanks, I will pull and give it a try. Just wondering if it is possible to pull out the exact bit depth as the camera, like before? Like for the imx296, it would be 10 bit values?

davidplowman, 2024-04-17:
Yes, look at `picam2.camera_configuration()['sensor']`. The [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) does talk about this (section 4.2.2.3 I think), though it gets more complicated than I would like because of backward compatibility issues, and the fact that we still aren't recommending Bookworm for Pi 3 or earlier devices.

tf4ong, 2024-04-18:
Thanks, I can get raw. I have looked at the manual and tried setting sensor={'bit_depth':10} when creating the configuration. If ran but still produced 16 bit data in the raw image

tf4ong, 2024-04-18:
Do you think there will be any work around to get the exact bit depth as before on the raspberry pi 4 (10/12 bit)?  I am binning the images (8x8, on 10 bit data in 16 bit array) on the go so I can get a 16 bit image right afterwards. If I do the shifts as well, it will probably drop my frame rate abit.

davidplowman, 2024-04-19:
No. Pi 5 gives you 16-bit values for all raw formats. If you want to reduce the dynamic range you will have to shift the values. In your case, I'd compute the sums of the 8x8 pixels (or whatever it is that you're doing!), and right-shift by 6 (substitute the correct value if that one is wrong!) at the end. In the previous case you had to have different code for 10/12 bit sensors, now they're all the same. On a Pi 5, this will be _way_ faster than what you were doing on a Pi 4.

tf4ong, 2024-04-19:
Just one last thing, about shiftting.  I might be misunderstanding shifiting. Do you mean I have to compute each new pixel value of the 16 bit array to 10 bit by mutiplying with 1023/65536 to shift it? Or is there an easier way to convert the 16 bit value to 10 bit while keeping the 16 bit array numpy structure?

davidplowman, 2024-04-19:
I would do `image_10bit = image_16bit >> 6` on your numpy array (or whatever it is!).

tf4ong, 2024-04-19:
Perfect! thanks for speedy replies. I think the probelm is solved and I will give that a go

closed 2024-04-19 by tf4ong