raspberrypi/picamera2 #1013: [HOW-TO] Encode frame rate in mp4, on Pi5

sciencedjinn, 2024-04-17:
Hi,
I'm recording from a Raspberry Pi HQ camera on a Pi 5. Despite trying a whole bunch of different ways, I'm having trouble getting a correct frame rate written into the output file's metadata.

Here's the simplest code fragment to reproduce the issue:
```
from picamera2 import Picamera2
picam2 = Picamera2()
video_config = picam2.create_video_configuration(controls={'FrameRate': 10})
picam2.start_and_record_video("test.mp4", config=video_config, duration=10)
```
The resulting video _is_ recorded at the proper 10 Hz framerate  +- some jitter (confirmed by filming a stopwatch over several minutes, or by reading out individual timestamps for each frame), but the mp4 reports the framerate as about 11.1 Hz (as read by Windows, VLC or Matlab). For longer videos, the discrepancy is smaller (e.g. 10.05Hz for a 3 minute video). Now, when analysing the videos, I could read out the individual timestamps for each frame, but for longer videos that would slow down analysis quite a bit. 
Is there any way to correct this, or manually write the correct frame rate into output files?

Many thanks in advance for your help!

For reference, I'm running picamera2 version 0.3.17 on a fresh Bookworm installation.

davidplowman, 2024-04-17:
Yes, there is something going on here. When I try this I get 93 frames, probably because that 10 second duration includes the camera startup time. The thing that goes slightly awry on a Pi 5 is that it uses a software encoder which spits encoded frames out with a several-frame delay. This means that they have to get flushed out at the very end, and they all get assigned with basically the same timestamp. So the video duration ends up short, even though it has the correct number of frames (so the fps is high).

I think it should be possible to improve that, so I'll have a look.

sciencedjinn, 2024-04-17:
Amazing, thanks for the speedy reply. Anything you can do will be much appreciated!