raspberrypi/picamera2 #1026: [HOW-TO] Dropping frames when recording w/ picamera2, but not with libcamera-vid (1080p60)

Phooood, 2024-04-30:
I'm running a Pi Zero 2 W, overclocked. I can consistently get a nice, smooth 1080p60 video out of it using:
`libcamera-vid --frames 300 -b 5000000 --width 1920 --height 1080 --framerate 60 -o test.h264 --level 4.2 --denoise cdn_off`

The end goal is essentially a dashcam for a rocket, where the Pi is recording and deleting videos at timed intervals until it detects a launch, where it will continue recording until it lands, then save said recording. It needs to be small & light (and not cost a small fortune), which is why I've chosen a Pi Zero. 

The video I'm getting out of my program is sped up (like the Pi isn't fast enough), though it is 60fps, which I assume means I'm dropping frames. 

Here is my configuration within the program:

```
ctrls = cam.video_configuration

#1080p60
ctrls.controls.FrameRate = 60
ctrls.size = (1920, 1080)
ctrls.buffer_count = 10
ctrls.controls.NoiseReductionMode = 1
ctrls.format = 'YUV420'
ctrls.controls.AfMode = controls.AfModeEnum.Continuous

encoder = H264Encoder(framerate=60, enable_sps_framerate=True)
```

The only setting I can't (or don't know how to) set in the program is the H264 target level. 

On the docs, it does mention that the H264 tops out at 1080p30, but [this](https://github.com/raspberrypi/picamera2/issues/737) thread seems to suggest that is not the case. 

Are there any other settings I can try? Why is the performance different between the CLI and Python? 

I still need to be sampling an accelerometer while recording to check for launch/landing. Will this just make the problem worse?

Thanks in advance.

davidplowman, 2024-04-30:
Hi, yes there are some challenges here, not least because there's quite a lot of Python code running now whereas libcamera-vid is a native C++ application. Running more Python code may cause additional problems, though course it depends what you're doing (we know that Python can't always make use of multi-threading very well, though if most of your Python tasks are sleeping or waiting for I/O then it mostly works).

Officially, only 1080p30 is supported. In practice, folks mostly manage to get up to about 1080p50 but I think anything beyond that is quite speculative.

Some other things to try:

* I assume you're running a Lite version of the OS. (Obviously I would recommend doing so if you aren't.) Also make sure you're not trying to display anything - again, I assume this is the case already.
* Set `NoiseReductionMode` to 3. The value 1 will still spend some time doing additional denoise which will make frames slightly late.
* In your `/boot/config.txt` try setting `force_turbo=1`. The system isn't always totally great at noticing how busy things are if you're not doing lots on the display, so pushing all the clocks up may help.
* Maybe try lowering the video bitrate. Or you might actually do better to leave the bitrate higher and reduce the resolution a bit. Or reduce the framerate a bit, it would at least be interesting to see what it will support.
* File systems can be unpredictable. Might be worth writing to `/dev/null` just to rule out any effects there. (If your videos are reasonably short, you might be able to write them to `/dev/shm`, though I realise Pi Zero 2s are a bit limited)

Phooood, 2024-05-25:
Thanks David, sorry for the late reply. Finals just wrapped up. I am running the lite version and my only interface with it is through SSH. I have double checked to make sure `force_turbo=1`. 

Setting `NoiseReductionMode` to 3 has made a big difference. I incorrectly assumed 1 meant it was off. I'm currently only dropping about 10% of frames (9s video out of a 10s recording period). 

I'm not exactly sure what I should be looking for when writing to `/dev/null`, nor am I really sure on how to do that within the program. I'm currently writing using this line: `filename = '{date:%Y-%m-%d_%H.%M.%S}.h264'.format(date = datetime.datetime.now())`. I'm a bit of a noob when it comes to Linux - I have only really used it for school. 

I may end up sacrificing some resolution/framerate and just be happy with the way it'll be. I'll have to play around with it a bit and see how I feel. Thanks again.

davidplowman, 2024-05-28:
As I think I said, mostly folks find that they can't quite reach 1080p60 using Python, but they get close. The videos may be better if you reduce the framerate to (for example) 50fps, as the output can look rather choppy if it's randomly dropping frames.

To try `/dev/null` just use `filename = '/dev/null'`, and see if the performance is better. This will cause the output to be discarded, so you can assess whether filesystem delays are causing frame drops. Another suggestion might be this:
```
filename = '/dev/shm/{date:%Y-%m-%d_%H.%M.%S}.h264'.format(date = datetime.datetime.now())
```
which will write to a memory filesystem. Again this should be faster than a real filesystem, but will only be possible if your files are not too large (compared to the amount of RAM your Pi has, though that could indeed be a problem on a Pi Zero 2). Also, those files get lost on a reboot, so you have to copy them to the "real" filesystem if you need to keep them.

kodonnell, 2024-06-01:
OK, this is somewhat unrelated, somewhat related. I'm going to include it here as I think it's relevant enough. First, I'm also seeing similar things to this on a non-overclocked Pi Zero 2W (my focus is low power). I'm running a lite buildroot kernel with most things disabled (including swap), and only aiming for 20FPS at 1024x768 using python ... which by all accounts should be fine. But every so often a few frames are skipped. E.g. over the last hour there have been 5 occurrences as below

[image]

I also have logs on the output which tell me that it's because `outputframe` is not being called at all (possibly 'cos it's blocking which then blocks further up the pipe?):

[image]

For clarity, it's only using 1.25W so definitely not working hard. (I'm running a computer vision pipeline normally, which is when I first saw this, so figured it was load based - so I've removed stuff but still seeing this.) I tried it with rpicam-vid outputting timestamps: ` sudo rpicam-vid -t 600000 --width 1024 --height 768 --framerate 20 -o /data/noncritical/recordings/$t.h264 --save-pts /data/noncritical/recordings/$t.timestamps.txt --level 4.2 --denoise cdn_off` and ... funnily enough, I'm seeing frame drops too - some pretty big!

[image]

So ... this possibly points to the SD card (in part influenced by https://forums.raspberrypi.com/viewtopic.php?t=180905 - yikes!). It is one of my dev ones (i.e. been flashed so many times!), and only running at 25mhz, so that's part of it. But it made me wonder ... if the filesystem can block for a second or so, why don't I just buffer the output so it can survive a few blocks? I'll try it. And back to the original issue - this might help there too. (I don't know if the SD card is an issue in that case, but buffering will probably help.)

Turns out it seems to work. When I hammer the SD card by writing 100mb at the same time, I see it gets behind by up to 40 frames (which points to the original cause being the SD card or some other OS process writing to it e.g. systemd flushing the journal or some such). But I don't see any skipped frames in the timestamps file i.e. no frame skips. Cool.

And ... after an hour, it's looking good, no frame skips. 

So wrapping up: there's probably a case for making the fileoutput buffered as above, given most users of picamera2 will be on a pi with an SD card (some of questionable quality!) I'm not sure it'll help at the extreme end e.g. 1080p @ 60fps ... but it might, because all the video buffers get released nice and fast as things are flushed instantly onto a queue. (And for short recordings it will help.) The code changes are also relatively trivial, and the only real difference a user will notice is a little higher memory usage while it's buffering (and not much given it's encoded data). We should also definitely add some logging around frame skips to warn the user - and possibly do some basic checks on the SD car (25mhz vs 50mhz, etc.). Happy to submit a PR at some point but pretty busy right now!

Appendix ... some other things I was going to consider, just in case I've spoken too soon.

- I could try the SD card overclock overlay.
- Bump video buffers up/down and see what happens ... seems unrelated as a 1s delay is a lot of buffers! But maybe there's some deadlocking or something going on, and more buffers will help?

kodonnell, 2024-06-02:
Yep, 6 hours / 500mb or recording and no frame skips. Cool. So the filesystem writes seem to be the cause.