raspberrypi/picamera2 #1028: [HOW-TO] RuntimeError: Failed to reserve DRM plane

novski, 2024-05-01:
I tried out some simple code:
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start_and_capture_file("test.jpg")
```
And get this Error:
```
$ python test.py 
[1:57:55.336613651] [1584]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[1:57:55.410688126] [1590]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:57:55.415292000] [1590]  WARN RPI vc4.cpp:392 Mismatch between Unicam and CamHelper for embedded data usage!
[1:57:55.416745704] [1590]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[1:57:55.416881746] [1590]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:57:55.422802708] [1584]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[1:57:55.483847649] [1593]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:57:55.488306522] [1593]  WARN RPI vc4.cpp:392 Mismatch between Unicam and CamHelper for embedded data usage!
[1:57:55.489648247] [1593]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[1:57:55.489735330] [1593]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:57:55.506401806] [1584]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SBGGR10_CSI2P
[1:57:55.507412279] [1593]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
Exception in thread Thread-2 (thread_func):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 29, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 83, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1280, in process_requests
    display.render_request(display_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 74, in render_request
    self.render_drm(self.picam2, completed_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 153, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
```
What can the reason be?

davidplowman, 2024-05-02:
Hi. can you say what OS you are running (`cat /etc/os-release`). Are you running a full desktop or the lite version? Are you logged in directory to the Pi, or connecting via ssh? What kind of display is attached? Does `libcamera-hello` work? Thanks.

novski, 2024-05-03:
Its a Raspbian lite 64bit and i am ssh in it libcamera-hello shows about 5s the incrementing last line:
```
<myuser>@<myhost>:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
<myuser>@<myhost>:~ $ libcamera-hello
[0:13:37.473193192] [705]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:13:37.579142799] [710]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:13:37.583950182] [710]  WARN RPI vc4.cpp:392 Mismatch between Unicam and CamHelper for embedded data usage!
[0:13:37.585426985] [710]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
[0:13:37.585603285] [710]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:13:37.591477892] [705]  INFO Camera camera.cpp:1183 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-SBGGR10_CSI2P
[0:13:37.592168248] [710]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
#0 (0.00 fps) exp 32306.00 ag 4.00 dg 1.46
#1 (30.01 fps) exp 33251.00 ag 5.12 dg 1.11
#2 (30.01 fps) exp 33251.00 ag 5.57 dg 1.02
#3 (30.01 fps) exp 33251.00 ag 5.57 dg 1.02
#4 (30.01 fps) exp 33251.00 ag 5.57 dg 1.02
#5 (30.01 fps) exp 33251.00 ag 5.95 dg 1.00
#6 (30.01 fps) exp 33251.00 ag 6.10 dg 1.01
.
.
.
#140 (30.01 fps) exp 33251.00 ag 6.10 dg 1.02
#141 (30.00 fps) exp 33251.00 ag 6.10 dg 1.02
```

davidplowman, 2024-05-03:
Are you logged in using X forwarding, and is the machine you are using to log in running an X server? In this case you should be able to use the Qt preview window (`Preview.QT`). If the Pi has a display attached but is not running desktop environment, then the DRM preview (`Preview.DRM`) should work. There is some information on this in section 3.4 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

novski, 2024-05-03:
Why?
All i am trying to do is to save a foto.

davidplowman, 2024-05-03:
Well, the problem you're having is because it can't create a preview window, which looks to be related to the fact that you are logged in remotely, and it doesn't have the right permissions or display resources. If all you want to do is save an image and don't need a preview, use something like this:
```
import time
from picamera2 import Picamera2

picam2 = Picamera2()
preview_config = picam2.create_preview_configuration()
capture_config = picam2.create_still_configuration()
picam2.configure(preview_config)

picam2.start()
time.sleep(2)
picam2.switch_mode_and_capture_file(capture_config, "test.jpg")
```
There are many examples [here](https://github.com/raspberrypi/picamera2/tree/main/examples).