raspberrypi/picamera2 #1029: [HOW-TO]

How to capture images with the picamera2 app continuously without exiting the app and save the images with continuous names/numbers (for example image_1, image_2, image_3, etc.) so that i can capture multiple images without needing to write the file name each time?

biprav, 2024-05-01:
Also, if possible can you tell me how to add the ISO control option in the picamera2 app?

Thank you.

davidplowman, 2024-05-02:
Hi, please do have a look at some of the introductory sections of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). Also, Picamera2 does not have a concept of ISO, instead you need to set the `"AnalogueGain"` of the sensor. There's information on this in section 5 of the manaul.

There are also many examples [here](https://github.com/raspberrypi/picamera2/tree/main/examples).

Please do get back to us if you are encountering specific difficulties with your code. It's often helpful to post small code examples so that we can see how to help you - remember to include them within "tripe backquotes" so that they are formatted correctly. Thanks!

biprav, 2024-05-02:
Actually I'm very new to programming and I don't have as such knowledge regarding it. Can you just tell me which lines of code to include in the code of the app_full.py application for changing the filename of images taken continuously as stated above.
A help in this regard would be highly appreciated. 
Thank you.

davidplowman, 2024-05-02:
The image file names look like they are created [here](https://github.com/raspberrypi/picamera2/blob/main/apps/app_full.py#L201). I would say that that's a fairly complicated app just for capturing a sequence of images - but obviously you need to do what fits with your requirements. Good luck!

sandyol55, 2024-05-02:
Might be worth looking at the timelapse example, where the range counter  i  is embedded in the file name to give an incrementing list of file names?
https://github.com/raspberrypi/picamera2/blob/main/examples/capture_timelapse.py#L19

davidplowman, 2024-05-02:
> Apologies for jumping in. Might be worth looking at the timelapse example, where the range counter i is embedded in the file name to give an incrementing list of file names? https://github.com/raspberrypi/picamera2/blob/main/examples/capture_timelapse.py#L19

On the contrary, good suggestion. Thanks!

biprav, 2024-05-03:
> On the contrary, good suggestion. Thanks!

> Apologies for jumping in. Might be worth looking at the timelapse example, where the range counter i is embedded in the file name to give an incrementing list of file names? https://github.com/raspberrypi/picamera2/blob/main/examples/capture_timelapse.py#L19

Thank you so much for the response. Just another query. In which lines of the app_full.py code should I put the range counter i code? I am a beginner, so asking such common questions. Apologies for that.
Thanks.

davidplowman, 2024-05-03:
app_full.py is an event-driven GUI app, so there's not really an obvious place where you would put a loop with a range counter. You could store a counter variable somewhere, and increment it when the user clicks the capture button, but I'm really not clear if this is what you want. Also, if you're new to programming, there's a lot of not-so-easy code there, so I'd consider starting from something _much_ simpler. Maybe:
```
import time
from picamera2 import Picamera2

picam2 = Picamera2()
capture_config = picam2.create_still_configuration()
picam2.start()

for i in range(5):
    time.sleep(1)
    picam2.switch_mode_and_capture_file(capture_config, f"test_{i}.jpg")
```
Or maybe you can give a more precise description of exactly what you wish to accomplish? Thanks.