raspberrypi/picamera2 #1031: [HOW-TO] Record at 50 fps with 2 camera V3 wide angle

Rayanne-M, 2024-05-02:
I've tried to set the framerate and change the frame duration limits  but it doesn't seems to change anything. I also tried to put the resolution from 2304x1296 to 1536x864, it able me to get a higher fps but the video is cropped. When I connect a second camera v3 the fps are divided by 2 (turning around 10 fps).

Do anyone know how I could get to 50 fps with the full veiw of the wide angle ? Weirdly it seems to work when I launch the recording from the cmd using libcamera-vid

davidplowman, 2024-05-02:
Hi, thanks for the question. You probably need to set the sensor mode explicitly. You don't say what OS you are using, but this should work:
```
from picamera2 import Picamera2

picam2 = Picamera2()
main = {'size': (1920, 1080)}
raw = {'size': (2304, 1296)}
controls = {'FrameRate': 50}
config = picam2.create_video_configuration(main, raw=raw, controls=controls)
picam2.configure(config)
picam2.start()
```
You don't say what kind of Pi you have, but in general only 1080p30 is guaranteed. You can normally just about achieve 1080p50 _with a single camera_, so long as you're not running too much else. You can't start recording with a 2nd camera without dropping frames. Also, note that 2304x1296 is too large for h264 on Pi 4s or earlier. A Pi 5 should encode that, but it is 44% more pixels than 1080p, so the maximum framerate will drop in proportion.

Rayanne-M, 2024-05-02:
Thanks for you answer ! I'm using the last version of Pi OS 64 bits on a RPi 5.  I'll try that tomorrow, and I'll let you know, thanks again :)

davidplowman, 2024-05-03:
Yes, I think the thing to be aware of is that you have up to about 100 MP/s of throughput. You can divide that up relatively arbitrarily between cameras, framerates and image resolution, though at some point you might find there are additional limits (e.g. really high framerates may struggle however small you make the images). It's always a case of try it and see.

On a Pi 5, always make sure you use "YUV420" format for any stream you wish to pass to the H264 encoder, as that saves an (expensive) software conversion. For best performance, avoid displaying anything, don't run too much else, and consider putting `force_turbo=1` into your `/boot/firmware/config.txt`.

Rayanne-M, 2024-05-07:
Thank you so much ! I mangaged to record at 55 fps in 960x540 with both camera. For context the goal of the project is to have 2 camera filming (one on the side and one on the ceiling) monkey performing on touchscreen and then do automated tracking of their hand on the screen. So we would like to have net view of the hand, is there a way to control something like the shutter speed ? 
Here is a little video :) 
https://github.com/raspberrypi/picamera2/assets/123652728/4bd05a50-ced7-4d21-884f-17d0e04f38dd

Rayanne-M, 2024-05-07:
Thank you so much ! I managed to record at max 55 fps  for a resolution of 960x540 with both camera. For context, we're recording monkeys performing on touchscreen with 2 camera (one on the side and one on the ceilling) with the aim to do an automated tracking of their hands on the screen. 

Thus, we'd like to avoid blurred image I've lower the exposure time and it seemed to work. We are also trying to synchronise the camera, for now I've just calculated the time difference between the begining of the cam0 et cam1 recording and i'll cut them afterward. If you have any advice or idea to improve it, you're welcome aha.

Thanks again ! 

Here is a video of monkey :) 

https://github.com/raspberrypi/picamera2/assets/123652728/b5b56e05-f9d7-40e3-a1c0-e4ce944dc4a0