raspberrypi/picamera2 #1032: [BUG] AttributeError: module 'kms' has no attribute 'PixelFormat'

xedrion, 2024-05-07:
**Describe the bug**
I am trying to import the module picamera2 using python 3.9.6 on my raspberry pi4 with Debian 12 OS

**Console Output, Screenshots**
>>> import picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/louis/.local/lib/python3.9/site-packages/picamera2/__init__.py", line 9, in <module>
    from .picamera2 import Picamera2, Preview
  File "/home/louis/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 28, in <module>
    from picamera2.previews import DrmPreview, NullPreview, QtGlPreview, QtPreview
  File "/home/louis/.local/lib/python3.9/site-packages/picamera2/previews/__init__.py", line 1, in <module>
    from .drm_preview import DrmPreview
  File "/home/louis/.local/lib/python3.9/site-packages/picamera2/previews/drm_preview.py", line 48, in <module>
    class DrmPreview(NullPreview):
  File "/home/louis/.local/lib/python3.9/site-packages/picamera2/previews/drm_preview.py", line 50, in DrmPreview
    "RGB888": pykms.PixelFormat.RGB888,
AttributeError: module 'kms' has no attribute 'PixelFormat'

**Hardware :**
I am using the Raspberry Pi 4 with the Pi Camera v2.1

davidplowman, 2024-05-20:
Are you running a standard Raspberry Pi OS or something else? What happens when you try this:
```
import pykms
print(pykms.PixelFormat.RGB888)
```

jacobstory09, 2025-01-30:
I'm also getting this error with a Raspberry Pi 5 and the AI camera on a standard Raspberry Pi OS. 

I get an attribute error (AttributeError: module 'pykms' has no attribute 'PixelFormat') from the code above with the current version of Pi Camera and the previous version (0.3.23)

davidplowman, 2025-01-31:
Can you try the above experiment too, please?
```
import pykms
print(pykms.PixelFormat.RGB888)
```
Also what does `apt list python3-kms++` report? Have you installed any other Python bindings for kms?

The other thing to try is to install a completely clean image and check whether the error goes away.

davidplowman, 2026-01-02:
I don't really know why the code looks like that. I think there are some different "kms" packages flying around, and pykms seems to be the one that works, at least on a Pi 5, but obviously if something else has installed another version of "kms" (e.g. using another OS), then obviously it could all go wrong.

Maybe the code needs changing back, or perhaps we need to check if that symbol  exists? I probably need to try and find out why that commit got made.