raspberrypi/picamera2 #1034: [HOW-TO] project.toml issue when trying to install rpi-libcamera in conda env with python 3.9

AlphaIkaros2, 2024-05-08:
i ran: `pip install rpi-libcamera`
output: 
error: subprocess-exited-with-error
Preparing metadata (pyproject.toml) did not run successfully.
exit code: 1
[24 lines of output]
      + meson setup /tmp/pip-install-ywh86yoy/rpi-libcamera_36fbda3d0c1b49ea92de0e2c3100a67d /tmp/pip-install-ywh86yoy/rpi-libcamera_36fbda3d0c1b49ea92de0e2c3100a67d/.mesonpy-i1av0au0 -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md -Drevision=v0.2.0+rpt20240418 -Drepository=https://github.com/raspberrypi/libcamera.git -Dversion=v0.2.0+120-eb00c13d -Drepository=https://github.com/raspberrypi/libcamera.git -Drevision=main --native-file=/tmp/pip-install-ywh86yoy/rpi-libcamera_36fbda3d0c1b49ea92de0e2c3100a67d/.mesonpy-i1av0au0/meson-python-native-file.ini
      The Meson build system
      Version: 1.4.0
      Source dir: /tmp/pip-install-ywh86yoy/rpi-libcamera_36fbda3d0c1b49ea92de0e2c3100a67d
      Build dir: /tmp/pip-install-ywh86yoy/rpi-libcamera_36fbda3d0c1b49ea92de0e2c3100a67d/.mesonpy-i1av0au0
      Build type: native build
      Project name: pylibcamera
      Project version: 0.1.0
      C compiler for the host machine: cc (gcc 12.2.0 "cc (Debian 12.2.0-14) 12.2.0")
      C linker for the host machine: cc ld.bfd 2.40
      C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Debian 12.2.0-14) 12.2.0")
      C++ linker for the host machine: c++ ld.bfd 2.40
      Host machine cpu family: aarch64
      Host machine cpu: aarch64
      Message: Cloning from: https://github.com/raspberrypi/libcamera.git, revision main
      Message: Version Check Not Done
      Found pkg-config: YES (/home/pi/miniconda3/envs/nanodet/bin/pkg-config) 0.29.2
      Found CMake: /home/pi/miniconda3/envs/nanodet/bin/cmake (3.23.1)
      Run-time dependency libcamera found: NO (tried pkgconfig and cmake)
      Message: Requires libcamera-dev package to be installed - apt install libcamera-dev
      
      ../meson.build:33:20: ERROR: Dependency "libcamera" not found, tried pkgconfig and cmake
      
      A full log can be found at /tmp/pip-install-ywh86yoy/rpi-libcamera_36fbda3d0c1b49ea92de0e2c3100a67d/.mesonpy-i1av0au0/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.
See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

will-v-pi, 2024-05-09:
> Message: Requires libcamera-dev package to be installed - apt install libcamera-dev

Have you installed the latest libcamera-dev package from apt?

AlphaIkaros2, 2024-05-09:
> > Message: Requires libcamera-dev package to be installed - apt install libcamera-dev
> 
> Have you installed the latest libcamera-dev package from apt?

Yes, i installed the libcamera-dev, the issue still remains. I switched to python environment with the `--system-site-packages` flags, everything works fine now. Thanks for the reply :)