raspberrypi/picamera2 #1036: Cannot run a simple example on Pi-5

EnziinSystem, 2024-05-17:
I have a Pi-5 with the camera v3 and I installed all packages in Readme.

I also know Pi-5 uses Wayland instead of X.

Run an example:

```
$ python3 real_time_with_labels.py --model mobilenet_v2.tflite --label coco_labels.txt
[0:43:40.625317821] [5733]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:43:40.633777732] [5740]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[0:43:40.644429310] [5740]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:43:40.645693273] [5733]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:43:40.655230277] [5743]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[0:43:40.666197281] [5743]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
QObject::moveToThread: Current thread (0x7f5402f720) is not the object's thread (0x7f54239740).
Cannot move to target thread (0x7f5402f720)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/developer/.local/lib/python3.11/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx.

Aborted
```

I don't want to go back to using X on the Pi-5 because it will probably break other things.

Any solution for Picam2 on Pi-5 with V3 camera module?

Thanks.

davidplowman, 2024-05-20:
Hi, thanks for the report. Could we perhaps start with a simpler example, maybe `examples/preview.py` - does that work?

I'm also a bit suspicious that maybe you've installed a version of OpenCV with a different version of PyQt from the one in Picamera2. Can you say how you installed OpenCV? I would strongly recommend the version from apt.

EnziinSystem, 2024-05-21:
> Hi, thanks for the report. Could we perhaps start with a simpler example, maybe `examples/preview.py` - does that work?
>
> I'm also a bit suspicious that maybe you've installed a version of OpenCV [...]

Version OpenCV

```
$ python
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> print("OpenCV version:", cv2.__version__)
OpenCV version: 4.9.0
>>>
```

I also know the problem is that the latest versions of OpenCV-python do not include the library for Wayland.

This isn't a Picam2 bug, I'm just looking for a solution for Wayland on the Pi-5.

davidplowman, 2024-05-21:
Thanks for the update. So you're right, this might be related to  Wayland, or possibly Qt versions. Nonetheless, I'd like to start with perhaps the simplest example we can. Are you able to say what that very simple example does?

It is the case that if you download a vanilla Raspberry Pi OS image and change nothing at all (other than letting it perform its updates), then `examples/preview.py` works, whether running Wayland or X11. Picamera2 doesn't talk to Wayland directly, it still (I believe, I might be wrong) uses the X11 protocols and relies on XWayland to handle them. So in this respect, Wayland might not be the issue, or at least, not directly.

But I have seen problems like this when OpenCV wants to use a different version of Qt from the one installed from apt for Picamera2. If you installed OpenCV from somewhere else (did you compile it yourself? did you install it using pip?), then my suggestion would be to uninstall it and go back to the version from apt. It might be worth trying out a fresh installation of Raspberry Pi OS, and verifying that it works as I described.