raspberrypi/picamera2 #1041: [HOW-TO] DRM preview in fullscreen

ArosPrince, 2024-05-29:
Hi, I am trying display a video feed on Pi OS Lite using the DRM preview. I want it to be fullscreen. By default though it creates a small frame in the top left of the screen. Looking at the default params of the `start_preview` I assume it's 640 x 480.

Now obviously I could hard-code the screen size, something like `picam2a.start_preview(Preview.DRM, x = 0, y = 0, width = 3840, height = 2160)` or import drm and query the screen size. But it would be really surprising if there was no standard way of doing it.

There has to be a built in Picamera2 param to enable this, right?

Thanks!

davidplowman, 2024-05-30:
Hi, I'm afraid I don't really know too much about that. I'm guessing the standard way might be to import drm and query that? Alternatively, asking on one of the more general, or perhaps graphics, forums might lead you to folks with more knowledge in this area. Sorry not to be more help!

ArosPrince, 2024-05-30:
Hi, 
basically what I'm asking is if there's a standard way to do it and from what you write it looks there isn't.

The C++ implementation has a `--fullscreen` param. When creating the preview it calls `drmModeGetCrtc` to get the screen dimensions and if fullscreen is on, then it uses that for the preview size. I am looking for the same feature here in the Python implementation.

If there isn't one, I can sure use drm to figure it out myself.

davidplowman, 2024-05-31:
Once you've started the DRM preview, you can get hold of these values
```
picam2._preview.crtc.mode.hdisplay
picam2._preview.crtc.mode.vdisplay
```
which might be the numbers you want, but I wouldn't know if that's just a hack and there's some "official" API to get them!