raspberrypi/picamera2 #1042: Photon transfer function, Full well capacity of the IMX708 [OTHER]

polijn, 2024-05-29:
Hi,
@davidplowman, do you know the relation of intensity values to measured photoelectrons for the IMX708?
Is there some kind of photon transfer function available for the IMX708? 
Or is there somewhere a statement of the full well capacity of the IMX708?

I found a value for the IMX219 [here](https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/platform/turtlebot3/more_info/appendix_raspi_cam.md)
I found a paper with an extensive setup to measure it [here](https://www.spiedigitallibrary.org/journals/journal-of-electronic-imaging/volume-26/issue-1/013014/Laying-the-foundation-to-use-Raspberry-Pi-3-V2-camera/10.1117/1.JEI.26.1.013014.full)
Which is based on the literature of  J. R. Janesick (Photon Transfer).

I'm aware that this is the wrong place to ask this question, but possibly a place where others would end up searching for it :)

davidplowman, 2024-05-30:
Hi, I'm afraid I'm not aware of anything like that. You would indeed probably do better to ask on the Camera forum as the folks who select and make the modules would probably read that.

polijn, 2024-06-01:
Thanks! I will do that!

closed 2024-06-01 by polijn