raspberrypi/picamera2 #1048: [HOW-TO] Reset camera from picamera2.start() after programme crash

Bobo-amg, 2024-06-03:
Is there any way to reset a camera after programme crash?

Scenario: The running python programme using picamera2 crashes. The camera is left in a busy state as per the following error message:

Pipeline handler in use by another process
Camera __init__ sequence did not complete.
RuntimeError: Failed to acquire camera: Device or resource busy

Aside from restarting the Raspberry Pi, is there any other less obtrusive way to bring the camera back online?

Thanks

davidplowman, 2024-06-03:
Hi, could you maybe provide a short example that we can try where the Python program is crashing and leaving the camera in a bad state. I assume you've tried closing the Picamera2 instance, but in the abstract it's a bit hard to guess what may have become unhappy. Sorry not to have any ideas off the top of my head, but if you can provide a bit more detail about how things are going wrong, that would be useful and give us something to look at. Thanks!

Bobo-amg, 2024-06-03:
Thanks for quick reply.

Perhaps I have not explained clearly enough. It is not picamera2 that is crashing the programme. It is my shenanigans. Mainly whilst in development mode.

If the programme crashes before the camera is sent a picamera2.stop() then that leaves the camera in "device busy" mode, which then doesn't allow access to it. How can the camera be recovered from this state, without a reboot of the RPi?

Thanks

davidplowman, 2024-06-03:
Thanks for the update. I guess I'd still like to get a better idea of how things are broken. For example

* What happens if you call `picam2.stop()`?
* What happens if you call `picam2.close()`?
* What happens if you kill the Python process completely? Can you now start something like `rpicam-hello`?

I'd have expected `picam2.close()` might free up the camera, but maybe that's hanging or not working now? I'd also have expected killing the Python process to sort things out too. Unless libcamera underneath or the kernel has become unhappy. Probably worth seeing what `rpicam-hello` reports after killing your Python process, and if that doesn't work, maybe a look at `dmesg`.

Bobo-amg, 2024-06-03:
The only thing that appears to work is a process KILL command. Strangely, HTOP was unable to kill it (which is what had me baffled), but issuing a KILL from CLI did the job.

Thanks for the help.

davidplowman, 2024-06-03:
I guess I'm still curious, but if things are behaving better for you, then that's great!!