raspberrypi/picamera2 #1050: [HOW-TO] Sync two cameras in software

mgineer85, 2024-06-05:
**Describe what it is that you want to accomplish**
With the Pi5 there are two camera ports and I would like to sync the cameras frames as best as possible. It doesn't need to be exact.
I wonder if something similar like this can be achieved with picamera2 also:
https://picamera.readthedocs.io/en/release-1.13/api_camera.html?highlight=pid#picamera.PiCamera.framerate_delta

**Describe alternatives you've considered**
None, if not possible.

**Additional context**

davidplowman, 2024-06-05:
This is an area we've quite wanted to do work on, but haven't got round to it.

It should be possible to synchronise a pair of cameras with a software control loop. The things you'd have to do would be:

1. Look at the `"SensorTimestamp"` value in the metadata with each frame. By comparing these you can obviously tell how unsynchronised the cameras are.

2. Adjust the `"FrameDurationLimits"` control value to bring the frames together.

mgineer85, 2024-06-05:
Thank you, that was the starter information I needed. I will check it out!

mgineer85, 2024-06-07:
``So, this is cool, seems to work.

Finally I wanted to save stills so I started directly using the `create_still_configuration` which implicitely uses just one buffer. When the control loop is close to zero delta it could lead to a jump in the `SensorTimestamp` by 1/FPS seconds due to a dropped frame. I fixed by using a buffer, here `buffer_count=3`.

This is my very basic script, that just syncronizes, but nothing more yet:
```

#!/usr/bin/python

from picamera2 import Picamera2

def P_controller(Kp: float = 0.05, setpoint: float = 0, measurement: float = 0, output_limits=(-10000, 10000)):
    e = setpoint - measurement
    P = Kp * e

    output_value = P

    # output and limit if output_limits set
    lower, upper = output_limits
    if (upper is not None) and (output_value > upper):
        return upper
    elif (lower is not None) and (output_value < lower):
        return lower
    return output_value

if len(Picamera2.global_camera_info()) <= 1:
    print("SKIPPED (one camera)")
    quit()

# Primary (leads)
picam2a = Picamera2(0)
# need buffer_count > 1 because if a frame is skipped, there will be a jump in SensorTimestamp due to dropped frame which messes with the control
config2a = picam2a.create_still_configuration(controls={"FrameRate": 10}, buffer_count=3)
picam2a.configure(config2a)

# Secondary (follows)
picam2b = Picamera2(1)
# need buffer_count > 1 because if a frame is skipped, there will be a jump in SensorTimestamp due to dropped frame which messes with the control
config2b = picam2b.create_still_configuration(controls={"FrameRate": 10}, buffer_count=3)
picam2b.configure(config2b)

picam2a.start()
picam2b.start()

print("Press Ctrl+C to exit")
try:
    while True:
        metadata_picam2a = picam2a.capture_metadata()
        metadata_picam2b = picam2b.capture_metadata()

        timestamp_picam2a = metadata_picam2a["SensorTimestamp"] / 1000  #  convert ns to µs because all other values are in µs
        timestamp_picam2b = metadata_picam2b["SensorTimestamp"] / 1000  #  convert ns to µs because all other values are in µs
        timestamp_delta = timestamp_picam2b - timestamp_picam2a

        controller_output_frameduration_delta = int(P_controller(0.05, 0, timestamp_delta, (-10000, 10000)))
        control_out_frameduration = int(metadata_picam2a["FrameDuration"] + controller_output_frameduration_delta)  # sync to a, so use that for ref

        print("Cam A: SensorTimestamp: ", timestamp_picam2a, " FrameDuration: ", metadata_picam2a["FrameDuration"])
        print("Cam B: SensorTimestamp: ", timestamp_picam2b, " FrameDuration: ", metadata_picam2b["FrameDuration"])
        print("SensorTimestampDelta: ", round(timestamp_delta / 1000, 1), "ms")
        print("FrameDurationDelta: ", controller_output_frameduration_delta, "new FrameDurationLimit: ", control_out_frameduration)

        with picam2b.controls as ctrl:
            # set new FrameDurationLimits based on P_controller output.
            ctrl.FrameDurationLimits = (control_out_frameduration, control_out_frameduration)

except KeyboardInterrupt:
    print("got Ctrl+C, exiting")

picam2a.stop()
[... 166 lines cut ...]
FrameDurationDelta:  9 new FrameDurationLimit:  99991
Cam A: SensorTimestamp:  10233023706.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233023525.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  9 new FrameDurationLimit:  99991
Cam A: SensorTimestamp:  10233123689.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233123509.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  9 new FrameDurationLimit:  99991
Cam A: SensorTimestamp:  10233223674.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233223493.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  9 new FrameDurationLimit:  99991
Cam A: SensorTimestamp:  10233323655.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233323476.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  8 new FrameDurationLimit:  99990
Cam A: SensorTimestamp:  10233423638.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233423461.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  8 new FrameDurationLimit:  99990
Cam A: SensorTimestamp:  10233523621.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233523444.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  8 new FrameDurationLimit:  99990
Cam A: SensorTimestamp:  10233623604.0  FrameDuration:  99982
Cam B: SensorTimestamp:  10233623428.0  FrameDuration:  99982  SensorTimestampDelta:  -0.2 ms
FrameDurationDelta:  8 new FrameDurationLimit:  99990
^Cgot Ctrl+C, exiting

```

davidplowman, 2024-06-10:
Very cool, thanks for the update!

mgineer85, 2024-06-10:
I added also sync capture triggered by gpio on pi5. Pictures came out synced perfectly and got no hiccups after photos were taken in the synchronization. 
You mind if I send a minimal example via pullrequest?

davidplowman, 2024-06-10:
Yes, please do send new examples. If you can check the [contribution guidelines](https://github.com/raspberrypi/picamera2?tab=readme-ov-file#contributing) that would be helpful - I think it boils down mostly to running flake8, signing the commit, and submitting it to the "next" branch.

475651582, 2024-08-20:
> ``So, this is cool, seems to work.
>
> Finally I wanted to save stills so I started directly using the `create_still_configuration` which implicitely uses just one buffer. When the control loop is [...]

I also manage to write a camera-sync code basically follow the idea by match the picam2.capture_metadata()['SensorTimestamp'] together of each camera. Just like @mgineer85 posted.
However, I found this sync result always end up with about 8~10ms error.
If the Sensortimestamp is exactly the time first pixel send to CSI. The error should not be so high.

Have you met the same issue? @mgineer85

475651582, 2024-08-20:
INoted that measured the sync error by shooting a stopwatch.

mgineer85, 2024-08-20:
Hi!
 I did not test with a stopwatch as I relied on the timestamp to be accurate enough. Visually it seems to be in sync as captures in motion line up very well.

But there is more to it. Above script just demoes the synchronization, how do you actually capture? It needs to be in line with sync and should not interfere.

Check this script, it works very well for me: https://github.com/photobooth-app/wigglecam/blob/main/wigglify/app_camera_service.py
The idea is basically to use another thread for the actual capture as it is done on button press.

You can make that work?