raspberrypi/picamera2 #1053: [OTHER]

xucaiqin, 2024-06-10:
when i use 4b to push stream to my rtsp server。this question has happed.

Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly

[image]

davidplowman, 2024-06-10:
Can you share a very short version of a script that produces this error? If it doesn't rely on external components, such as RTSP servers, that would be helpful.

In my experience we do get this error from Ffmpeg sometimes, but I don't really understand why. Ffmpeg is supposed to choose a timestamp based on when it gets the packet so I find it surprising that you get errors this large. I often find I get a few warnings which then stop, and the stream works OK. Are you finding this too, or is the behaviour different?

xucaiqin, 2024-06-10:
> Can you share a very short version of a script that produces this error? If it doesn't rely on external components, such as RTSP servers, that would be helpful.
>
> In my experience we do get this [...]

This question is related to the issue #1037 I submitted earlier. So I reinstalled the Raspberry PI OS, then resubmitted my project code and ran it. But this time the problem was different. This time the question is the one I'm asking. I don't know why.
I'm using a server, set up an rtsp service. The video stream is then pushed to this rtsp service. It uses an open source rtsp server called monibuca.

> 您能否分享一个产生此错误的脚本的简短版本？如果它不依赖于外部组件（例如 RTSP 服务器），那将会很有帮助。
> 
> 根据我的经验，我们有时会从 Ffmpeg 收到此错误，但我真的不明白为什么。Ffmpeg 应该根据它获取数据包的时间来选择时间戳，因此我发现您会收到如此大的错误，这令人惊讶。我经常发现我会收到一些警告，然后停止，并且流可以正常工作。你也发现了这一点，还是行为有所不同？

xucaiqin, 2024-06-10:
> This question is related to the issue #1037 I submitted earlier. So I reinstalled the Raspberry PI OS, then resubmitted my project code and ran it. But this time the problem was different. This time [...]

[image]

davidplowman, 2024-06-10:
I'm afraid I can't really comment on the RTSP server and the various other packages you might be using, as I'm not familiar them. If you are able to capture video streams using the FfmpegOutput where we can look at and analyse the timestamps, then that would be helpful. For example, if you save your frames to an mp4 file, what timestamps do you get in the file? If we can see a problem in this simpler case, then maybe we can find a problem that we can investigate.

davidplowman, 2024-06-10:
You could try, for example, `output = FfmpegOutput("test.mp4")`.

xucaiqin, 2024-06-10:
> 例如，您可以尝试`output = FfmpegOutput("test.mp4")`。

ok ,i understand it

tatterz91, 2024-07-12:
`[mp4 @ 0x1399890] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[mp4 @ 0x1399890] Non-monotonous DTS in output stream 0:0; previous: 240000, current: 232195; changing to 240001. This may result in incorrect timestamps in the output file.`

I'm getting the same error. It happens at random. The instance above rendered the output file broken after several seconds of playback, so a bit frustrating it can happen at random. Pi4 with Bullseye and everything updated. I'm using the Full GUI that is provided in the repository, so It's not even my own code

davidplowman, 2024-07-12:
Thanks for the info. Could you please report:

* What kind of Pi you are using.
* What kind of camera you are using.
* The OS version (`cat /etc/os-release`). I would recommend using Bookworm, not Bullseye. Bullseye no longer receives fixes and updates (except critical security ones), and has not done so for some time.
* The kernel version (`uname -a`).
* The camera software version (`libcamera-hello --version`).
* The Picamera2 version if you're using Python (`apt list python3-picamera2` will tell you).
* The firmware version (`vc_gencmd version`).
* I'm afraid I don't understand what you mean by "Full GUI". A short script that demonstrates the problem is most helpful. I'm expecting such a script should be no more than a dozen or so lines.
* A link to an mp4 file that is broken in this way, and which we can download, might also be useful. Please also say what software you are using that is unable to play this file.

Thanks!