raspberrypi/picamera2 #1055: [SUGGESTION] give autofocus an initial value to start focusing with (or a bracket of values)

quitmeyer, 2024-06-12:
**Please only include one suggestion or features request per issue!**
I use the big clunky 64mp cameras for taking photos of insects on a screen 30cm away. the 64mp camera takes like 50-90 seconds to do a real decent autofocus

[image]
im using this code from @davidplowman to focus it and monitor its progress https://github.com/raspberrypi/picamera2/issues/670

and sometimes it will jump right to like 4.5 or something (i generally want something around 7.6 for the lensposition)
but then it will drop back to 0 and slowly increment and bounce around until it starts going between 7 and 8

**Describe the solution you'd like**
It would be great if i could "seed" the autofocus with a value for it to start working from. or if i could even just give the autofocus a range to stay between. I feel that could speed things immensely.

**Describe alternatives you've considered**
I tried using the AFRange, but it doesn't seem to have much, if any, effect on my focusing

We need speed for our focusing because we have to have bright lights on in the rainforest during the focus cycle and that eats batteries.

davidplowman, 2024-06-13:
My belief is that when it's running in continuous AF mode, then the position it's at should "seed" an autofocus cycle, though I don't know if that will help in practice if it's falling back to the slow contrast-based search. You can also [edit the ranges ](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/vc4/data/imx708.json#L619)in the imx708.json tuning file (substitute the name of the correct tuning file if it's a third party camera), which I would hope might make a significant difference.

@njhollinghurst Should anything else be changed if the ranges are shortened? Anything else you can think of?

njhollinghurst, 2024-06-13:
I tried to implement the libcamera controls spec, though it wasn't always clear, which is how the ranges (and speeds) came to be defined as they are. Yes just changing the range limits should be fine.

Currently the scan (when it falls back to CDAF) always starts from one end; otherwise it might have to do a sort of triangular-wave thing which wouldn't necessarily be faster. I'll take another look at it...

quitmeyer, 2024-06-13:
> My belief is that when it's running in continuous AF mode, then the position it's at should "seed" an autofocus cycle,

Yeah I had tried code like this modified from your earlier example:

```
picam2 = Picamera2()
picam2.set_controls({"AfMode":2,"AfSpeed":1,"AfRange":0, "LensPosition":8.0})
time.sleep(2)
picam2.pre_callback = print_af_state
picam2.start(show_preview=True)
success = picam2.autofocus_cycle()
picam2.pre_callback = None
```

And setting the Lensposition sometimes makes its initial behavior slightly different like it will rapidly jump from 1.0-near the target, but then it will give up quickly, and go back to 0 and start slowly focusing

here's the output from above
```
Idle 1.0
Idle 1.0
Idle 1.0
Idle 1.0
Scanning 1.0
Scanning 2.1493079662323
Scanning 2.1493079662323
Scanning 2.1493079662323
Scanning 3.2953310012817383
Scanning 3.2953310012817383
Scanning 3.2953310012817383
Scanning 3.2953310012817383
Scanning 4.4032111167907715
Scanning 4.4032111167907715
Scanning 4.4032111167907715
Scanning 4.4032111167907715
Scanning 5.3449788093566895
Scanning 5.3449788093566895
Scanning 5.3449788093566895
Scanning 5.3449788093566895
Scanning 5.862688064575195
Scanning 5.862688064575195
Scanning 5.862688064575195
Scanning 5.862688064575195
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 6.252727508544922
Scanning 2.252727508544922
Scanning 2.252727508544922
Scanning 0.25272753834724426
Scanning 0.25272753834724426
Scanning 0.0
Scanning 0.0
Scanning 0.0
Scanning 0.0
(and then it keeps going for 90 seconds)
```

njhollinghurst, 2024-06-13:
OK I see the issue with starting position. Actually two issues:

1. Controls within a single request cannot be ordered. As described by the comment [here](https://github.com/raspberrypi/libcamera/blob/eb00c13d7c9f937732305d47af5b8ccf895e700f/src/ipa/rpi/common/ipa_base.cpp#L721), the mode change will be applied before the lens position, which means that the given lens position is ignored.

Now, picamera2's `autofocus_cycle()` will change the AF mode to Auto, so you can omit `"AfMode":2,` from the initial set of controls. That should set an initial lens position. However it might not help because...

2. As noted, the CDAF scan always starts from the lower limit. (Does this camera have PDAF support?) There are no immediate plans to change that; you might get better results by changing the limits in the tuning file, to cover just the range of distances you expect to need.

(The 90-second timeout should be dealt with elsewhere.)

quitmeyer, 2024-06-13:
> 1. Controls within a single request cannot be ordered.

oh interesting, so i could do it piecemeal like

picam2.set_controls({ "LensPosition":8.0})
picam2.set_controls({"AfSpeed":1})
picam2.set_controls({"AfRange":0})

> (Does this camera have PDAF support?) 
Arducam says they have had PDAF support for a year on these cameras, but it does still look like it's going CDAF

Any idea where these tuning files live on a raspberry pi?

njhollinghurst, 2024-06-13:
>Any idea where these tuning files live on a raspberry pi?

It depends on how libcamera was installed, and if it's Raspberry Pi 5 or earlier. It might be something like:
```
/usr/share/libcamera/ipa/rpi/pisp/arducam_64mp.json
/usr/share/libcamera/ipa/rpi/vc4/arducam_64mp.json
```
(the "pisp" one is for Raspberry Pi 5). If they're not found, try `find /usr -name arducam_64mp.json` (or name of camera if different).

Within the file, search for the string `"ranges"` then it's the numbers found underneath.

Rather than edit the file in situ it's safer to make a local copy, then set the environment variable `LIBCAMERA_RPI_TUNING_FILE` to point to your modified version.

davidplowman, 2024-06-13:
> > 1. Controls within a single request cannot be ordered.
> 
> oh interesting, so i could do it piecemeal like
> 
> picam2.set_controls({ "LensPosition":8.0}) picam2.set_controls({"AfSpeed":1}) picam2.set_controls({"AfRange":0})

No, because they'll still get bundled up and sent together whether the next bunch of stuff is forwarded to the camera. But it won't matter here because I don't think any of these race against one another. And the camera starts in manual AF mode, so it's fine to use `"LensPosition"`. (It was sending the `"AfMode"` that would cause indeterminate behaviour.)

quitmeyer, 2024-06-13:
ok, so through some bashing around, i figured out a system that seems to let me

-  focus completely and consistently 
- AND focus quicker 
- AND not show a preview (because that was crashing when cron ran the scripts)

you can see a lot of my brute force examinations all commented out ;)
also note that 

```picam2 = Picamera2()
#capture_main = {"size": (9000, 6000), "format": "RGB888", }
#capture_main = {"size": (1920, 1080), }

#capture_config = picam2.create_still_configuration(main=capture_main,raw=None, lores=None)

#picam2.configure(capture_config)

#preview_config = picam2.create_preview_configuration(main={"size": (4000, 3000)})
#picam2.configure(preview_config)
#picam2.start_preview(show_preview=True)
#picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous,"AfSpeed": controls.AfSpeedEnum.Fast,})

#picam2.set_controls({"AfSpeed":1,"AfRange":0, "LensPosition":8.0})
#picam2.set_controls({"LensPosition":8.0})
#picam2.set_controls({"AfRange":1})

time.sleep(2)
picam2.pre_callback = print_af_state
#picam2.start(show_preview=True)
picam2.start(show_preview=False)

time.sleep(2)

picam2.set_controls({"LensPosition":8.0})
#picam2.set_controls({"AfSpeed":1})

time.sleep(3)
start=time.time()
#picam2.set_controls({"AfMode": 2})
success = picam2.autofocus_cycle()

time.sleep(15)

#success = picam2.autofocus_cycle()
#picam2.pre_callback = None
flashOff()
```

Also note that the intial value still doesn't seem to be "seeding" the focus directly. Look it drops right back down to 0. **BUT if i remove the lenposition 8.0  thing, it WILL not focus correctly**
so maybe the autofocus isn't starting from there, but it, um, feels more confident about getting to the correct value quicker? (doing some psychoanalysis of the code ahahha)
```
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
Idle 8.0
[... 333 lines cut ...]
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Scanning 7.706932067871094
Success 7.706932067871094
Autofocus completed! 8.347043752670288
Success 7.706932067871094
Success 7.706932067871094
Success 7.706932067871094
```

quitmeyer, 2024-06-13:
> Anyway, my bizarre hack works for me now, but im confused why!

Well after running totally fine for an hour on cron. I restarted it and had cron run the exact same things again, and unfortunately now it's also doing the partial focusing thing again? so weird. nothing changed!

njhollinghurst, 2024-06-14:
Could you repeat the experiments after `export LIBCAMERA_LOG_LEVELS=RPiAf:0`? That ought to generate a lot more logging, though it's rather hard to decipher.

My guess is that PDAF is only working (and perhaps not very well) when the lens is already close to the optimal focus position. It could be an artifact of the particular camera, which is not one we have tested, or because the scene doesn't contain much detail at the appropriate scale.

quitmeyer, 2024-06-17:
Changing the range and stuff was not working, and I set the log levels too, but have found the solution in the form of an Arducam user posting his own corrected version of Arducams install files in the past day or so. Looks like they had something janky going on in the past couple updates, and the user JakeBrowning's fix fixed it.

https://forum.arducam.com/t/arducam-64mpx-hawkeye-b0399-no-autofocus/7031/8

Also happy to report that focusing now takes about 1 second, and it **DOES in fact start focusing from the value that the LenPosition was at.** So @davidplowman 's original assumptions were correct if your camera is functional!

Thanks for your help everyone! really appreciate it, and the moths do too. they are looking much more glamorous now!

```
dle 8.0  time  2024-06-17 10:07:45
Scanning 8.0  time  2024-06-17 10:07:45
Scanning 7.7463836669921875  time  2024-06-17 10:07:45
Scanning 7.7463836669921875  time  2024-06-17 10:07:45
Scanning 7.574095726013184  time  2024-06-17 10:07:45
Scanning 7.574095726013184  time  2024-06-17 10:07:45
Scanning 7.4649176597595215  time  2024-06-17 10:07:45
Scanning 7.4649176597595215  time  2024-06-17 10:07:45
Scanning 7.4649176597595215  time  2024-06-17 10:07:46
Scanning 7.4649176597595215  time  2024-06-17 10:07:46
Scanning 7.4649176597595215  time  2024-06-17 10:07:46
Scanning 7.4649176597595215  time  2024-06-17 10:07:46
Scanning 7.4649176597595215  time  2024-06-17 10:07:46
Success 7.4649176597595215  time  2024-06-17 10:07:46
```