raspberrypi/picamera2 #1056: [BUG]ImportError: cannot import name 'ControlType' from 'libcamera' (unknown location)

boyit1309, 2024-06-18:
**Please only report one bug per issue!**
**Describe the bug**
After install Rasberry Pi OS updates, picamera2 now fails to initialize
i also run :

source a-env/bin/activate
sudo apt install libcamera-apps libcamera-dev python3-libcamera
pip install numpy --upgrade
pip install picamera2

**To Reproduce**
import io
import time
import libcamera
from picamera2 import Picamera2

picam2 = Picamera2()
capture_config = picam2.create_still_configuration()
picam2.configure(picam2.create_preview_configuration())
picam2.start()

time.sleep(1)
data = io.BytesIO()
picam2.capture_file(data, format='jpeg')
print(data.getbuffer().nbytes)

time.sleep(1)
data = io.BytesIO()
picam2.switch_mode_and_capture_file(capture_config, data, format='jpeg')
print(data.getbuffer().nbytes)

**Expected behaviour**
run file a.py to test 

**Console Output, Screenshots**
$ python a.py 

Traceback (most recent call last):
  File "/home/pi/Documents/a.py", line 4, in <module>
    from picamera2 import Picamera2
  File "/home/pi/Documents/a-env/lib/python3.11/site-packages/picamera2/__init__.py", line 5, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/pi/Documents/a-env/lib/python3.11/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/pi/Documents/a-env/lib/python3.11/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Rectangle, Size
ImportError: cannot import name 'ControlType' from 'libcamera' (unknown location)

**Hardware :**
Pi 3B
Camera Module V3 – 12MP

**Additional context**
uname -a
Linux pi 6.6.32-v8+ #1769 SMP PREEMPT Thu Jun  6 11:38:54 BST 2024 aarch64 GNU/Linux
python -V
Python 3.11.2

apt show python3-libcamera 
Package: python3-libcamera
Version: 0.3.0+rpt20240617-1

libcamera-hello --version
rpicam-apps build: 49344f2a8d18 17-06-2024 (12:09:08)
libcamera build: v0.3.0+65-6ddd79b5

Name: picamera2
Version: 0.3.18
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /home/pi/Documents/a-env/lib/python3.11/site-packages
Requires: av, numpy, PiDNG, piexif, pillow, python-prctl, simplejpeg, v4l2-python3
Required-by:

davidplowman, 2024-06-18:
Installing libcamera into a venv is actually quite tricky. There are some instructions on our camera forum, but I really wouldn't recommend trying that if you can avoid it. The easiest solution is to create your venv with `--system-site-packages` and you will have all the packages you need automatically, without the need to install anything else.

boyit1309, 2024-06-22:
> Installing libcamera into a venv is actually quite tricky. There are some instructions on our camera forum, but I really wouldn't recommend trying that if you can avoid it. The easiest solution is [...]

i reinstall os and try your solution and finally it works. Thank you.

closed 2024-06-22 by boyit1309