raspberrypi/picamera2 #1061: [HOW-TO] change the exposition mode and constraint?

oralian, 2024-06-21:
**Describe what it is that you want to accomplish**
I'm trying to play around with `AeConstraintMode` and `AeExposureMode` as [documented here](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). However, after trying out multiple configurations, I see no change in my exposure times and gains.

**Describe alternatives you've considered**
He're a snippet of what I tried:

```python
import time

from libcamera import controls
from picamera2 import Picamera2

controls = {
    "constraint_normal": {
        "AeConstraintMode": controls.AeConstraintModeEnum.Normal,
    },
    "constraint_highlight": {
        "AeConstraintMode": controls.AeConstraintModeEnum.Highlight,
    },
    "constraint_shadows": {
        "AeConstraintMode": controls.AeConstraintModeEnum.Shadows,
    },
    "exposure-mode_normal": {
        "AeExposureMode": controls.AeExposureModeEnum.Normal,
    },
    "exposure-mode_short": {
        "AeExposureMode": controls.AeExposureModeEnum.Short,
    },
    "exposure-mode_long": {
        "AeExposureMode": controls.AeExposureModeEnum.Long,
    },
    "constraint_normal_and_exposure-mode_normal": {
        "AeConstraintMode": controls.AeConstraintModeEnum.Normal,
        "AeExposureMode": controls.AeExposureModeEnum.Normal,
    },
    "constraint_highlight_and_exposure-mode_short": {
        "AeConstraintMode": controls.AeConstraintModeEnum.Highlight,
        "AeExposureMode": controls.AeExposureModeEnum.Short,
    },
    "constraint_shadows_and_exposure-mode_long": {
        "AeConstraintMode": controls.AeConstraintModeEnum.Shadows,
        "AeExposureMode": controls.AeExposureModeEnum.Long,
    },
}

if __name__ == "__main__":
    picam2 = Picamera2(1)

    for control in controls:
        picam2.set_controls(controls[control])
        picam2.configure(
            picam2.create_video_configuration(
                main={"format": "RGB888", "size": (1640, 1232)}
            )
        )
        picam2.start()

        time.sleep(4)

        metadata = picam2.capture_metadata()
        exposure = metadata["ExposureTime"]
        gain = metadata["AnalogueGain"]

        print(f"{control} -> exposure: {exposure / 1_000} millisec, gain: {gain}")

        picam2.capture_file(f"output/{control}.jpg")
        picam2.stop()
```

**Additional context**
I'm using a Raspberry Pi camera v2 with a Raspberry Pi Compute Module 4. Thanks!

davidplowman, 2024-06-21:
Hi, I think you might just need to set the controls after calling `configure()` (see section 5.1.1.2 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)). (Though I guess the manual could be clearer that `configure()` resets any pending controls, which is why you need to set them afterwards, but before `start()`.)

oralian, 2024-06-21:
Thanks @davidplowman, it was exactly that :upside_down_face:

closed 2024-06-21 by oralian

KlausDecOpt, 2024-08-05:
We are using a RPI V3 camera wide w/ a RPI4b. As recommended we put the picam2.set_controls behind the picam2.configure. But still ran into two problems:
1) WARN RPiAgc agc.cpp:564 No constraint list shadows
2) All other configurations are working, but little changes in the camera images.
Any hints?

        Thanks,
              Klaus

davidplowman, 2024-08-05:
Hi again. Yes, looking at the imx708_wide.json file it seems like the "shadows" mode is missing. Don't really know what's happened there. However, it's very easy for you to add it into the tuning file yourself. Tuning files for a Pi 4 are normally found in /usr/share/libcamera/ipa/rpi/vc4. Look for the HQ cam tuning file (imx477.json) and search for "shadows", which you should find just after a block of parameters named "highlight". We'll be wanting to copy and paste that parameter block.

Now go to imx708_wide.json and search for "highlight". At the end of those parameters add an extra comma (,) and then paste in the "shadows" parameters from imx477.json. Be sure to add the parameter block exactly as it is in imx477.json, otherwise your camera may refuse to start! I'd also recommend taking a back-up of imx708_wide.json beforehand just in case things go wrong.

It's possible that the "shadows" mode doesn't give you the effect that you want, in which case you're of course very welcome to edit the definition to something that works better for you.

Many of those control parameters have quite subtle effects. For example, changing the exposure mode doesn't normally change the overall brightness, it just changes the balance between shutter time and gain. It's normally the constraint mode that has by far the greatest effect. If you haven't done so, please do read section 5.9 of the [tuning guide](https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf).

Section 5.11 might also be of interest. If you want brighter shadows without blowing everything else out completely, then increasing the values at the bottom end of the gamma curve might be helpful.