raspberrypi/picamera2 #1064: [OTHER] Please improve Documentation

savejeff, 2024-06-25:
TLDR: Getting the camera V2 running on a Raspberry Pi 4 is more frustrating than it should be. Please Update old tutorials 

Here is what happened to me: I got the Camera a while ago and today installed a blank image of Raspberry Pi OS 64bit. I Followed the [official tutorial](https://projects.raspberrypi.org/en/projects/getting-started-with-picamera) which is completely outdated. Not a single command in that tutorial still works. other tutorials like [this](https://randomnerdtutorials.com/guide-to-raspberry-pi-camera-v2-module/) also use old commands like raspi-still and picamera in python. 

only be multiple web searches I had to go through this gits example folder to get something going. Also when using X-forwarding via SSH most of the commands crash. 

this is all solvable for me as an experienced embedded developer, but as Raspberry Pis and accompanying products are intended as educational products. while my described experience is pretty much realistic in terms of what you will experience as a software developer but it should be easier to get a first picture going

This post might be off-topic here but its the best place I found to post this. 
greetings

davidplowman, 2024-06-26:
Hi, and thank you for raising this. The legacy camera system has been deprecated for a number of years now, however, as you've noted, getting documentation updated can be very difficult.

The first link you mention belongs to Raspberry Pi Foundation. Obviously we have a strong relationship with them so I can raise this with them and ask if them if they might be able to improve the situation. The other link is to a third party website. Unfortunately the web is just full of these, with outdated documentation, and we have no significant ability to get anything updated. You could try posting a message if you can, asking them to withdraw or update the page, but you will probably have no more success than we do.

Some pages that you should follow would include:
https://www.raspberrypi.com/documentation/accessories/camera.html - about the camera hardware and connecting it
https://www.raspberrypi.com/documentation/computers/camera_software.html - camera software and rpicam-apps
https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf - for Python users

The biggest take away is that, once you've connected your camera and installed the Raspberry Pi OS, you _don't need to do anything special_. The cameras should "just work".

savejeff, 2024-06-28:
Thanks for the fast and fair response. I know how things are usually behind the scenes. I wanted to give a perspective from somebody on the outside. 
Raspi and the also the Camera stuff is such an important thing for the next generation of coders/developers. we should make it attractive for them to enter the field. 

also thanks for the links. I've managed to get the images into OpenCV 👌

closed 2024-06-28 by savejeff