raspberrypi/picamera2 #1074: disable logging when taking picture

LeifSec, 2024-07-22:
Following [issue](#307) I want to switch off logging / set logging level to ERROR for calling a function 

```
camera.switch_mode_and_capture_image(
                camera_config=self.__settings.still_configuration,
                name="main",
                wait=None,
                signal_function=None)
``` 
.
So I did:
``` 
if "picamera2" not in logging.Logger.manager.loggerDict.keys():
            raise ValueError("not in tt")
logger = logging.getLogger("picamera2")
logger.setLevel(logging.ERROR)
camera.switch_mode_and_capture_image(....
logger.setLevel(old_level)
```

Unfortunately there is still this INFO level logging:
```
163:48:59.073898409] [27530]  INFO Camera camera.cpp:1183 configuring streams: (0) 1536x864-BGR888 (1) 4608x2592-SBGGR10_CSI2P
[163:48:59.097958056] [27500]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
[163:48:59.516137626] [27530]  INFO Camera camera.cpp:1183 configuring streams: (0) 1536x864-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[163:48:59.525447463] [27500]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
``` 

I am using `imx708` camera and latest Raspberian 12 software.

davidplowman, 2024-07-22:
Yes, this is all libcamera logging and it can be quite chatty. It might be easiest just to put something like `export LIBCAMERA_LOG_LEVELS=*:3` in your `.bashrc`...

LeifSec, 2024-07-22:
`export LIBCAMERA_LOG_LEVELS=*:3`works - thank you very much. But as I need it in the code (to disable it just for that call of `switch_mode_and_capture_image` ) I need something like:
```
os.environ["LIBCAMERA_LOG_LEVELS"] = "*:3"
camera.switch_mode_and_capture_image(...`
os.environ["LIBCAMERA_LOG_LEVELS"] = "*:old_level"
``` 

But this does not work.

davidplowman, 2024-07-22:
Yes, libcamera probably only reads that environment variable when it starts up. So far as I know, there's no Python interface to libcamera's logging system after that. So I don't really have a workaround for you there. Can you say which of the logging messages are actually useful? I'm wondering whether you can turning them all off, and add your own logging for your application.