raspberrypi/picamera2 #1081: [OTHER] IndexError

mbauer575, 2024-07-30:
Getting an IndexError when running picamera2 on rpi 5.  I was able to run this program awhile ago but now im getting this error every time I run it. 

```
    picam2 = Picamera2()
             ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 246, in __init__
    camera_num = self.global_camera_info()[camera_num]['Num']
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
IndexError: list index out of range
```

Any ideas what could be wrong?

davidplowman, 2024-07-31:
What did `self.global_camera_info()` return, and what was `camera_num`? One possibility is that your camera (or cameras) are not being detected for some reason.

mbauer575, 2024-08-09:
I reseated the camera cable and it seems to work fine now.

closed 2024-08-09 by mbauer575