raspberrypi/picamera2 #1089: [OTHER] Blocks of code overflowing in PDF documentation

Dad0u, 2024-08-09:
Hello, 

I noticed several instances of text overflowing out of the page in the PDF documentation at https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf
Here is an example (opened in Firefox) :
[image]

The pdf renders the same under Acrobat Reader. This only seems to affect blocks of code. An extreme example can be seen on page 60 where and entire block of code is squashed on a single line : 

[image]

I ran a quick visual check and noticed the issue on pages 11, 20, 23, 24, 25, 27, 60, 61, 62 and 65

This harms readability of the document. Could this be fixed ? Is there another location where the doc can be accessed that is not subject to these formatting issues ?

davidplowman, 2024-08-09:
Hi, and thanks for reporting this.

@nathan-contino Hi Nate, could you maybe take a look at this one? When I generate the pdf myself locally, it all looks OK - so I don't know what's up with the "official" version. Thanks!

nathan-contino, 2024-08-09:
Thanks for letting us know.

No clue what went wrong with the older version of the PDF. Sorry about that.

I've uploaded a new version now that doesn't have any of these overflow issues. If you notice anything else that warrants an update, give me a holler and I'll fix it up.

nathan-contino, 2024-08-09:
@davidplowman feel free to close this issue now; alternatively, you could give me permissions to close issues, and I promise I'll only close docs ones. Thanks.

davidplowman, 2024-08-09:
Thanks @nathan-contino !

closed 2024-08-09 by davidplowman

Dad0u, 2024-08-09:
Thank you for the quick fix !