raspberrypi/picamera2 #1098: [BUG] LensPosition must be set to 4 in manual focus for "Infinity"

tutoduino, 2024-08-25:
The manual focus is not "Infinity" when LensPosition is set to 0.0. 
With LensPosition between 0 and 3, the horizon is blurred, it is clear when LensPosition = 4

My hardware is : Raspberry Pi Zero 2 W + Raspberry camera v3 (wide 120°)

Here is the test program:
```
import libcamera
import time
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.configure(picam2.create_still_configuration())
picam2.start()

for lens_position in range(35):
    picam2.set_controls({"AfMode": libcamera.controls.AfModeEnum.Manual, "LensPosition": lens_position})
    time.sleep(1)
    r = picam2.capture_request()
    filename = "lens-position" + str(lens_position) + ".jpg"
    metadata = r.get_metadata()
    actual_lens_position = metadata["LensPosition"]
    print("Actual LensPosition = {}, file capture in file {}".format(actual_lens_position,filename))
    r.save("main", filename)
    r.release()

picam2.stop()
```

davidplowman, 2024-08-27:
Hi, thanks for the report.

It's worth saying that the lens position is always somewhat approximate because there is variation between different modules. However, in this case it seems to be behaving in a significantly different manner from what we expect. By any chance, do you have any other modules to compare this one against?

It does seem to me like something is significantly wrong here, and you may have a faulty module which would need to be returned to the place where you bought it.

@njhollinghurst Anything else to add, do you think?

njhollinghurst, 2024-08-27:
There is some variation, but this is well outside the expected range and should have been caught by factory test. Unfortunately this does suggest the camera may have been damaged during handling or shipping, and should be raised with the vendor.

If the camera otherwise works well and you want to keep it, it may be possible to edit the camera tuning file.

Other things to check: Do you have any other optics (lenses, mirrors, etc)? Is the camera tilted vertically upwards?

tutoduino, 2024-08-29:
Thanks for your reply.
No other optics. 
The camera is tilted 45°.

davidplowman, 2024-08-29:
Tilting the camera will generally make a difference, though it's hard to say how much. Does it behave better when it's not tilted?

tutoduino, 2024-09-06:
Tilting the camera seems to have an impact, but quite difficult to measure with my test conditions... Any official input from Raspberry would help.

njhollinghurst, 2024-09-06:
The effect is intrinsic to the camera. But if it's permanently mounted in that orientation, it can be calibrated for.

Take a copy of the file `/usr/share/libcamera/ipa/rpi/vc4/imx708_wide.json` and find the `"map":` line which is near the bottom.
Edit it to something this (to increase focus by 4.0):
```
                "map": [ 0.0, 475, 33.0, 950 ]
```

Test the modification using:
```
tuning = Picamera2.load_tuning_file("imx708_wide.json", dir=".")
picam2 = Picamera2(tuning=tuning)
```
If you're happy with it, you can put it back into the default directory.

Alternatively I guess you could edit "tuning" programmatically in Python:
```
tuning = Picamera2.load_tuning_file("imx708_wide.json")
next(m for m in tuning['algorithms'] if 'rpi.af' in m.keys())['rpi.af']['map']=[0.0, 475, 33.0, 950]
picam2 = Picamera2(tuning=tuning)
```