raspberrypi/picamera2 #1100: [HOW-TO] Reduce Global Shutter with HQ-Camera

mmoollllee, 2024-08-27:
We want to use the HQ-Camera attached to a slightly moving object.
Currently we're facing blurry parts in the image, due to global shutter of the camera.

How can we reduce this effect as good as possible, operating with a RPi 3A+?
Is the ExposureTime in metadata the only value we have to consider to reduce?

Hope to find some tipps on how to make the most out of it :)

Our Code:
```py
# Limit ExposureTime to reduce blur due to GlobalShutter by daytime
if (timeofday == "day"):
   AeExposureMode = controls.AeExposureModeEnum.Short
   FrameDurationLimits = (50, 5000)
else:
   AeExposureMode = controls.AeExposureModeEnum.Long
   FrameDurationLimits = (100, 10000000)

with Picamera2() as picam2:
   picam2.start(show_preview=False)

   picam2.set_controls({
      "AeFlickerMode": controls.AeFlickereModeEnum.FlickerOff,
      "AeMeteringMode": controls.AeMeteringModeEnum.Custom if metering else controls.AeMeteringModeEnum.Spot,
      "AeExposureMode": AeExposureMode,
      "FrameDurationLimits": FrameDurationLimits
   })

   # auto exposure might take a few moments settle
   time.sleep(0.5)

   ctrl={}
   capture_config = picam2.create_still_configuration(ctrl)
   time.sleep(2)

   metadata = picam2.capture_metadata()

   picam2.switch_mode_and_capture_file(capture_config, "file.jpg")

   picam2.stop()
```

davidplowman, 2024-08-27:
Yes, I think the exposure time is the only thing that will affect the blurriness of the final image. You can keep this short by setting it directly (and the analogue gain will adapt according to the exposure time chosen), or by selecting an appropriate exposure mode.

I notice that you've chosen the "short" exposure mode which is an appropriate thing to do. If you wanted, you are free to edit the parameters of the "short" exposure mode in the camera tuning file (search for imx296.json on the Pi's filesystem).

FrameDuration can limit the exposure time, so it's a reasonable way to constrain what the AE algorithm will do, but won't directly affect blurriness (except by forcing a different exposure time to be used).

mmoollllee, 2024-08-27:
For a deeper understanding of what is happening: Does it make a difference if we use a faster RaspberryPi? Or is the speed limited by the camera sensor?

davidplowman, 2024-08-27:
No. It's all down to the camera sensor. I think the only factor is the exposure time.

You mentioned that you're using the HQ camera. I just wanted to add that the HQ cam is not a "global shutter camera". It employs a "rolling shutter", like most of our cameras (except the "Global Shutter Camera" that we also sell). The rolling shutter can have an effect on an image, in that the top of the image is not exposed at the same moment as the bottom, so you can get some strange motion artefacts. However, the _blurriness_ of any part of the image is not affected by when the exposure happened, only how long it was.

mmoollllee, 2024-08-27:
I'm so sorry, I've been talking about rolling shutter and wrote global shutter in the beginning... I changed the title here for anyone searching for this...
Thanks for the informations! I'll give it a try and report back :)

davidplowman, 2024-08-27:
The exposure time in your metadata is extremely short, so I don't think you can improve that. The capture should have the same exposure time, but of course you may be subject to increased rolling shutter effects. I guess I'm a bit confused about what the problem really is - motion blur, rolling shutter effects, or lens problems? Could you maybe clarify that?

You can certainly capture a stream of images, there's no problem with that. Though if single images are showing a problem, then you may end up with a stream of images with the same problem. Capturing a stream of images is usually limited by the time taken to write them out. The HQ cam can run up to 10fps in the full resolution mode, though to achieve that you need fast SSDs - a micro SD card is unlikely to keep up.

mmoollllee, 2024-08-27:
[Here's an example](https://cam6.blautalbau.de/)
If you click the last day in the timeline at the bottom and hover over those images you can see, that little movements of the crane cause parts of the image to be blurry and distorted.
I'd except capture time to be less than 100ms, but then this seems to be too much of distortion...

mmoollllee, 2024-08-28:
Okay you nailed it to the lens and you're right. It seems to be just a little little bit loose! And there seems to be some erruption, combined with the rolling shutter to make this effect.
I'm sorry I was on the wrong track. Thank you so much for your time and the final hint!

closed 2024-08-28 by mmoollllee